Koenkk/zigbee2mqtt #23125: Cannot bind an endpoint

ProjectInitiative, 2024-06-21:
### What happened?

Tried to bind a secondary endpoint. This worked a few months ago, but have stopped with the following error:

```bash
error 2024-06-21 12:30:35Frame remoteSetBindingHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readUInt8 (node:internal/buffer:254:5) at Buffer.readUIntLE (node:internal/buffer:184:17) at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23 at Array.every (<anonymous>) at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15) at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35)
error 2024-06-21 12:30:35Unparsed frame 0x31. Skipped
```

### What did you expect to happen?

Expected to use the secondary functions of a switch previously.

### How to reproduce it (minimal and precise)

Using Jasco Enbrighten switches, try to add another binding with source = 2, destination = coordinator, destination endpoint = 1

### Zigbee2MQTT version

1.38.0 commit: fe048e6

### Adapter firmware version

6.7.8.0 build 373

### Adapter

EZSP

### Setup

Docker container, arm

Koenkk, 2024-06-22:
You are using the old `ezsp` driver, it's recommended to use the new `ember` drivers ([info](https://github.com/Koenkk/zigbee2mqtt/discussions/21462))

ProjectInitiative, 2024-06-23:
The ember driver does not work with my HSUSBZB-1 adapter. I've upgraded its firmware.

ProjectInitiative, 2024-06-23:
I see, I'll probably need to get a newer adapter

closed 2024-12-29 by github-actions[bot]