Koenkk/zigbee2mqtt #23127: disabled device still pinged

srg777, 2024-06-22:
### What happened?

I have disabled my device to prevent it is been binged, but not a luck - I still receiving error messages about failed pings

### What did you expect to happen?

i expect that zigbe2mqtt will not ping disabled device

### How to reproduce it (minimal and precise)

come to devise setting page and check disabled option, in the device list it marked as disabled

### Zigbee2MQTT version

1.38.0-dev commit: a9b8808

### Adapter firmware version

20220103

### Adapter

MSH

### Setup

MSH device

Koenkk, 2024-06-23:
Could you try with the latest dev branch?

srg777, 2024-06-23:
I have 1.38.0-dev

Koenkk, 2024-06-24:
But not the latest one (https://github.com/Koenkk/zigbee2mqtt/commit/a9b8808237554316c5c97e9d9c990048733779e8)

srg777, 2024-06-24:
> But not the latest one (a9b8808 <https://github.com/Koenkk/zigbee2mqtt/commit/a9b8808237554316c5c97e9d9c990048733779e8>)

It seems I have the latest

https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.38.0
￼

closed 2024-12-29 by github-actions[bot]