Koenkk/zigbee2mqtt #23128: GL-C-006P (mini) bricked after OTA update

webns, 2024-06-22:
### What happened?

I've updated GL-C-006P (mini) firmware via Z2M OTA update, after that controller don't response to any commands (state, brightness, color...). Controller still can be paired/unpaired.

I assume that it's a same issue as #22287 but in reverse — mini controller got firmware for normal version. I've tried to flash this version — https://github.com/catduckgnaf/zigbee-OTA/blob/3cdcc512f4cc589ab112ecc1ed0adf585fbea005/images/Gledopto/GL-C-006P(MINI)_20451203_20240227.ota but without any success, logs says:
```
Images currently unavailable for device 'GL-C-006P', hardwareVersion '2', manufacturerName GLEDOPTO, {"fieldControl":0,"manufacturerCode":4687,"imageType":5176,"fileVersion":604057601}'
```

Current state of the controller:
```json
{
    "color_mode": "xy",
    "linkquality": 160,
    "state": "OFF",
    "update": {
        "installed_version": 604057601,
        "latest_version": 604057601,
        "state": "idle"
    },
    "update_available": false
}
```

### What did you expect to happen?

Mini version of controller doesn't receive updates for normal version 

### How to reproduce it (minimal and precise)

Update mini version of GL-C-006P via OTA to latest version.

### Zigbee2MQTT version

1.38.0 commit: fe048e6

### Adapter firmware version

 7.4.1 [GA]

### Adapter

SLZB-07

### Setup

Docker

webns, 2024-06-22:
Disabling OTA updates for GL-C-006P could be a good idea for preventing bricking more devices, at least while problem is being solving

Koenkk, 2024-06-22:
Until it's more clear how to properly update these devices, I've disabled all GL-C-006P OTAs: https://github.com/Koenkk/zigbee-OTA/commit/ae36142be9858dd8eaf974215ad300806ede8515

webns, 2024-06-22:
Noticed that both updates (normal and mini) had the same `"imageType": 5138` but after the update in the normal version it changes to 5176 while the mini version keeps 5138. That's why it's possible to recover a normal controller with mini's firmware.

I've tried to edit imageType in the update for the mini controller but OTA update doesn't start with an error
```
"error":"Update of '0xa4c138a4688554cd' failed (OTA: Timeout, device did not request any image blocks)"
```

For now I have no idea how to recover mini version of the controller

CoolID2, 2024-06-22:
I've got the same problem on two of my normal GL-C-006P.
Actually no chance zu bring them up again. The workaround #22287 won't help.

webns, 2024-06-23:
Looks like the name of the controller doesn't matter, mini/normal is only a name, the real difference is the hardware they used to build the controller and we can't differentiate one hw from another

Lukagian, 2024-06-23:
I have the same problem. I try to contact gledopto for a working firmware, I hope they will help us to recover the device.

Andrik45719, 2024-06-23:
workaround
https://github.com/Andrik45719/GL-C-006P_mini_fix

Andrik45719, 2024-06-23:
> I have the same problem. I try to contact gledopto for a working firmware, I hope they will help us to recover the device.

https://github.com/Koenkk/zigbee-OTA/blob/6384e5e89df73bf4b141485129dccdfe091eedbf/images/Gledopto/GL-C-006P(MINI)_20451203_20240227.ota

webns, 2024-06-23:
> https://github.com/Koenkk/zigbee-OTA/blob/6384e5e89df73bf4b141485129dccdfe091eedbf/images/Gledopto/GL-C-006P(MINI)_20451203_20240227.ota

It can't be installed because imageType of the firmware is 5138, while broken mini's imageType is 5176

Lukagian, 2024-06-23:
> workaround https://github.com/Andrik45719/GL-C-006P_mini_fix

With this workaround now I can control the device. But it is recognized as GL-C-008P.
[image]

Andrik45719, 2024-06-23:
To flash FW header is patched to cheat z2m
002P to 5176
006p mini to 5145

webns, 2024-06-23:
> To flash FW header is patched to cheat z2m 002P to 5176 006p mini to 5145

Have you patched only header? I was trying to patch OTA update but my controller doesn't accept patched file with error thats says image is not correct

Andrik45719, 2024-06-23:
> > To flash FW header is patched to cheat z2m 002P to 5176 006p mini to 5145
> 
> Have you patched only header? I was trying to patch OTA update but my controller doesn't accept patched file with error thats says image is not correct

https://github.com/Andrik45719/GL-C-006P_mini_fix/blob/main/GL-C-006P_mini_fix.7z

'GL.ota_002'  is patched GL-C-002P_V20551203_20240125.ota 
'GL.ota_mini_patched' is patched GL-C-006P(MINI)_20451203_20240227.ota

webns, 2024-06-23:
> With this workaround now I can control the device. But it is recognized as GL-C-008P.

There is a list of different versions of the controller at the end of the OTA update 
[image]

Maybe the same firmware can work on different controllers

Andrik45719, 2024-06-23:
> > workaround https://github.com/Andrik45719/GL-C-006P_mini_fix
> 
> With this workaround now I can control the device. But it is recognized as GL-C-008P. <img alt="Screenshot 2024-06-23 alle 20 16 29" width="1247" src="https://private-user-images.githubusercontent.com/116172987/342107007-46f6a765-de1f-4df8-8f94-1956c0c9eb6c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkxNjY5MTgsIm5iZiI6MTcxOTE2NjYxOCwicGF0aCI6Ii8xMTYxNzI5ODcvMzQyMTA3MDA3LTQ2ZjZhNzY1LWRlMWYtNGRmOC04Zjk0LTE5NTZjMGM5ZWI2Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyM1QxODE2NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNjI2YTQwMTczNWMwNzQ2NDVkMmEwNmQ0MTE4ZTcyZDM3NTcyYWYwNjcwNjYzOTg5MmJiMWYzYzcyZjE3YzE5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.7dDobU4KEf3AUHeED36tfTHPxDqbipqCEg-NwWEHtZk">

You have to update two times
1st one to GL-C-002P/008P/007P
2nd to GL-C-006P mini

Andrik45719, 2024-06-23:
> > With this workaround now I can control the device. But it is recognized as GL-C-008P.
> 
> There is a list of different versions of the controller at the end of the OTA update <img alt="image" width="689" src="https://private-user-images.githubusercontent.com/26829020/342107383-96f899d4-9835-4046-8ad4-25c15a423718.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkxNjc0NDYsIm5iZiI6MTcxOTE2NzE0NiwicGF0aCI6Ii8yNjgyOTAyMC8zNDIxMDczODMtOTZmODk5ZDQtOTgzNS00MDQ2LThhZDQtMjVjMTVhNDIzNzE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjIzVDE4MjU0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNmZjNkNGRlMThhODUxYzBiYTI1YTE5NWFkN2I1ZmMzN2M1MTU3NDI5Mjk2MzY2MzhjYjFhYzdiOTEzY2E5ODYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.lJZbZUDp2TzJU7IYnvKyVjiHEBeYFPS1-1zbNDZCFRQ">
> 
> Maybe the same firmware can work on different controllers

The problem is to detect right modification of hardware.

webns, 2024-06-23:
> You have to update two times
> 1st one to GL-C-002P/008P/007P
> 2nd to GL-C-006P mini

That's interesting way, will try later, thanks for your research and helpful information

One thing I'm sure about is that OTA updates should be disabled for any controllers 😅

Lukagian, 2024-06-23:
After a restart of z2m the device was recognized as GL-C-006P. For me the fix worked great. Thank you very very much @Andrik45719

Reinhard-M, 2024-06-23:
My device was also bricked. With the patch and step by step description from @Andrik45719 it's fixed, thanks a lot for this 😊

webns, 2024-06-24:
Patch from @Andrik45719 worked for my controller too, thank you 👍

robertoclg, 2024-06-24:
The fix worked great for my two devices. Thank you very very much @Andrik45719

crazych1, 2024-06-24:
Maybe not in the subject, but suddenly who knows if it is possible to make the GL-C-006P_mini controller work at full capacity in neutral light mode? 12 volts per channel? In white or yellow light mode, the channel operates at 12 volts, but as soon as the neutral light is turned on, both channels operate at 10 volts. Maybe there's a way to fix it?

CoolID2, 2024-07-02:
Hi,
I have the normal GL-C-006P. Unfortunately also bricked by z2m. Now I have installed the patch from Andrik45719. 

- z2m rebooted 
- device relearned
- It is also recognized as GL-C-006P
But it cannot be controlled. It can only be switched off via z2m after relearning.

What am I doing wrong? 
Or does this firmware only work with the 006P mini?

Klikini, 2024-07-02:
@CoolID2 this issue is only for the mini. Since you have the normal version, take a look at this one: https://github.com/Koenkk/zigbee2mqtt/issues/21988#issuecomment-2052111667

closed 2024-12-29 by github-actions[bot]