Koenkk/zigbee2mqtt #23134: Zigbee2MQTT Web Ui not working after host update

nicola-spreafico, 2024-06-22:
### What happened?

Hello,
before the activities of this morning, all the infrastructure were working very fine since many months, Z2M ok, MQTT ok, HA ok, no issue.

This morning I did a Proxmox update as it offered me to install some updates. After that, the Z2M web interface stopped working with the ERR_CONNECTION_REFUSED error. 

I did not change anything in the Z2M LXC specifically, I even tried restoring a backup from yesterday, but the issue is still persisting. This is my configuration yaml, this file has had no changes for many weeks. The last modification I made around april is moving to the ember driver, as defined in the following configuration:
```
homeassistant: true
advanced:
  network_key:
    ....
  pan_id: 55519
  ext_pan_id:
    ....
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  transmit_power: 20
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 514
    protocol: udp4
    type: '5424'
  channel: 15
  log_level: debug
mqtt:
  base_topic: zigbee2mqtt_01
  server: mqtt://192.168.50.208:1883
  user: xxxxxx
  password: xxxx
  keepalive: 60
  reject_unauthorized: true
  version: 4
serial:
  port: tcp://192.168.50.204:6638
  baudrate: 115200
  adapter: ember
  disable_led: false
  rtscts: false
frontend:
  port: 9442
device_options:
  legacy: false
```

Besides the web interface not being accessible, it seems that the MQTT is not actually receiving any data, so all the functionalities are currently offline.

I did a quick check on all existing VM and LXC and it seems that the only issue generated by this update is this one, so I'm not sure if the root cause of the issue is on the Proxmox host side or the specific Z2M LXC.

I already lost 3 hours on this, I tried to reboot many times, restoring backup from yesterday, updating the coordinator firmware, rebooting both Z2M and MQTT, rebooting the coordinator itself, rebooting the entire Proxmox, checking the Z2M logs over and over, I'm currently out of ideas and a bit hopeless on this matter.

I appreciate any help I can receive on this.
Thank you

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

core firmware: v2.3.6, zigbee firmware: 20240510

### Adapter
[... 91 lines cut ...]
[2024-06-22 19:43:48] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:43:48] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:43:48] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "16298".
[2024-06-22 19:43:49] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "16298".
[2024-06-22 19:43:51] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "7782".
[2024-06-22 19:44:00] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:00] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:00] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:01] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "16298".
[2024-06-22 19:44:01] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:01] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:01] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:02] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:02] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:03] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:03] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "16298".
[2024-06-22 19:44:03] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "7782".
[2024-06-22 19:44:03] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "16298".
[2024-06-22 19:44:38] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "7782".
[2024-06-22 19:44:47] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:52] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "16298".
[2024-06-22 19:44:52] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:44:52] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "16298".
[2024-06-22 19:44:52] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".
[2024-06-22 19:45:02] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27845".

yayitazale, 2024-06-23:
I have a similar issue, but I just updated the coordinator firmware. Everything is working except the web ui.

BertrumUK, 2024-07-02:
I am seeing the same issue after updating Proxmox this morning. I am running the latest version 1.39.0-1, but when trying to get into the UI I get a cloudflare error as I run HA behind CF.

I am using port 8099 and even trying to connect via the local IP and port I get no response. Nothing is being reported in the logs either.

Z2MQTT works fine in every other way, it's just that I cannot reach the Web UI.

nicola-spreafico, 2024-07-02:
I want to add another experience I have, after I made it working as mentioned in my edit of first message, I had to shutdown the host where Proxmox is running, and once again after the startup I was having the same Z2M issue. So it seems something related of when the host is booting up (or at least Z2M is booting up).

I did again the trick of changing the driver back and forth, but it is not a definitive solution

BertrumUK, 2024-07-02:
I am running the Sonoff CC2652P adaptor, but I get no errors for the adapter or the Web UI in the logs. It was working this morning before the Proxmox updates so I am guessing that something changed there to stop the Web UI from working now.

Will wait and see if others have similar issues after carrying our a Proxmox update.

BertrumUK, 2024-07-02:
Rolling back to 1.38.0-1 fixed the issue for me. I updated to the latest update and I lose access to the UI again. Rolling back a 2nd time restores it.

mpuff, 2024-07-03:
i have this problem on a raspberry pi installation, was working for min 1 year, but now I see in the logs the ui port is open, but no access to the ui, also no open port when I check with ss -ltn in the terminal

Micr0mega, 2024-07-05:
I have similar symptoms, but have not updated my host system (Home Assistant Yellow). I did update to Home Assistant 2024.7.0 yesterday and for a few hours the web UI worked fine.

But after adding two new devices, the UI is not reachable anymore and gives `502: Bad gateway`. However, all devices are working fine. Tried multiple restarts of Z2M and HA itself, but only the web UI stays unreachable, everything else functions as expected.

Rolling back to 1.38.0-1 fixed it for me as well.

mpuff, 2024-07-05:
also another problem, with the safari browser the gui is never working, only with a browser like chrome for me :/

BertrumUK, 2024-07-06:
similar issue to [here](`https://github.com/Koenkk/zigbee2mqtt/issues/23266#issue-2392067402`)

Timmmy001, 2024-07-13:
Fix it with a STRG+F5 in Firefox (Delete Cache/Reload).
Chrome was ok.

BertrumUK, 2024-08-18:
I have managed to get mine working again with the latest update after removing all entries in the group.yaml file. I saw it mentioned [here ](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/624) and that allows the frontend to work with the latest update.

I deleted the 2 entries I had in groups from the UI before updating, and it has fixed it for me.

DaveDavenport, 2024-10-07:
I have the same issue with latest.  I get no UI, not even an open port (at 8080), everything is working, except the web UI.

Nothing in the log file about frontend or any other error.

stahlblau-max, 2024-10-11:
same here. no UI. no connection from z2m to MQTT host. happened over night with no change made.

DaveDavenport, 2024-11-14:
For me mqtt and everything works, just not the web interface.

DaveDavenport, 2024-12-11:
odd addition to the weirdness,,  the web-ui starts working after +- 3 days.  If I then restart, it takes 3 days again.

Is there some query that happens on startup that can block things?

mcneilc, 2025-01-10:
I have this issue as well as have raised a bug report at [https://github.com/Koenkk/zigbee2mqtt/issues/25715](url)

In my case, whether connected or not, Home Assistant isn't picking up any data from Z2M

TaylorWatson, 2025-01-15:
My solution was this

`frontend:
  enabled: true
  port: 8080
  host: 0.0.0.0`

seems like the newest update isn't automatically binding the container port to 0.0.0.0:8080 .... annoying but whatever. Add that to your configuration.yaml and it works.

cr0wm4n, 2025-02-25:
> My solution was this
> 
> `frontend: enabled: true port: 8080 host: 0.0.0.0`
> 
> seems like the newest update isn't automatically binding the container port to 0.0.0.0:8080 .... annoying but whatever. Add that to your configuration.yaml and it works.

Thankyou, this worked for me too :)

closed 2025-05-05 by github-actions[bot]