Koenkk/zigbee2mqtt #23143: The switch TS0001 (_TZ3000_hzlsaltw) does not display consumption data, as well as voltage

Iversat, 2024-06-23:
### What happened?

I purchased this relay unit via Aliexpress, but when connected to the z2m, it was identified only as a switch, but it does not output data (current and voltage)

### What did you expect to happen?

Relay with full output of indicators

### How to reproduce it (minimal and precise)

I don't know

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20220301

### Adapter

zstack сс2652

### Setup

huawei matebook d14 2020

### Debug log

[image]
[image]
[image]

Anry777, 2024-07-09:
The same problem
Cant see consumption on this device in z2m.. But in ZHA it works correctly.

Koenkk, 2024-07-10:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-10 by Koenkk

Anry777, 2024-07-13:
It work correctly. Grazz!!