Koenkk/zigbee2mqtt #23147: [New device support]:

MisterE1970, 2024-06-23:
### Link

https://de.aliexpress.com/item/1005006808281792.html?spm=a2g0o.order_list.order_list_main.5.521e5c5fvLJbvK&gatewayAdapt=glo2deu

### Database entry

0xa4c138b82718ecdb

### Comments

IEEE-Addresse
0xa4c138b82718ecdb
Netzwerk-Adresse
0xCF27

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_myd45weu',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

ustolt, 2024-06-25:
These device is recognized, but no soil_moisture detected.
I'm running two sensors TS0601, _TZE204_myd45weu under Z2M-Homeassistant addon  (1.38.0-dev commit: 30227a1) with zigbee herdsman converter version 19.59.2.

```json
{
    "battery": 90,
    "battery_state": "high",
    "last_seen": "2024-06-23T11:10:19+02:00",
    "linkquality": 128,
    "soil_moisture": 0,
    "temperature": 24,
    "temperature_unit": "celsius"
}
```

ustari28, 2024-06-26:
Here the same issue, I have configured zigbee2mqtt -> mqtt server -> homeassistant. How can I set up this sensor to get battery, humidity, temperature, soil_moisture?

semirhadzic, 2024-06-27:
Same here,

Tried to write external definition and implement it step by step but to no avail:

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601_soil',
    vendor: '_TZE204_myd45weu',
    description: 'Soil sensor',
    extend: [],
    fromZigbee: [
        fz.temperature,
        fz.soil_moisture,
        fz.battery
    ],
    toZigbee: [],
    exposes: [
        e.temperature(), 
        e.soil_moisture(), 
        exposes.enum('temperature_unit', ea.SET, ['celsius', 'fahrenheit']).withDescription('Temperature unit'),
        e.battery()
    ],
    options: [
        {name: 'temperature_calibration', type: 'number', description: 'Calibrates the temperature value (absolute offset)', minimum: -30, maximum: 60},
        {name: 'temperature_precision', type: 'number', description: 'Number of digits after decimal point for temperature', minimum: 0, maximum: 3},
        {name: 'soil_moisture_calibration', type: 'number', description: 'Calibrates the soil moisture value (absolute offset)', minimum: 0, maximum: 100},
        {name: 'soil_moisture_precision', type: 'number', description: 'Number of digits after decimal point for soil moisture', minimum: 0, maximum: 3},
    ]
};

module.exports = definition;
```

Zigbee2MQTT version
1.38.0 commit: unknown

[image]

[image]

Without external definition Z2M just reports unsupported and only report lqi strenght

Koenkk, 2024-07-03:
`_TZE204_myd45weu` is supported in z2m 1.39.0

closed 2024-07-03 by Koenkk