Koenkk/zigbee2mqtt #23155: _TZ3000_x3ewpzyr - failed to configure so won't report energy, power, current or voltage

curtash, 2024-06-24:
### What happened?

I two versions of the _TZ3000_x3ewpzyr which is a single channel relay with power monitor. The first version I purchased some time ago works perfectly but the new one won't fully configure so it won't show power monitoring. I spoke to the manufacture who stated that this is due to my second device using more up-to-date firmware that appears to not be compatible with zigbee2MQTT.

I've tried using the standard and edge versions of z2M, many different modules (in case it's a faulty module), dongle e 7.4.1.0, 7.4.2.0.

In the about section of the device its states

Zigbee Model: TS0001
Zigbee Manufacturer: _TZ3000_x3ewpzyr
Description: Switch with power monitoring
Support status: Supported
Firmware build date: 20210625
Firmware version: 500+TZTDQ31AC101
Manufacturer: [TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model: [TS0001_power](https://www.zigbee2mqtt.io/devices/TS0001_power.html#tuya-ts0001_power)

### What did you expect to happen?

I expected it to report Power, Current, Voltage & Energy. I can turn the relay on and off, adjust the switch type and power outage memory. I can even adjust "Settings  (specific)" options such as power, current and voltage calibration

### How to reproduce it (minimal and precise)

Just pair

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.4.1 [GA]

### Adapter

sonoff dongle e (using ember)

### Setup

HA raspberry PI 4B 

### Debug log

Error 2024-06-24 07:42:02Failed to configure '0x70b3d52b600f0b1c', attempt 2 (Error: ZCL command 0x70b3d52b600f0b1c/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:316:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:771:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:557:9) at Object.rmsVoltage (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:224:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2967:13) at Configure.configure (/app/lib/extension/configure.ts:121:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Error 2024-06-24 12:52:32z2m: Failed to configure '0x70b3d52b600bf199', attempt 2 (Error: ZCL command 0x70b3d52b600bf199/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:316:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:771:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:557:9) at Object.rmsVoltage (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:224:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3011:13) at Configure.configure (/app/lib/extension/configure.ts:121:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

curtash, 2024-06-27:
Unfortunately that does not seems to have fixed it. I even removed and repaired it. This time though no errors but voltage, current, energy and power remain at Null.

I'm quite sure I added the converter correctly but here are a couple of images to make sure (I restarted after adding) :

[image]

[image]

Koenkk, 2024-06-27:
When configuring the device via the z2m frontend -> yellow refresh button, does that succeed now?

curtash, 2024-06-27:
Unfortunately not no but the error is there. See below:

`Info 2024-06-27 10:59:54z2m: Configuring '0x70b3d52b600bf199'
Error 2024-06-27 10:59:55z2m: Failed to configure '0x70b3d52b600bf199', attempt 4 (Error: ZCL command 0x70b3d52b600bf199/1 haElectricalMeasurement.configReport([{"attribute":"rmsCurrent","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":50}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:316:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:771:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:557:9) at Object.rmsCurrent (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:220:5) at Object.configure (/app/data/extension/externally-loaded.js:27:9) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Info 2024-06-27 10:59:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x70b3d52b600bf199"},"error":"Failed to configure (ZCL command 0x70b3d52b600bf199/1 haElectricalMeasurement.configReport([{\"attribute\":\"rmsCurrent\",\"minimumReportInterval\":5,\"maximumReportInterval\":3600,\"reportableChange\":50}], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE'))","status":"error","transaction":"bhjke-1"}'`

Koenkk, 2024-06-27:
Updated https://gist.github.com/Koenkk/2f54ba89ade7460ed3941dbba81eec48, could you try again?

curtash, 2024-06-27:
Did not work

```
Info 2024-06-27 11:22:49z2m: Configuring '0x70b3d52b600bf199'
Error 2024-06-27 11:22:49z2m: Failed to configure '0x70b3d52b600bf199', attempt 4 (Error: ZCL command 0x70b3d52b600bf199/1 haElectricalMeasurement.configReport([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":10}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:316:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:771:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:557:9) at Object.activePower (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:208:5) at Object.configure (/app/data/extension/externally-loaded.js:28:9) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Info 2024-06-27 11:22:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x70b3d52b600bf199"},"error":"Failed to configure (ZCL command 0x70b3d52b600bf199/1 haElectricalMeasurement.configReport([{\"attribute\":\"activePower\",\"minimumReportInterval\":5,\"maximumReportInterval\":3600,\"reportableChange\":10}], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE'))","status":"error","transaction":"j4sol-1"}'
```

Koenkk, 2024-06-28:
So for power, voltage and current you got a  `UNREPORTABLE_ATTRIBUTE` error from the device, I'm starting to wonder wether yours support energy measurements. I've updated https://gist.github.com/Koenkk/2f54ba89ade7460ed3941dbba81eec48 to configure it for energy only.

curtash, 2024-06-29:
I have tried and still no joy (not even energy only). I suspect you are correct, annoyingly the earlier version of the device worked perfectly. I think the new firmware that they have added has caused problems. 

The device I'm using by the way is a 1CH version of this https://www.zigbee2mqtt.io/devices/XSH01B.html which is known as XSH01A.

Not sure if this helps but I added one to the Tuya App and it is reporting as expected. 

I'm in contact with the manufacturer of the device so if you have any information that may be helpful to them then please let me know and I'll pass it on.

I feel the manufacturer has broken their device with the firmware they have added. Unrelated to this issue  I know but these new modules don't come on when powered about 30% - 40% of the time, you have to remove and restore power again to have them come back alive.

```
Info 2024-06-29 08:59:18z2m: Configuring '0x70b3d52b600bf199'
Error 2024-06-29 08:59:23z2m: Failed to configure '0x70b3d52b600bf199', attempt 3 (Error: ZCL command 0x70b3d52b600bf199/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":154}) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:96:31) at EmberAdapter.onMessageSentDeliveryFailed (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:610:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:3960:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:797:18) at Ezsp.tick (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:451:22) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))
Info 2024-06-29 08:59:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x70b3d52b600bf199"},"error":"Failed to configure (ZCL command 0x70b3d52b600bf199/1 genBasic.read([\"manufacturerName\",\"zclVersion\",\"appVersion\",\"modelId\",\"powerSource\",65534], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Delivery failed for {\"profileId\":260,\"clusterId\":0,\"sourceEndpoint\":1,\"destinationEndpoint\":1,\"options\":4160,\"groupId\":0,\"sequence\":154}))","status":"error","transaction":"xqg6e-1"}'
```

Koenkk, 2024-06-29:
> Not sure if this helps but I added one to the Tuya App and it is reporting as expected.

Maybe it does polling here, do you get energy values when doing a read like this?

[image]

rmghhh1, 2024-06-29:
It looks like that is the case as I received a power value when I plugged a heater into the device and did as you suggested.

`Info 2024-06-29 17:13:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600bf199', payload '{"current":null,"energy":null,"linkquality":255,"power":432,"power_outage_memory":"restore","state":"ON","switch_type":"state","voltage":234}'`

curtash, 2024-06-29:
It looks like that is the case as I received a power value when I plugged a heater into the device and did as you suggested.

`Info 2024-06-29 17:13:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600bf199', payload '{"current":null,"energy":null,"linkquality":255,"power":432,"power_outage_memory":"restore","state":"ON","switch_type":"state","voltage":234}'`

Koenkk, 2024-06-30:
Does it work with the following ext converter? https://gist.github.com/Koenkk/2f54ba89ade7460ed3941dbba81eec48

curtash, 2024-07-01:
HI, I can confirm that all values are now updating.

```
Info 2024-07-01 07:54:42z2m: Successfully configured '0x70b3d52b600bf199'
Info 2024-07-01 07:54:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x70b3d52b600bf199"},"status":"ok","transaction":"tifds-3"}'
Info 2024-07-01 07:54:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b600bf199', payload '{"current":1.91,"energy":0.03,"linkquality":255,"power":463,"power_outage_memory":"restore","state":"ON","switch_type":"state","voltage":242}'
```

Koenkk, 2024-07-01:
Great, integrated the fix, you can remove the ext converter after updating to the 1 August release.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-01 by Koenkk

mrmaximas, 2025-01-05:
Dear @Koenkk 
As has been said before me _TZ3000_x3ewpzyr is the TUYA Model: XSH01A aka 1-gang variant of https://www.zigbee2mqtt.io/devices/XSH01B.html
I have one of this device, it shows up as a switch, not a relay.
[image]

Right device picture 512*512 with transparent background here:

[image]

Please fix it. Thank you!

Koenkk, 2025-01-05:
Fixed!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `XSH01A.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

mrmaximas, 2025-01-05:
> Fixed!
>
> Could you also submit a picture for the docs? This can be done by:
>
> 1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
> 2. Go to the `public/images/devices` [...]

fixed

Basskongen, 2025-03-06:
I now have this exact problem.

Zigbee Model: TS0001
Zigbee Manufacturer: _TZ3000_xkap8wtb
Description: Switch with power monitoring
Support status: Supported
Firmware build date: 20210625
Firmware version: 500+TZSKT31Z3101
Manufacturer: TuYa
Model: TS0001_power

**Zigbee2MQTT version**

[2.1.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.3) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

**Coordinator type**
zStack3x0

**Adapter**
Sonoff ZBDongle P

**Error**

(z2m: Failed to configure '0x70b3d52b60135065', attempt 8 (Error: ZCL command 0x70b3d52b60135065/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9) at Object.rmsVoltage (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:219:5) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3902:13) at Configure.configure (/app/lib/extension/configure.ts:124:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:49:25) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17)))

Koenkk, 2025-05-27:
@Cyanide05 could you post the data/database.db entry of your device?

Koenkk, 2025-05-28:
For your device it has been fixed in the dev branch already.

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Koenkk, 2025-11-30:
@curtash @mrmaximas could you both provide the data/database.db entry of this device?

note to self; https://github.com/Koenkk/zigbee-herdsman-converters/issues/4843

mrmaximas, 2025-11-30:
@Koenkk my model is https://www.zigbee2mqtt.io/devices/XSH01A.html#tuya-xsh01a
I believe that it should not be pulled.
`{"id":88,"type":"Router","ieeeAddr":"0xa4c138e2bc230f6c","nwkAddr":54240,"manufId":4417,"manufName":"_TZ3000_x3ewpzyr","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"65503":"�\u0003\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_x3ewpzyr","zclVersion":3,"appVersion":74,"modelId":"TS0001","powerSource":1,"hwVersion":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":232,"rmsCurrent":483,"activePower":95}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":33952}},"genOnOff":{"attributes":{"onOff":1,"moesStartUpOnOff":2,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":2}},"manuSpecificTuya3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1291,"minRepIntval":30,"maxRepIntval":3600,"repChange":10},{"cluster":2820,"attrId":1288,"minRepIntval":30,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1285,"minRepIntval":30,"maxRepIntval":3600,"repChange":5},{"cluster":1794,"attrId":0,"minRepIntval":300,"maxRepIntval":3600,"repChange":10}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1764510672045}`

Koenkk, 2025-12-07:
Didn't get a reply, so I've disabled the polling for your version @mrmaximas 

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

mrmaximas, 2025-12-07:
@Koenkk it's works 🙏🏻 Later, I will write about what to change in the default reporting settings.