Koenkk/zigbee2mqtt #23158: SNZB-02D doing OTA checks every 5mins

MFornander, 2024-06-25:
### What happened?

Looking at the logs I see that the otherwise awesome SONOFF SNZB-02D T&H w/ display, does an OTA check (genOta commandQueryNextImageRequest) every 5mins which will really drain the battery since the other updates are set to very infrequent.

Now I realize that this is probably not zigbee2mqtt's fault since it responds with a "NO_IMAGE_AVAILABLE" (genOta queryNextImageResponse) but I'm wondering if the reply payload doesn't satisfy the SNZB-02D so it keeps trying.

Is anyone else seeing this and/or working on a solution?  I'm about to order a SONOFF Zigbee Bridge Pro Hub and boot up my Zigbee sniffer to figure this out.  I'm wondering if there is a response or arcane set of commands that quiet the chatter if paired with a standard SONOFF controller.

Any ideas or workarounds?
Is everyone else's SNZB-02D reporting softwareBuildID: 1.0.9 and maybe I have an old firmware?

SNZB-02D device state below:
```
{
    "battery": 100,
    "humidity": 54.4,
    "last_seen": "2024-06-25T04:47:56.030Z",
    "linkquality": 240,
    "temperature": 22.6,
    "device": {
        "applicationVersion": 0,
        "dateCode": "20230105",
        "friendlyName": "SNZB-02D",
        "hardwareVersion": 0,
        "ieeeAddr": "0x0ceff6fffedf950f",
        "manufacturerID": 4742,
        "manufacturerName": "SONOFF",
        "model": "SNZB-02D",
        "networkAddress": 17356,
        "powerSource": "Battery",
        "softwareBuildID": "1.0.9",
        "type": "EndDevice",
        "zclVersion": 8
    }
}
```

### What did you expect to happen?

I expected that OTA checks would be at most every day.

### How to reproduce it (minimal and precise)

1. Install zigbee2mqtt
2. add SNZB-02D
3. Enable z2m debug log
4. See "z2m: Device 'SNZB-02D' requested OTA" every 5mins

### Zigbee2MQTT version

1.38.0-dev commit: 889e41ca

### Adapter firmware version

 7.4.1 [GA]

### Adapter

SLZB-06M v2.3.6 on ember 20240510 over TCP

### Setup

Plain linux x86-64 using manual npm start to debug

### Debug log

[2024-06-24 22:32:54] debug: 	zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=52 Len=36]
[2024-06-24 22:32:54] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":64,"groupId":0,"sequence":227}], [lastHopLqi=236], [lastHopRssi=-41], [sender=17356], [bindingIndex=255], [addressIndex=255], [messageContents=117101008612050809100000]
[2024-06-24 22:32:54] debug: 	zh:controller: Received payload: clusterID=25, address=17356, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=236, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":113,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4742,"imageType":2053,"fileVersion":4105},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
[2024-06-24 22:32:54] debug: 	z2m: Received Zigbee message from 'SNZB-02D', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":4105,"imageType":2053,"manufacturerCode":4742}' from endpoint 1 with groupID 0
[2024-06-24 22:32:54] info: 	z2m:mqtt: MQTT publish: topic 'z2m/SNZB-02D', payload '{"battery":100,"device":{"applicationVersion":0,"dateCode":"20230105","friendlyName":"SNZB-02D","hardwareVersion":0,"ieeeAddr":"0x0ceff6fffedf950f","manufacturerID":4742,"manufacturerName":"SONOFF","model":"SNZB-02D","networkAddress":17356,"powerSource":"Battery","softwareBuildID":"1.0.9","type":"EndDevice","zclVersion":8},"humidity":54.4,"last_seen":"2024-06-25T04:32:54.980Z","linkquality":236,"temperature":22.6}'
[2024-06-24 22:32:54] debug: 	z2m: Device 'SNZB-02D' requested OTA
[2024-06-24 22:32:54] debug: 	zh:controller:endpoint: CommandResponse 0x0ceff6fffedf950f/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-06-24 22:32:54] debug: 	zh:controller:endpoint: Request Queue (0x0ceff6fffedf950f/1): send request
[2024-06-24 22:32:54] debug: 	zh:ember:queue: Status queue=0 priorityQueue=0.
[2024-06-24 22:32:54] debug: 	zh:ember: ~~~> [ZCL to=17356 apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"manufacturerCode":null,"transactionSequenceNumber":113,"commandIdentifier":2}]
[2024-06-24 22:32:54] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=53 Len=25]
[2024-06-24 22:32:55] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=53 Len=7]
[2024-06-24 22:32:55] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=9 messageTag=7 status=SUCCESS]
[2024-06-24 22:32:55] debug: 	z2m: Responded to OTA request of 'SNZB-02D' with 'NO_IMAGE_AVAILABLE'

weljajoh, 2024-06-25:
I'm seeing the same issue on my SONOFF SNZB-02D also on firmware 1.0.9. I hadn't noticed but you issue made me check my logs.

MFornander, 2024-06-28:
Hijacking my own issue to let people know I'm working on a more complete z2m driver for SNZB-02D.

After some zniffing between it and a ZBBridge using eWeLink, I'm seeing attribute conversations that allows you to set high/low temp, high/low humidity, and temperature unit (using ZB not double click).  It also looks like the button in the back sends an on/off event which could be exposed, and I'm seeing a sun icon when it's bright so maybe it even has a light sensor with events.  Nice little $12 device if so!

I reached out to SONOFF support about the OTA issue and actually got an initial reply asking for more details.  It looks like the 1.0.9 version is the latest according to eWeLink so no firmware patch for it yet.

jonacaz, 2024-10-18:
I have exactly the same problem.  Items arrived in the last couple of days.  Firmware is the same as yours:
IEEE Address  0x44e2f8fffe17c5b6
Network address  0x327C
Firmware build date  20230105
Firmware version 1.0.9

closed 2024-12-28 by github-actions[bot]