Koenkk/zigbee2mqtt #23160: [New device support]:

larrriiin, 2024-06-25:
### Link

https://www.zigbee2mqtt.io/devices/GM46.html

### Database entry

{"id":17,"type":"EndDevice","ieeeAddr":"0x6c5cb1fffe9bd22b","nwkAddr":21059,"manufId":4098,"manufName":"_TZE200_gnw1rril","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":">\\\r.f>\\\r.\u0012?\\\r.g?\\\r.\u0012@\\\r.gI\\\r.g","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_gnw1rril","stackVersion":0,"dateCode":"","appVersion":65}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1719301521376}

### Comments

**The link to the device does not quite correspond to reality. I did not find a link specifically to my device (I bought it by hand on a local marketplace)
I have a GM46D model**

Hi!
This is the first time I've come across the fact that the device is not supported in z2m.
This is a curtain motor with a battery.
Model: GM4D-ZIGBEE.
It says on the box that it's from MOES.
Inside the ZS3L module.
I tried using the instructions and another external converter to add support, but without success. Nothing has changed after the restart - no new properties have been added.
I tried using the developer version of z2m (Zigbee2MQTT Edge), but it didn't help.

### External definition

```shell
// Название: Tuya Zigbee Smart Curtain
// Модель: TS0601_cover
// modelID: TS0601
// manufacturerName: _TZE200_gubdgai2

const fz = require( 'zigbee-herdsman-converters/converters/fromZigbee' );
const tz = require( 'zigbee-herdsman-converters/converters/toZigbee' );
const exposes = require( 'zigbee-herdsman-converters/lib/exposes' );
const reporting = require( 'zigbee-herdsman-converters/lib/reporting' );
const extend = require( 'zigbee-herdsman-converters/lib/extend' );
const tuya = require( 'zigbee-herdsman-converters/lib/tuya' );
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja', // Curtain motors
            'mcdj3aq', 'mcdj3aq\u0000', // Tubular motors
            ],
        fingerprint: [
            { modelID: 'TS0601', manufacturerName: '_TZE200_gnw1rril' },
            ],
        model: 'TS0601_cover',
        vendor: 'TuYa',
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Tuya', model: 'M515EGZT'},
            ],
        fromZigbee: [
            fz.tuya_cover,
            fz.ignore_basic_report,
            ],
        toZigbee: [
            tz.tuya_cover_control,
            tz.tuya_cover_options,
            ],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            ],
    };

module.exports = definition;
```

larrriiin, 2024-06-25:
Okay, it works, but there are few control units. All that works is to open/close/stop and change the direction of rotation.

I would like to be able to set the boundaries for opening / closing and, of course, control the position.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja', // Curtain motors
            'mcdj3aq', 'mcdj3aq\u0000', // Tubular motors
        ],
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_gnw1rril'},
    ],
    model: 'TS0601_cover_1',
    vendor: 'Tuya',
    description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
    whiteLabel: [
        tuya.whitelabel('Shenzhen Golden Security Technology', 'GM46D', 'Curtain motor', ['_TZE200_gnw1rril']),
    ],
    fromZigbee: [legacy.fromZigbee.tuya_cover, fz.ignore_basic_report],
    toZigbee: [legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options],
    exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            e.composite('options', 'options', ea.STATE_SET)
                .withFeature(e.numeric('motor_speed', ea.STATE_SET)
                    .withValueMin(0).withValueMax(255).withDescription('Motor speed'))
                .withFeature(e.binary('reverse_direction', ea.STATE_SET, true, false)
                    .withDescription('Reverse the motor direction'))],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(2), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(0)})],
            [2, 'position', tuya.valueConverter.coverPositionInverted],
            [3, 'position', tuya.valueConverter.coverPositionInverted],
            [4, 'opening_mode', tuya.valueConverterBasic.lookup({'tilt': tuya.enum(0), 'lift': tuya.enum(1)})],
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'learning': tuya.enum(2)})],
            [13, 'battery', tuya.valueConverter.raw],
            [101, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],
            [107, 'factory_reset', tuya.valueConverter.setLimit],
        ],
    },
};

module.exports = definition;
```

[image]

larrriiin, 2024-06-25:
Position is not working

my5123, 2024-07-05:
Facing a same problem.  
https://github.com/Koenkk/zigbee2mqtt/issues/23237

I am gone try all datapoint to find the right one some other time. Such as
[102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],
[103, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],
[104, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],
…………
[255, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],

larrriiin, 2024-07-06:
> Facing a same problem. #23237
>
> I am gone try all datapoint to find the right one some other time. Such as [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': [...]

I have solved the problem. He assembled the cornice and turned on the engine first to the far left, then to the far right, and he remembered them. Then the position definition started working.

Thank you.

closed 2024-07-06 by larrriiin