Koenkk/zigbee2mqtt #23161: [New device support]:  model: 'TS0601', vendor: '_TZE204_fwondbzy'

Destroyer061090, 2024-06-25:
### Link

https://www.aliexpress.us/item/1005004148507209.html?spm=a2g0o.order_list.order_list_main.17.25a23696CjBMer&gatewayAdapt=4itemAdapt

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0xa4c138e6873f79fd","nwkAddr":46683,"manufId":4417,"manufName":"_TZE204_fwondbzy","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_fwondbzy","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1719323586268}

### Comments

Just pared the device and Z2M say "not supported"

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_fwondbzy',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

XFNeo, 2024-07-19:
I'm really looking forward to support for this device.
Thanks!

tommaineult, 2024-07-19:
For information:
Vendor : MOES
Ref : ZSS-QY-HP-MS

Koenkk, 2024-07-26:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/7801

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-26 by Koenkk

TatoMCSE, 2024-07-29:
Does it also come with Edge version for HA?

Koenkk, 2024-07-30:
Yes, dev = edge

TatoMCSE, 2024-07-30:
Hmmm, should it work out of the box when switched to dev? Asking as it still shows as unsupported on zigbee2mqtt-edge v1.39.0-dev with herdsman converters v19.71.1 <crying>

Koenkk, 2024-07-31:
Yes, otherwise wait for tomorrows z2m release.

pehon, 2024-08-02:
I've updated zigbee2mqtt to 1.39.1 and the device works like a charm. Thank you!

XFNeo, 2024-08-02:
But unfortunately this sensor is also spamming in to zigbee network =(
[image]

Destroyer061090, 2024-08-04:
> But unfortunately this sensor is also spamming in to zigbee network =( [...]

What could be the side effect of this?

XFNeo, 2024-08-05:
@Destroyer061090 you can read this topic
https://github.com/Koenkk/zigbee2mqtt/issues/19045

arnevanhove, 2025-02-21:
Crap just ordered a ZSS-QY-HP from moes in the hopes it would not spam my zigbee network. Anybody knows if this device can be configured so it does not spam? for example, disabling the distance feature?