Koenkk/zigbee2mqtt #23164: Delivery of BROADCAST failed for "65532" [apsFrame={"profileId":0,"clusterId":54,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":116} messageTag=61]

msoaresp, 2024-06-25:
### What happened?

This error message constantly appears when I enter the Zigbee2MQTT interface, especially when I enable JOIN.
This happened after I migrated to the ember protocol. Furthermore, there is a strange behavior of the service and many error messages from it trying to read device states, as well as changing them.

### What did you expect to happen?

Just as it was before, no error message appeared. It's strange that red window in the inner right corner every now and then when I access the zigbee2MQTT interface

### How to reproduce it (minimal and precise)

I don't need to do anything. Without me doing anything, sometimes it appears. If I give a join, it appears. If I try to change the state of something on a device, sometimes it appears...

### Zigbee2MQTT version

 1.38.0 commit: unknown

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff Dongle-E

### Setup

Running on a machine with Home Assistant OS and x86-64

### Debug log

[2024-06-22 09:22:19] error: 	z2m: Failed to read state of '0xa4c1381d69243574' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":154},"zclSequence":237,"commandIdentifier":1} timed out after 10000ms))
[2024-06-22 09:27:35] error: 	z2m: Publish 'set' 'power_on_behavior' to 'Interruptor Churrasqueira' failed: 'Error: ZCL command 0xa4c1381d69243574/1 genOnOff.write({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":249},"zclSequence":250,"commandIdentifier":4} timed out after 10000ms)'
[2024-06-22 09:52:09] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":59} messageTag=255]
[2024-06-22 12:55:43] error: 	z2m: Failed to read state of 'Interruptor Churrasqueira' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":217},"zclSequence":23,"commandIdentifier":1} timed out after 10000ms))
[2024-06-22 13:31:38] error: 	z2m: Failed to read state of 'Interruptor Churrasqueira' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":4},"zclSequence":35,"commandIdentifier":1} timed out after 10000ms))
[2024-06-22 13:31:48] error: 	z2m: Failed to read state of 'Interruptor Churrasqueira' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":7},"zclSequence":36,"commandIdentifier":1} timed out after 10000ms))
[2024-06-22 13:31:58] error: 	z2m: Failed to read state of 'Interruptor Churrasqueira' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":8},"zclSequence":37,"commandIdentifier":1} timed out after 10000ms))
[2024-06-22 13:32:08] error: 	z2m: Failed to read state of 'Interruptor Churrasqueira' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":10},"zclSequence":38,"commandIdentifier":1} timed out after 10000ms))
[2024-06-22 13:32:28] error: 	z2m: Failed to read state of 'Interruptor Churrasqueira' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":15},"zclSequence":41,"commandIdentifier":1} timed out after 10000ms))
[2024-06-22 13:32:38] error: 	z2m: Failed to read state of 'Interruptor Churrasqueira' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":18},"zclSequence":42,"commandIdentifier":1} timed out after 10000ms))
[2024-06-22 13:32:48] error: 	z2m: Failed to read state of 'Interruptor Churrasqueira' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":20},"zclSequence":43,"commandIdentifier":1} timed out after 10000ms))
[2024-06-22 13:32:58] error: 	z2m: Failed to read state of 'Interruptor Churrasqueira' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":21},"zclSequence":44,"commandIdentifier":1} timed out after 10000ms))
[2024-06-22 13:33:08] error: 	z2m: Failed to read state of 'Interruptor Churrasqueira' after reconnect (ZCL command 0xa4c1381d69243574/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":57796,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":24},"zclSequence":45,"commandIdentifier":1} timed out after 10000ms))
[... cut ...]
[2024-06-25 19:20:25] error: 	zh:ember: Delivery of BROADCAST failed for "65532" [apsFrame={"profileId":0,"clusterId":54,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":116} messageTag=61]
[2024-06-25 19:33:08] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":133} messageTag=255]
[2024-06-25 19:33:59] error: 	z2m: Failed to check if update available for 'PIR Quarto Casal' (OTA: Device didn't respond to OTA request)

andyblac, 2024-06-26:
same, have to restart Z2M in order to allow a device to join.

msoaresp, 2024-06-26:
I thank. Does this mean I have to restart Z2M before clicking the JOIN button? Will this make this error stop? Is this a palliative or new behavior for Z2M?

andyblac, 2024-06-27:
> I thank. Does this mean I have to restart Z2M before clicking the JOIN button? Will this make this error stop? Is this a palliative or new behavior for Z2M?

yes, if you restart Z2M, then as soon as it come back online click `Join` it does allow you add new devices, but the error will come back after a while.

andyblac, 2024-06-27:
my log..

```
[2024-06-26 15:39:13] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":120} messageTag=255]
[2024-06-26 15:44:42] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":43} messageTag=255]
[2024-06-26 16:06:03] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":93} messageTag=255]
[2024-06-26 16:35:57] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":242} messageTag=255]
[2024-06-26 17:22:55] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":248} messageTag=255]
[2024-06-26 18:59:54] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":227} messageTag=255]
[2024-06-26 20:57:02] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":165} messageTag=255]
[2024-06-26 21:47:03] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":105} messageTag=255]
[2024-06-26 21:47:13] error: 	z2m: Publish 'set' 'state' to 'Lounge Patio Window Shade' failed: 'Error: ZCL command 0x54ef4410003a2750/1 genAnalogOutput.write({"presentValue":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":23595,"apsFrame":{"profileId":260,"clusterId":13,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":82},"zclSequence":145,"commandIdentifier":4} timed out after 10000ms)'
[2024-06-26 22:08:56] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":79} messageTag=255]
[2024-06-26 23:36:51] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":115} messageTag=255]
[2024-06-26 23:53:56] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":235} messageTag=255]
[2024-06-26 23:57:36] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":137} messageTag=255]
[2024-06-27 00:06:09] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":149} messageTag=255]
[2024-06-27 00:34:49] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":89} messageTag=255]
[2024-06-27 02:14:15] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":128} messageTag=255]
[2024-06-27 04:59:34] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":23} messageTag=255]
[2024-06-27 06:46:28] warning: 	z2m: Failed to ping 'Media Room Sideboard Lamp Left' (attempt 1/2, ZCL command 0x0017880103b85242/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":23677,"apsFrame":{"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":11,"options":4416,"groupId":0,"sequence":102},"zclSequence":125,"commandIdentifier":1} timed out after 10000ms))
[2024-06-27 07:14:22] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":215} messageTag=255]
[2024-06-27 08:58:41] error: 	z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0x54ef4410003a2875/1 genAnalogOutput.read(["presentValue"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":6754,"apsFrame":{"profileId":260,"clusterId":13,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":129},"zclSequence":184,"commandIdentifier":1} timed out after 10000ms)
```

smartmatic, 2024-07-22:
I have the same issue with a Sonoff Dongle-E and Firmware 7.4.3

`{"friendly_name":"0x0ceff6fffe17f579","ieee_address":"0x0ceff6fffe17f579"},"type":"device_announce"}'
[2024-07-22 08:56:22] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":59} messageTag=255]
[2024-07-22 08:56:25] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":67} messageTag=255]
[2024-07-22 08:58:44] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":72} messageTag=255]
[2024-07-22 09:00:59] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":73} messageTag=255]`

mvadu, 2024-07-24:
same here with a ZB-GW03 with 7.4.3 firmware (from https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/ZigbeeBridge_SM-011-signed) 
```
[2024-07-24 18:30:10] warning: 	zh:ember: [EzspConfigId] Failed to SET "ADDRESS_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-24 18:30:12] warning: 	zh:ember: [EzspConfigId] Failed to SET "SOURCE_ROUTE_TABLE_SIZE" TO "200" with status=ERROR_INVALID_VALUE. Firmware value will be used instead.
[2024-07-24 18:30:12] warning: 	zh:ember: [EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-24 18:30:16] error: 	zh:ember: Delivery of BROADCAST failed for "65532" [apsFrame={"profileId":0,"clusterId":54,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":183} messageTag=1]
[2024-07-24 18:31:05] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":160} messageTag=255]
[2024-07-24 18:31:08] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":161} messageTag=255]
[2024-07-24 18:31:11] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":163} messageTag=255]
[2024-07-24 18:31:13] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":131} messageTag=255]
```

larstobi, 2024-08-08:
Same here with SkyConnect adapter:
```
[2024-08-07 21:23:33] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":61} messageTag=255]
[2024-08-07 21:25:43] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":52} messageTag=255]
[2024-08-07 22:02:10] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":51} messageTag=255]
[2024-08-07 22:03:25] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":40} messageTag=255]
[2024-08-07 22:18:33] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":251} messageTag=255]
[2024-08-07 22:39:26] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":251} messageTag=255]
[2024-08-07 22:53:04] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":76} messageTag=255]
[2024-08-07 23:22:11] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":85} messageTag=255]
[2024-08-07 23:48:34] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":4} messageTag=255]
[2024-08-08 00:10:37] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":120} messageTag=255]
[2024-08-08 00:34:24] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":254} messageTag=255]
[2024-08-08 02:00:11] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":183} messageTag=255]
[2024-08-08 02:00:26] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":188} messageTag=255]
[2024-08-08 02:00:40] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":193} messageTag=255]
[2024-08-08 02:27:43] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":59} messageTag=255]
[2024-08-08 02:28:18] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":52} messageTag=255]
[2024-08-08 02:28:43] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":244} messageTag=255]
[2024-08-08 02:38:37] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":105} messageTag=255]
[2024-08-08 02:46:36] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":234} messageTag=255]
[2024-08-08 03:12:18] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":255} messageTag=255]
[2024-08-08 03:12:39] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":101} messageTag=255]
[2024-08-08 04:10:37] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":132} messageTag=255]
[2024-08-08 04:35:10] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":221} messageTag=255]
[2024-08-08 06:32:23] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":13} messageTag=255]
[... cut ...]
[2024-08-08 06:37:13] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":183} messageTag=255]
[2024-08-08 06:37:22] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":113} messageTag=255]
[2024-08-08 06:37:27] error: 	zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":188} messageTag=255]
```

closed 2024-12-28 by github-actions[bot]

JericoPablo, 2025-01-23:
i already commented on a closed one but i also have this. [issue 22453](https://github.com/Koenkk/zigbee2mqtt/issues/22453#issuecomment-2599330604) @Koenkk anything we could provide more to solve this? Because im sure its not a inteference issue.