Koenkk/zigbee2mqtt #23169: Firmware version and build date are wrong for IKEA Tradfri Open/Close Remote (E1766)

konqi, 2024-06-26:
### What happened?

I wrote my whole story [here](https://github.com/Koenkk/zigbee-OTA/issues/516) but I'll summarize:
- I have a IKEA Tradfri Open/Close Remote 
- According to the `fileVersion` the device is up to date (see logs)
- the ui shows the device  as version 2.2.010 and an old build date (20190311)
- the Battery says it is at 200% (because [that fix](https://github.com/Koenkk/zigbee-herdsman-converters/blob/3e56ca454b942b4780f08976e81de93982f89606/src/lib/ikea.ts#L114) needs the correct `softwareBuildID`
- I believe this issue is related to Issue #9439. I had similar symptoms as described in that issue and the bind-during-discovery-fix worked (something seems off with the device though, e.g. I can't properly bind it to the blinds). If the `softwareBuildID` is incorrect, conditional fixes for versions aren't properly applied (the ones where binding to only groups and binding to only devices works).

I tried to figure out where the `softwareBuildID` is set (or where it is coming from). I assume it should be in the herdsman-converters but I wasn't able to locate the piece of code.

### What did you expect to happen?

The UI should show the correct version (`24.4.6`). Since I wasn't able to find the code location where `softwareBuildID` is coming from I can't say much else. I somehow expect the code to create the version string from the `fileVersion`, since `604241926 = 0x0x24040006 = "24.4.6"` but don't know where this happens.

### How to reproduce it (minimal and precise)

No idea. But I should probably say that I have two of those buttons and only removed and re-paired one of them. So it could be some sort of invalid or corrupted cache?

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20230507

### Adapter

Electrolama zig-a-zig-ah! (zzh!)

### Setup

openhabian on Raspberry Pi

### Debug log

[2024-06-26 11:00:14] debug: 	zh:zstack:unpi:parser: <-- [254,34,68,129,0,0,25,0,203,5,1,13,0,98,0,10,49,222,0,0,14,1,80,1,1,124,17,197,17,6,0,4,36,1,0,51,158,28,203]
[2024-06-26 11:00:14] debug: 	zh:zstack:unpi:parser: --- parseNext [254,34,68,129,0,0,25,0,203,5,1,13,0,98,0,10,49,222,0,0,14,1,80,1,1,124,17,197,17,6,0,4,36,1,0,51,158,28,203]
[2024-06-26 11:00:14] debug: 	zh:zstack:unpi:parser: --> parsed 34 - 2 - 4 - 129 - [0,0,25,0,203,5,1,13,0,98,0,10,49,222,0,0,14,1,80,1,1,124,17,197,17,6,0,4,36,1,0,51,158,28] - 203
[2024-06-26 11:00:14] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":1483,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":98,"securityuse":0,"timestamp":14561546,"transseqnumber":0,"len":14,"data":{"type":"Buffer","data":[1,80,1,1,124,17,197,17,6,0,4,36,1,0]}}
[2024-06-26 11:00:14] debug: 	zh:controller: Received payload: clusterID=25, address=1483, groupID=0, endpoint=1, destinationEndpoint=13, wasBroadcast=false, linkQuality=98, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":80,"commandIdentifier":1},"payload":{"fieldControl":1,"manufacturerCode":4476,"imageType":4549,"fileVersion":604241926},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
[2024-06-26 11:00:14] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-26 11:00:14] debug: 	z2m: Received Zigbee message from 'TradfriRemoteOpenClose2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":604241926,"imageType":4549,"manufacturerCode":4476}' from endpoint 1 with groupID 0
[2024-06-26 11:00:14] debug: 	zhc:ota:common: Got request '{"fieldControl":1,"manufacturerCode":4476,"imageType":4549,"fileVersion":604241926}'
[2024-06-26 11:00:14] debug: 	zhc:ota:common: Is new image available for '0x842e14fffe783e62' (TRADFRI open/close remote), current '{"fieldControl":1,"manufacturerCode":4476,"imageType":4549,"fileVersion":604241926}'
[2024-06-26 11:00:14] debug: 	zhc:ota:tradfri: Call getImageMeta for TRADFRI open/close remote
[2024-06-26 11:00:14] debug: 	zhc:ota:common: Is new image available for '0x842e14fffe783e62' (TRADFRI open/close remote), latest meta '{"fileVersion":604241926,"url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/10005778-tradfri_onoff_controller-24.4.6-prod.ota.ota.signed"}'
[2024-06-26 11:00:14] debug: 	zhc:ota:common: Update available for '0x842e14fffe783e62' (TRADFRI open/close remote): NO
[2024-06-26 11:00:14] info: 	z2m: No update available for 'TradfriRemoteOpenClose2'
[2024-06-26 11:00:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/TradfriRemoteOpenClose2', payload '{"battery":200,"linkquality":98,"update":{"installed_version":604241926,"latest_version":604241926,"state":"idle"}}'
[2024-06-26 11:00:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"TradfriRemoteOpenClose2","updateAvailable":false},"status":"ok","transaction":"6onb5-5"}'

sjorge, 2024-06-26:
Try the reinterview button

konqi, 2024-06-27:
Just tried that. The first attempt failed (very likely because the device become unreachable), the second attempt went through without a problem (I pushed a button every couple of seconds so the device would stay awake).
Everything still looks the same.

```
[2024-06-27 08:47:32] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,5,0,97]
[2024-06-27 08:47:32] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 5 - [0] - 97
[2024-06-27 08:47:32] debug: 	zh:zstack:znp: SRSP: <-- ZDO - activeEpReq - {"status":0}
[2024-06-27 08:47:32] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-27 08:47:32] debug: 	zh:zstack:unpi:parser: <-- [254,23,68,129,0,0,2,1,203,5,1,1,0,91,0,246,124,110,0,0,3,1,101,0,51,158,28,118]
[2024-06-27 08:47:32] debug: 	zh:zstack:unpi:parser: --- parseNext [254,23,68,129,0,0,2,1,203,5,1,1,0,91,0,246,124,110,0,0,3,1,101,0,51,158,28,118]
[2024-06-27 08:47:32] debug: 	zh:zstack:unpi:parser: --> parsed 23 - 2 - 4 - 129 - [0,0,2,1,203,5,1,1,0,91,0,246,124,110,0,0,3,1,101,0,51,158,28] - 118
[2024-06-27 08:47:32] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":258,"srcaddr":1483,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":91,"securityuse":0,"timestamp":7240950,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[1,101,0]}}
[2024-06-27 08:47:32] debug: 	zh:controller: Received payload: clusterID=258, address=1483, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=91, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":101,"commandIdentifier":0},"payload":{},"command":{"ID":0,"parameters":[],"name":"upOpen"}}
[2024-06-27 08:47:32] debug: 	zh:controller:endpoint: ZCL command 0x842e14fffe783e62/1 closuresWindowCovering.defaultRsp({"cmdId":0,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":101,"writeUndiv":false})
[2024-06-27 08:47:32] debug: 	zh:controller:endpoint: Request Queue (0x842e14fffe783e62/1): send defaultRsp request immediately (sendPolicy=undefined)
[2024-06-27 08:47:32] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-27 08:47:32] debug: 	z2m: Received Zigbee message from 'TradfriRemoteOpenClose2', type 'commandUpOpen', cluster 'closuresWindowCovering', data '{}' from endpoint 1 with groupID 0
[2024-06-27 08:47:32] debug: 	z2m: Skipping message, still interviewing
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: <-- [254,7,69,133,203,5,0,203,5,1,1,199]
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,7,69,133,203,5,0,203,5,1,1,199]
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --> parsed 7 - 2 - 5 - 133 - [203,5,0,203,5,1,1] - 199
[2024-06-27 08:47:33] debug: 	zh:zstack:znp: AREQ: <-- ZDO - activeEpRsp - {"srcaddr":1483,"status":0,"nwkaddr":1483,"activeepcount":1,"activeeplist":[1]}
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-27 08:47:33] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x842e14fffe783e62:1483/1 (0,0,1)
[2024-06-27 08:47:33] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":1483,"destendpoint":1,"srcendpoint":1,"clusterid":258,"transid":26,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,101,11,0,0]}}
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,203,5,1,1,2,1,26,0,30,5,24,101,11,0,0,144]
[2024-06-27 08:47:33] debug: 	zh:controller:device: Interview - got active endpoints for device '0x842e14fffe783e62'
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-06-27 08:47:33] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,26,220]
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,26,220]
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,26] - 220
[2024-06-27 08:47:33] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":26}
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-27 08:47:33] debug: 	zh:zstack:znp: SREQ: --> ZDO - simpleDescReq - {"dstaddr":1483,"nwkaddrofinterest":1483,"endpoint":1}
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:writer: --> frame [254,5,37,4,203,5,203,5,1,37]
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,4,0,96]
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,4,0,96]
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 4 - [0] - 96
[2024-06-27 08:47:33] debug: 	zh:zstack:znp: SRSP: <-- ZDO - simpleDescReq - {"status":0}
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-27 08:47:33] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,4,11,230,184,1,1,0,32,0,58,36,111,0,0,8,24,196,10,11,5,41,86,0,230,184,29,53]
[... cut ...]
[2024-06-27 08:47:37] debug: 	z2m: Received Zigbee message from 'TradfriRemoteOpenClose2', type 'readResponse', cluster 'genPollCtrl', data '{"checkinInterval":172800}' from endpoint 1 with groupID 0
[2024-06-27 08:47:37] debug: 	z2m: Skipping message, still interviewing
[2024-06-27 08:47:37] debug: 	zh:controller:device: Interview - completed for device '0x842e14fffe783e62'
[2024-06-27 08:47:37] debug: 	zh:controller:database: Writing database to '/opt/zigbee2mqtt/data/database.db'
```
I believe this is only a partial interview but the relevant stuff should be in there.

konqi, 2024-06-28:
This log is from the beginning of the interview
```
[2024-06-28 14:53:44] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/interview' with data '{"id":"TradfriRemoteOpenClose2","transaction":"t7k0z-3"}'
[2024-06-28 14:53:44] debug: 	zh:controller:device: Interview - start device '0x842e14fffe783e62'
[2024-06-28 14:53:44] debug: 	zh:controller:device: Interview - skip node descriptor request for '0x842e14fffe783e62', already got it
[2024-06-28 14:53:44] debug: 	zh:zstack:znp: SREQ: --> ZDO - activeEpReq - {"dstaddr":1483,"nwkaddrofinterest":1483}
[2024-06-28 14:53:44] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,5,203,5,203,5,36]
[2024-06-28 14:53:44] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,5,0,97]
[2024-06-28 14:53:44] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,5,0,97]
[2024-06-28 14:53:44] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 5 - [0] - 97
[2024-06-28 14:53:44] debug: 	zh:zstack:znp: SRSP: <-- ZDO - activeEpReq - {"status":0}
[2024-06-28 14:53:44] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-28 14:53:45] debug: 	zh:zstack:unpi:parser: <-- [254,7,69,133,203,5,0,203,5,1,1,199]
[2024-06-28 14:53:45] debug: 	zh:zstack:unpi:parser: --- parseNext [254,7,69,133,203,5,0,203,5,1,1,199]
[2024-06-28 14:53:45] debug: 	zh:zstack:unpi:parser: --> parsed 7 - 2 - 5 - 133 - [203,5,0,203,5,1,1] - 199
[2024-06-28 14:53:45] debug: 	zh:zstack:znp: AREQ: <-- ZDO - activeEpRsp - {"srcaddr":1483,"status":0,"nwkaddr":1483,"activeepcount":1,"activeeplist":[1]}
[2024-06-28 14:53:45] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-28 14:53:45] debug: 	zh:controller:device: Interview - got active endpoints for device '0x842e14fffe783e62'
[2024-06-28 14:53:45] debug: 	zh:zstack:znp: SREQ: --> ZDO - simpleDescReq - {"dstaddr":1483,"nwkaddrofinterest":1483,"endpoint":1}
[2024-06-28 14:53:45] debug: 	zh:zstack:unpi:writer: --> frame [254,5,37,4,203,5,203,5,1,37]
[2024-06-28 14:53:45] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,4,0,96]
[2024-06-28 14:53:45] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,4,0,96]
[2024-06-28 14:53:45] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 4 - [0] - 96
[2024-06-28 14:53:45] debug: 	zh:zstack:znp: SRSP: <-- ZDO - simpleDescReq - {"status":0}
[2024-06-28 14:53:45] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-28 14:53:46] debug: 	zh:zstack:unpi:parser: <-- [254,42,69,132,203,5,0,203,5,36,1,4,1,3,2,1,7,0,0,1,0,3,0,9,0,32,0,0,16,124,252,7,3,0,4,0,6,0,8,0,25,0,2,1,0,16,115]
[2024-06-28 14:53:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,42,69,132,203,5,0,203,5,36,1,4,1,3,2,1,7,0,0,1,0,3,0,9,0,32,0,0,16,124,252,7,3,0,4,0,6,0,8,0,25,0,2,1,0,16,115]
[2024-06-28 14:53:46] debug: 	zh:zstack:unpi:parser: --> parsed 42 - 2 - 5 - 132 - [203,5,0,203,5,36,1,4,1,3,2,1,7,0,0,1,0,3,0,9,0,32,0,0,16,124,252,7,3,0,4,0,6,0,8,0,25,0,2,1,0,16] - 115
[2024-06-28 14:53:46] debug: 	zh:zstack:znp: AREQ: <-- ZDO - simpleDescRsp - {"srcaddr":1483,"status":0,"nwkaddr":1483,"len":36,"endpoint":1,"profileid":260,"deviceid":515,"deviceversion":1,"numinclusters":7,"inclusterlist":[0,1,3,9,32,4096,64636],"numoutclusters":7,"outclusterlist":[3,4,6,8,25,258,4096]}
[2024-06-28 14:53:46] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-28 14:53:46] debug: 	zh:controller:device: Interview - got simple descriptor for endpoint '1' device '0x842e14fffe783e62'
[2024-06-28 14:53:46] debug: 	zh:controller:device: Interview - Poll control - binding '0x842e14fffe783e62' endpoint '1'
[2024-06-28 14:53:46] debug: 	zh:controller:endpoint: Bind 0x842e14fffe783e62/1 genPollCtrl from '0x00124b0021cc0aea/1'
[2024-06-28 14:53:46] debug: 	zh:zstack:znp: SREQ: --> ZDO - bindReq - {"dstaddr":1483,"srcaddr":"0x842e14fffe783e62","srcendpoint":1,"clusterid":32,"dstaddrmode":3,"dstaddress":"0x00124b0021cc0aea","dstendpoint":1}
[2024-06-28 14:53:46] debug: 	zh:zstack:unpi:writer: --> frame [254,23,37,33,203,5,98,62,120,254,255,20,46,132,1,32,0,3,234,10,204,33,0,75,18,0,1,49]
[2024-06-28 14:53:46] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,33,0,69]
[2024-06-28 14:53:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,33,0,69]
[2024-06-28 14:53:46] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 33 - [0] - 69
[2024-06-28 14:53:46] debug: 	zh:zstack:znp: SRSP: <-- ZDO - bindReq - {"status":0}
[2024-06-28 14:53:46] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-28 14:53:47] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,203,5,1,51,158,230,254,3,69,161,203,5,0,41]
[2024-06-28 14:53:47] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,203,5,1,51,158,230,254,3,69,161,203,5,0,41]
[2024-06-28 14:53:47] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [203,5,1,51,158] - 230
[2024-06-28 14:53:47] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":1483,"relaycount":1,"relaylist":[40499]}
[2024-06-28 14:53:47] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,161,203,5,0,41]
[2024-06-28 14:53:47] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 161 - [203,5,0] - 41
[2024-06-28 14:53:47] debug: 	zh:zstack:znp: AREQ: <-- ZDO - bindRsp - {"srcaddr":1483,"status":0}
[2024-06-28 14:53:47] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-28 14:53:47] debug: 	zh:controller:endpoint: ZCL command 0x842e14fffe783e62/1 genPollCtrl.read(["checkinInterval"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"sr [... cut ...]
[2024-06-28 14:54:00] debug: 	z2m: Received Zigbee message from 'TradfriRemoteOpenClose2', type 'commandUpOpen', cluster 'closuresWindowCovering', data '{}' from endpoint 1 with groupID 0
[2024-06-28 14:54:00] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/TradfriRemoteOpenClose2', payload '{"action":"open","battery":200,"linkquality":105,"update":{"installed_version":604241926,"latest_version":604241926,"state":"idle"}}'
```

I can't find what you're looking for tough. 
Could this have to do with the workaround mentioned in #9439 ? The interview during paring is failing there when I don't bind the device to the coordinator during the interview.

sjorge, 2024-06-28:
It shouldn't matter if the device is bound or not. I don't see any read for those attributes. We could try and read them manually, z2m should still update the database if we do so.

You can do this with the dev tab on the frontend:
[image]

konqi, 2024-06-28:
Well – that is interesting.
[image]
[image]

Koenkk, 2024-06-29:
@sjorge we do update them (here: https://github.com/Koenkk/zigbee-herdsman/blob/99ee8f494d2c8d49040d8aa354772a3c16fa79c0/src/controller/controller.ts#L744) but they are not saved directly to the DB. Restarting Z2M will force a save, then it should appear correctly in the frontend.

sjorge, 2024-06-29:
Triggering a read and restarting z2m should miss the wrong version being displayed then. Based on the read, the device was already updated.