Koenkk/zigbee2mqtt #23172: [New device support]: Sleep Monitoring Pad SPM915

mario42004, 2024-06-26:
### Link

https://www.owon-smart.com/sleep-monitoring-pad-spm915-product/

### Database entry

{"id":12,"type":"EndDevice","ieeeAddr":"0x3c6a2cfffed3ea3a","nwkAddr":48421,"manufId":4412,"manufName":"OWON","powerSource":"Battery","modelId":"SPM915","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,1,32],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"SPM915","manufacturerName":"OWON","powerSource":3,"zclVersion":3,"appVersion":4,"stackVersion":0,"hwVersion":1,"dateCode":"20221019"}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffe71985f","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe71985f","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe71985f","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null}],"meta":{}}},"appVersion":4,"stackVersion":0,"hwVersion":1,"dateCode":"20221019","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1719427668763,"checkinInterval":3600}

### Comments

The converter is ready to be used with this sensor. Please could you include this in your next release. Thanks

### External definition

```shell
const {identify, iasZoneAlarm, battery} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SPM915'],
    model: 'SPM915',
    vendor: 'OWON',
    description: 'Owon sleeping pad monitor',
    extend: [identify(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]}), battery()],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-06-27:
Added, thanks!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `MODEL.png` (update `model` accordingly) 
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

freakshock88, 2024-07-01:
@mario42004 If I may ask do you know a shop where this device can be bought?

mario42004, 2024-07-02:
> Dear all, we have contacted the seller, and they have told us that so far, they do not have an online shop, but you can contact them directly to their mail. ***@***.******@***.***>. We have bought them few units in Spain without any problem.
>
> If you have another doubt, do not hesitate in contact us.
>
> I hope this information is useful for you.
>
> With my best regards,
>
> Mario

....An additional information, the commercial mail of the seller that have attend us is,
Name: Fini
 ***@***.******@***.***>

The experience was very nice with him,

Regards again,

Mario