Koenkk/zigbee2mqtt #23174: [New device support]: TS0601 _TZE204_hi5qczzt

nttrung1097, 2024-06-27:
### Link

https://vi.aliexpress.com/item/32396473743.html?gatewayAdapt=glo2vnm

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c13826989e09e8","nwkAddr":51660,"manufId":4417,"manufName":"_TZE204_hi5qczzt","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":56,"65508":0,"appVersion":73,"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1719455369020}

### Comments

I a newbie on this guide and i tried to follow the supporting new device but no library in the case, can i help me with this. Thanks a lot

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_hi5qczzt',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

ajimenezdev, 2024-08-23:
I have the same device and issue.

@nttrung1097 have you manage to configure it on z2mqtt?

nttrung1097, 2024-08-28:
> I have the same device and issue.
> 
> @nttrung1097 have you manage to configure it on z2mqtt?

not yet :(

closed 2024-12-28 by github-actions[bot]