Koenkk/zigbee2mqtt #23180: Sky Connect Interview Failed

BobD61, 2024-06-27:
### What happened?

Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'Sky Connect' (0x18d793fffed0f566) failed: Error: Interview failed because can not get node descriptor ('0x18d793fffed0f566')'
Info 2024-06-27 11:54:38MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{},"error":"interview of 'Sky Connect' (0x18d793fffed0f566) failed: Error: Interview failed because can not get node descriptor ('0x18d793fffed0f566')","status":"error","transaction":"bfcuq-3"}'

### What did you expect to happen?

Sky Device no longer being recognised although it still seems to be functional

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

7.4.2 (GA)

### Adapter

Sky Connect

### Setup

Addon in HA OS running on RPI 4

BobD61, 2024-12-21:
Still an issue

gvlekke, 2025-02-25:
maybe related? https://github.com/Koenkk/zigbee2mqtt/issues/25438

closed 2025-05-05 by github-actions[bot]

gvlekke, 2025-05-05:
I think this issue is bigger then just this device. Some help to debug this would be usefull