Koenkk/zigbee2mqtt #23183: [New device support]: TZE204_nbkshs6k

gu1234, 2024-06-27:
### Link

https://www.aliexpress.com/item/1005007033046059.html

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c138656f408bde","nwkAddr":24446,"manufId":4417,"manufName":"_TZE204_nbkshs6k","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"VG\u0010.i[G\u0010.i`G\u0010.ieG\u0010.i\u0006G\u0010.i\u000bG\u0010.i\u0010G\u0010.i\u0015G\u0010.i","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_nbkshs6k","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1719509726379}

### Comments

Looks like this is very similar device to other supported Tuya human presence devices but it's not supported

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_nbkshs6k',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

BazarnyiM, 2024-06-30:
Same issue. As I understand its need external converter?

himself1985, 2024-06-30:
I also have 2 of these devices.

That works here:
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_e5m9c5hl',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

That one doesn't work:
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_laokfqwu',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

Koenkk, 2024-06-30:
Can you try to: 
- stop z2m
- change `_TZE204_nbkshs6k ` to `_TZE200_ikvncluo` in the data/database.db
- start z2m, and check if the device works.

Hoops79, 2024-06-30:
> Can you try to:
> 
> * stop z2m
> * change `_TZE204_nbkshs6k ` to `_TZE200_ikvncluo` in the data/database.db
> * start z2m, and check if the device works.

I just gave that a go. No joy.

If it helps - previously I've managed to get what I assume is presence on datapoint 1. At least, an inverted value. Moving in front of the device changes the state to "false". Also, datapoint 12 appears to be lux. Shining a light changes the value, anyway.

I don't have a Tuya gateway so that's as far as I can go.

drathbone, 2024-07-01:
I have a few TZE204_nbkshs6k units as well, and same issue as above. I've tried a couple of convertor definitions with no joy, and tried the above (changing db to TZE200_ikvncluo), again no luck.

Subscribed to the issue and if I find anything that progresses it I'll post up.

drathbone, 2024-07-01:
> @drathbone try to change `TZE204_nbkshs6k` to `TZE200_nbkshs6k`

Thanks, tried that just now but no good.

Koenkk, 2024-07-01:
there are a few more you can try: `_TZE200_lu01t0zl`, `_TZE204_sxm7l9xa`, `_TZE200_hl0ss9oa`, `_TZE200_0u3bj3rc`, `_TZE200_2aaelwxk`, `_TZE200_2aaelwxk`, `_TZE204_sooucan5`, `_TZE204_ijxvkhd0`, `_TZE204_7gclukjs`, `_TZE204_e9ajs4ft, `_TZE204_sbyx0lm6`, `_TZE204_mhxn2jso`, `_TZE204_kyhbrfyl`.

An alternative is to find the correct [datapoints](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)

drathbone, 2024-07-01:
Many thanks @Koenkk - I've just tried all of those, and none of them seem to give returns for any of the exposed entities. 
I don't have a Tuya ZB GW, but I might pick one up as (I think) they're only cheap, and see if I can get anywhere. If not it's no big deal; I bought 5 of those sensors (thinking they were the same as one I tried from Amazon as the enclosure was the same - stupid I know :)), but they weren't expensive, and I'm sure support will eventually show up.

drathbone, 2024-07-02:
So I decided to grab a cheap used Tuya gateway and go down the dev route to find the datapoints.

1 = presence state (presence/No one)
9 = sensitivity (Low/Middle/High)
10 = keep time (30s/60s/120s)
12 = illuminance value (xxlux)

Still very new at this so following the docco breadcrumbs at what to do next :)

drathbone, 2024-07-02:
So far the following has got illuminance and presence working, just need to sort it so you can set the keep time and sensitivity:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_nbkshs6k',
        },
    ],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Testing Human Presence Detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.presence(),
        e.illuminance_lux(),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverterBasic.lookup({'occupied': 0, 'clear': 1})],
            [12, 'illuminance_lux', tuya.valueConverter.raw],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
`

drathbone, 2024-07-02:
The code below has this device fully working now:

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_nbkshs6k',
        },
    ],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Human Presence Detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.presence(),
        e.illuminance_lux(),
        e
            .enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high'])
            .withDescription('PIR sensor sensitivity (refresh and update only while active)'),
        e
            .enum('keep_time', ea.STATE_SET, ['30', '60', '120'])
            .withDescription('PIR keep time in seconds (refresh and update only while active)'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverterBasic.lookup({'occupied': 0, 'clear': 1})],
            [9, 'sensitivity', tuya.valueConverterBasic.lookup({low: tuya.enum(0), medium: tuya.enum(1), high: tuya.enum(2)})],
            [10, 'keep_time', tuya.valueConverterBasic.lookup({'30': tuya.enum(0), '60': tuya.enum(1), '120': tuya.enum(2)})],    
            [12, 'illuminance_lux', tuya.valueConverter.raw],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

I've timed the 30/60/120sec keep time as:

30sec = 90sec
60sec = 120sec
120sec = 166sec

This was timed from activating occupancy and then turning the device over until the occupancy cleared in device logs.
Not tested the ranges yet of low/medium/high, but I've run out of will to do it today :)
Also it will need an update to get a picture in there.

Hopefully this helps anyone still struggling to get this device running. ^^ The above may not be the tidiest I'm afraid, first time diving into this to get an unsupported device running, so go easy on me :)

drathbone, 2024-07-02:
[image]

drathbone, 2024-07-02:
[image]

drathbone, 2024-07-02:
Hmm, maybe not working - seems to be fine now in z2m but in hass the sensors aren't returning values:

[image]

I'm obviously missing something silly.

drathbone, 2024-07-02:
[image]

[image]

Hopefully someone can suggest what I've done wrong

drathbone, 2024-07-02:
Sorted by changing the presence line to True/False: 
```            [1, 'presence', tuya.valueConverterBasic.lookup({'True': 0, 'False': 1})],```

Hoops79, 2024-07-03:
Excellent @drathbone! That works for me, too. Thanks for your effort. I've found that the keep time on my sensor manages to clear closer to the 30 second time set. But it appears to vary quite a bit.

So, is this likely everything these things expose? No distance measurement is a shame. You get what you pay for, I guess.

drathbone, 2024-07-03:
Cheers. Yeah it was a new experience for me setting up the Tuya dev stuff, but that's all they expose. I have another bought from Amazon, looks identical so I guess lots use the same casing, but that one lets you choose detection distances etc, was hoping for the same but 🤷   I got 5 of these super cheap, and I thought they were a write-off, so at least there's a few more sensors to throw around 😂  I got a Tuya gateway used off Amazon for about £13, so I'll keep that in case I need to go datapoint hunting again, very useful.

I'm going to try and tidy it up a little, and maybe get the matching picture in there somehow.
Glad it's working for you too 👍

gu1234, 2024-07-03:
> Sorted by changing the presence line to True/False: ` [1, 'presence', tuya.valueConverterBasic.lookup({'True': 0, 'False': 1})],`
> 
> Low sensitivity seems to be around 4m detection range, perfect for small room. Lux measurement seems to be within 1-2units of other sensors I have, so pretty decent.
> 
> Not the best or worst sensor I've used, but at least the 5 I bought are now usable 😄

Thanks!! that's solved it for me

Koenkk, 2024-07-04:
@drathbone would you mind making a pull request by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

drathbone, 2024-07-05:
Hi @Koenkk - I will, I just need some free time to look over the tuya.ts file and work out exactly how the code fix/add needs to fit into it 😂  

Apologies, but js coding really isn't my forté. 
I'll find an hour or two over the weekend to look at it and get a PR raised 👍

drathbone, 2024-07-05:
^^ hopefully correct @Koenkk

drathbone, 2024-07-08:
As per PR, I've changed the Model line to this, which now shows the correct image:

```    model: 'ZY-M100-S_2',
```

Koenkk, 2024-07-09:
Merged, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-09 by Koenkk

fabian19vs, 2025-04-21:
hi! 
Is it possible to have more exposes for this device? The vendor says it is capable of detecting things like ''static or moving'', or even distance (which can be used to create zones), currently Z2M only shows presence (true or false) and iluminance (which is not working for me right now).