Koenkk/zigbee2mqtt #23187: [New device support]:  Zemismart Tuya Zigbee 4 Gang Wireless Scene Push Button Portable Switch App Control Battery Powered ZMR4

MiAutomations, 2024-06-28:
### Link

https://www.zemismart.com/products/zmr4

### Database entry

{"id":166,"type":"EndDevice","ieeeAddr":"0x881a14fffeb63b62","nwkAddr":52639,"manufId":4098,"manufName":"_TZ3000_11pg3ima","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_11pg3ima","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1719558560094}

### Comments

I tried to add this device to the Z2M without success. The device  appears in the lista as not supported 

Can you please include this device as a supported device? 

Thank you for your support in advance 

### External definition

```shell
const {battery, identify, onOff, commandsOnOff, commandsLevelCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS004F'],
    model: 'TS004F',
    vendor: '_TZ3000_11pg3ima',
    description: 'Automatically generated definition',
    extend: [battery(), identify(), onOff({"powerOnBehavior":false}), commandsOnOff(), commandsLevelCtrl()],
    meta: {},
};

module.exports = definition;
```

caduvalsesia, 2024-08-02:
The device is now shown as compatible, but none of the button actions work yet.

rodrigosiviero, 2024-08-31:
Same on my side, _TZ3000_11pg3ima say its unsupported.

Koenkk, 2024-09-02:
Could you provide the debug log when trigger actions on this device?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

rodrigosiviero, 2024-09-02:
> Could you provide the debug log when trigger actions on this device?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Here's me triggering all four buttons:

`Debug 2024-09-02 18:08:20Received Zigbee message from '0x28dba7fffe4aec22', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2024-09-02 18:08:20Skipping message, still interviewing
Debug 2024-09-02 18:08:21Received Zigbee message from '0x28dba7fffe4aec22', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug 2024-09-02 18:08:21Skipping message, still interviewing
Debug 2024-09-02 18:08:22Received Zigbee message from '0x28dba7fffe4aec22', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":51,"transtime":10}' from endpoint 1 with groupID 0
Debug 2024-09-02 18:08:22Skipping message, still interviewing
Debug 2024-09-02 18:08:22Received Zigbee message from '0x28dba7fffe4aec22', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":51,"transtime":10}' from endpoint 1 with groupID 0
Debug 2024-09-02 18:08:22Skipping message, still interviewing`

[image]

caduvalsesia, 2024-09-02:
This happens when I press the button1 1 time.

debug 2024-09-02 18:20:26zh:zstack:unpi:parser: <-- [254,23,68,129,0,0,6,0,176,34,1,1,0,98,0,220,217,120,0,0,3,1,52,1,176,34,11,247]
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: --- parseNext [254,23,68,129,0,0,6,0,176,34,1,1,0,98,0,220,217,120,0,0,3,1,52,1,176,34,11,247]
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: --> parsed 23 - 2 - 4 - 129 - [0,0,6,0,176,34,1,1,0,98,0,220,217,120,0,0,3,1,52,1,176,34,11] - 247
debug 2024-09-02 18:20:26zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":8880,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":98,"securityuse":0,"timestamp":7920092,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[1,52,1]}}
debug 2024-09-02 18:20:26zh:controller: Received payload: clusterID=6, address=8880, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=98, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":52,"commandIdentifier":1},"payload":{},"command":{"ID":1,"parameters":[],"name":"on"}}
debug 2024-09-02 18:20:26zh:controller:endpoint: ZCL command 0x28dba7fffe45580d/1 genOnOff.defaultRsp({"cmdId":1,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":52,"writeUndiv":false})
debug 2024-09-02 18:20:26zh:zstack: sendZclFrameToEndpointInternal 0x28dba7fffe45580d:8880/1 (0,0,1)
debug 2024-09-02 18:20:26zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":8880,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":7,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,52,11,1,0]}}
debug 2024-09-02 18:20:26zh:zstack:unpi:writer: --> frame [254,15,36,1,176,34,1,1,6,0,7,0,30,5,24,52,11,1,0,132]
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: --- parseNext []
debug 2024-09-02 18:20:26z2m: Received Zigbee message from '0x28dba7fffe45580d', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2024-09-02 18:20:26z2m: No converter available for 'ZMR4' with cluster 'genOnOff' and type 'commandOn' and data '{}'
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
debug 2024-09-02 18:20:26zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: --- parseNext []
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,7,193]
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,7,193]
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,7] - 193
debug 2024-09-02 18:20:26zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":7}
debug 2024-09-02 18:20:26zh:zstack:unpi:parser: --- parseNext []

This happens when I press the button3 1 time.
ebug 2024-09-02 18:22:42zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,8,0,176,34,1,1,0,98,0,103,99,10,0,0,7,1,57,2,0,51,10,0,176,34,11,181,254,27,68,129,0,0,1,0,176,34,1,1,0,98,0,117,100,10,0,0,7,8,58,10,33,0,32,200,176,34,29,77]
debug 2024-09-02 18:22:42zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,8,0,176,34,1,1,0,98,0,103,99,10,0,0,7,1,57,2,0,51,10,0,176,34,11,181,254,27,68,129,0,0,1,0,176,34,1,1,0,98,0,117,100,10,0,0,7,8,58,10,33,0,32,200,176,34,29,77]
debug 2024-09-02 18:22:42zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,8,0,176,34,1,1,0,98,0,103,99,10,0,0,7,1,57,2,0,51,10,0,176,34,11] - 181
debug 2024-09-02 18:22:42zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":8,"srcaddr":8880,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":98,"securityuse":0,"timestamp":680807,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[1,57,2,0,51,10,0]}}
debug 2024-09-02 18:22:42zh:controller: Received payload: clusterID=8, address=8880, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=98, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":57,"commandIdentifier":2},"payload":{"stepmode":0,"stepsize":51,"transtime":10},"command":{"ID":2,"parameters":[{"name":"stepmode","type":32},{"name":"stepsize","type":32},{"name":"transtime","type":33}],"name":"step"}}
debug 2024-09-02 18:22:42zh:controller:endpoint: ZCL command 0x28dba7fffe45580d/1 genLevelCtrl.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":57,"writeUndiv":false})
debug 2024-09-02 18:22:42zh:zstack: sendZclFrameToEndpointInternal 0x28dba7fffe45580d:8880/1 (0,0,1)
debug 2024-09-02 18:22:42zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":8880,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":24,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,57,11,2,0]}}
debug 2024-09-02 18:22:42zh:zstack:unpi:writer: --> frame [254,15,36,1,176,34,1,1,8,0,24,0,30,5,24,57,11,2,0,155]
[... cut ...]
debug 2024-09-02 18:22:42zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
debug 2024-09-02 18:22:42zh:zstack:unpi:parser: --- parseNext []
debug 2024-09-02 18:22:42zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,25,223]
debug 2024-09-02 18:22:42zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,25,223]
debug 2024-09-02 18:22:42zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,25] - 223
debug 2024-09-02 18:22:42zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":25}
debug 2024-09-02 18:22:42zh:zstack:unpi:parser: --- parseNext []

Koenkk, 2024-09-03:
- @rodrigosiviero the interview failed in your case, try re-pairing it
- @caduvalsesia could you provide a screenshot of the z2m frontend -> device about tab?

caduvalsesia, 2024-09-03:
Of course, here's the screenshot.
Thanks.
[image]

smartmatic, 2024-09-04:
> this

I've also created a debug log. Interview is finished but button press does yet work. Hope that helps?

[ZMR4 Debug.log](https://github.com/user-attachments/files/16871179/ZMR4.Debug.log)

Koenkk, 2024-09-05:
@Burki24 since you added the ZMR4 in https://github.com/Koenkk/zigbee-herdsman-converters/pull/7734, can you check wether it correctly sends the actions? It seems we need to use the standard definition instead (https://github.com/Koenkk/zigbee-herdsman-converters/blob/4a81f8cf5ea7e1562b1b337a1c4360ae378ea071/src/devices/tuya.ts#L3077)

@caduvalsesia can you check if the device works after:
- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZ3000_11pg3ima` to `_TZ3000_nuombroo`
- Start Z2M and see if the device now works

Koenkk, 2024-09-05:
@caduvalsesia could you provide the debug log when triggering it twice?

caduvalsesia, 2024-09-05:
The double press and hold options only worked with buttons 3 and 4.

Here I press button 3 twice.

debug 2024-09-05 14:01:11zh:zstack:unpi:parser: <-- [254,32,68,129,0,0,0,3,228,7,1,1,0,87,0,246,249,154,0,0,12,1,104,76,1,50,0,10,0,153,0,244,1,228,7,11,83]
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: --- parseNext [254,32,68,129,0,0,0,3,228,7,1,1,0,87,0,246,249,154,0,0,12,1,104,76,1,50,0,10,0,153,0,244,1,228,7,11,83]
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: --> parsed 32 - 2 - 4 - 129 - [0,0,0,3,228,7,1,1,0,87,0,246,249,154,0,0,12,1,104,76,1,50,0,10,0,153,0,244,1,228,7,11] - 83
debug 2024-09-05 14:01:11zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":768,"srcaddr":2020,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":87,"securityuse":0,"timestamp":10156534,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,104,76,1,50,0,10,0,153,0,244,1]}}
debug 2024-09-05 14:01:11zh:controller: Received payload: clusterID=768, address=2020, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=87, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":104,"commandIdentifier":76},"payload":{"stepmode":1,"stepsize":50,"transtime":10,"minimum":153,"maximum":500},"command":{"ID":76,"parameters":[{"name":"stepmode","type":32},{"name":"stepsize","type":33},{"name":"transtime","type":33},{"name":"minimum","type":33},{"name":"maximum","type":33}],"name":"stepColorTemp"}}
debug 2024-09-05 14:01:11zh:controller:endpoint: ZCL command 0x28dba7fffe45580d/1 lightingColorCtrl.defaultRsp({"cmdId":76,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":104,"writeUndiv":false})
debug 2024-09-05 14:01:11zh:zstack: sendZclFrameToEndpointInternal 0x28dba7fffe45580d:2020/1 (0,0,1)
debug 2024-09-05 14:01:11zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":2020,"destendpoint":1,"srcendpoint":1,"clusterid":768,"transid":157,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,104,11,76,0]}}
debug 2024-09-05 14:01:11zh:zstack:unpi:writer: --> frame [254,15,36,1,228,7,1,1,0,3,157,0,30,5,24,104,11,76,0,123]
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: --- parseNext []
debug 2024-09-05 14:01:11z2m: Received Zigbee message from 'int4_zemis', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":500,"minimum":153,"stepmode":1,"stepsize":50,"transtime":10}' from endpoint 1 with groupID 0
debug 2024-09-05 14:01:11z2m: No converter available for 'TS004F' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":500,"minimum":153,"stepmode":1,"stepsize":50,"transtime":10}'
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
debug 2024-09-05 14:01:11zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: --- parseNext []
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,157,91]
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,157,91]
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,157] - 91
debug 2024-09-05 14:01:11zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":157}
debug 2024-09-05 14:01:11zh:zstack:unpi:parser: --- parseNext []

Here I press button 4 twice. 
debug 2024-09-05 14:02:15zh:zstack:unpi:parser: <-- [254,32,68,129,0,0,0,3,228,7,1,1,0,80,0,42,247,215,0,0,12,1,105,76,3,50,0,10,0,153,0,244,1,228,7,11,200]
debug 2024-09-05 14:02:15zh:zstack:unpi:parser: --- parseNext [254,32,68,129,0,0,0,3,228,7,1,1,0,80,0,42,247,215,0,0,12,1,105,76,3,50,0,10,0,153,0,244,1,228,7,11,200]
debug 2024-09-05 14:02:15zh:zstack:unpi:parser: --> parsed 32 - 2 - 4 - 129 - [0,0,0,3,228,7,1,1,0,80,0,42,247,215,0,0,12,1,105,76,3,50,0,10,0,153,0,244,1,228,7,11] - 200
debug 2024-09-05 14:02:15zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":768,"srcaddr":2020,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":80,"securityuse":0,"timestamp":14153514,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,105,76,3,50,0,10,0,153,0,244,1]}}
debug 2024-09-05 14:02:15zh:controller: Received payload: clusterID=768, address=2020, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=80, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":105,"commandIdentifier":76},"payload":{"stepmode":3,"stepsize":50,"transtime":10,"minimum":153,"maximum":500},"command":{"ID":76,"parameters":[{"name":"stepmode","type":32},{"name":"stepsize","type":33},{"name":"transtime","type":33},{"name":"minimum","type":33},{"name":"maximum","type":33}],"name":"stepColorTemp"}}
debug 2024-09-05 14:02:15zh:controller:endpoint: ZCL command 0x28dba7fffe45580d/1 lightingColorCtrl.defaultRsp({"cmdId":76,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":105,"writeUndiv":false})
[... cut ...]
debug 2024-09-05 14:03:51zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
debug 2024-09-05 14:03:51zh:zstack:unpi:parser: --- parseNext []
debug 2024-09-05 14:03:51zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,167,97]
debug 2024-09-05 14:03:51zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,167,97]
debug 2024-09-05 14:03:51zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,167] - 97
debug 2024-09-05 14:03:51zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":167}
debug 2024-09-05 14:03:51zh:zstack:unpi:parser: --- parseNext []

Koenkk, 2024-09-05:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/812b59512556d54b7164440a9ec27649): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

caduvalsesia, 2024-09-05:
Koenkk,
Now it works perfectly.
I just had to choose the event option.
Excellent. Thank you very much and congratulations on your integration.
Today I'm releasing a video on how to configure zigbee2mqtt with sonoff dongle-e.
Success.

Koenkk, 2024-09-05:
@caduvalsesia what actions does it send with the event option?

caduvalsesia, 2024-09-05:
This sends:
1_single
2_single
3_single
4_single
1_double
2_double
3_double
4_double
1_hold
2_hold
3_hold
4_hold
[image]

Koenkk, 2024-09-05:
@caduvalsesia thanks!

@rodrigosiviero 

> empty string on action object:

thats expected: https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

Integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-05 by Koenkk

comdinet, 2024-11-05:
ZMR4, identified as model: TS0044, manufacturer _TZ3000_xwuveizv
I've added the external converter js and tried renaming in database.
Best it got was showing 6 total actions, completely random, not all buttons shown. Like 1_single action, 2_single action and then 4_double action.
Also wrong picture identified.