Koenkk/zigbee2mqtt #23191: TS0049 Variation (Sprinkler)

ian27ukish, 2024-06-28:
### Link

https://www.aliexpress.com/item/1005006859734151.html?spm=a2g0o.order_list.order_list_main.10.271b1802tAupaT

### Database entry

0xa4c138965701416c (0xC50A) 	_TZ3000_kz1anoi8	TS0049

### Comments

I've spent a SOLID 3 hours on this now and im at a complete loss. Could really use some help.

The device as added, does not work for on/off or show battery percentage.

If i "generate_external_defintion", and copy and paste that as an external converter.....the on/off works fine.

I've been trying an absolute bunch of different ways to use exposes and setting the meta datapoints, but all result in the device reverting to the default (not my external converter). I can't find any documentation or examples of using extends as in the generated code....,my search skills fail me.

I've mapped the data points for the various settings fine via Tuya cloud, i just need to get them configured as the example on https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html does not work for me.

The datapoints are:

> Battery
> 7
> 
> Switch
> 1
> 
> Accumulated Usage time
> 9
> 
> Irrigation time
> 11
> 
> Work state (is it on/off/idle etc)
> 12
> 
> Countdown (how long it runs for in seconds)
> 17
> 
> Once using time (seems to be time its ran for)
> 15
> 
> Failure to report
> 4
> 
> Weather delay
> 10

If anyone can help me with just one of them i'll happily do the rest and add the fingerprint + submit a PR with image etc for others. Just pulling my hair out.....first attempt is always the hardest 😃 

### External definition

```shell
const {identify, battery, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0049'],
    model: 'TS0049',
    vendor: '_TZ3000_kz1anoi8',
    description: 'Automatically generated definition',
    extend: [identify(), battery(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

ian27ukish, 2024-06-28:
Apologies but i cant figure out or find how to access the database via the homeassistant addon.....searched up and down for it without success.

This info is from the gui if it helps (not i left the description as is so i could tell when it was using the default vs my definition):

Friendly name
    Sprinkler
Description

Last seen
    N/A
Availability
    Online
Device type
    EndDevice
Zigbee Model
    TS0049
Zigbee Manufacturer
    _TZ3000_kz1anoi8
Description
    Automatically generated definition
Support status

    Supported
IEEE Address
    0xa4c138965701416c
Network address
    0xC50A
Manufacturer
    [_TZ3000_kz1anoi8](https://www.zigbee2mqtt.io/supported-devices/#v=_TZ3000_kz1anoi8)
Model
    [TS0049](https://www.zigbee2mqtt.io/devices/TS0049.html#_tz3000_kz1anoi8-ts0049)
Power
    %
Interview completed
    True

ian27ukish, 2024-07-02:
I've just used the original external definition to add the basic on/off and battery. I don't really need the extra nice to haves and can't figure it out.

const {identify, battery, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0049'],
    model: 'TS0049',
    vendor: '_TZ3000_kz1anoi8',
    description: 'Automatically generated definition',
    extend: [identify(), battery(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;

closed 2024-07-02 by ian27ukish

Koenkk, 2024-07-02:
Added.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `TS0049_1.png` (update `model` accordingly) 
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

Koenkk, 2024-07-02:
This device should be supported out-of-the-box in z2m 1.39.0

cyberdie, 2024-08-19:
I'm on z2m 1.39.1  and receiving the following error for this device:

`Error 2024-08-19 13:12:17z2m: Failed to configure 'Manguera', attempt 3 (Error: Bind 0xa4c1382f2b168d43/1 genPowerCfg from '0x00124b0018e207d6/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23) at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1102:40) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:843:28 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at ZStackAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:785:9) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:509:13) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:109:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:155:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:190:21) at Configure.configure (/app/lib/extension/configure.ts:126:13))`

And no data is exposed by the device. Any help on this?
[image]

Koenkk, 2024-08-20:
Try re-pairing the device

cyberdie, 2024-09-09:
Once repaired the device is not controllable through Z2m.
Exposes Battery and State but doesn't work.
The device, supossedly exposes more functions as seen here: https://es.aliexpress.com/item/1005007315283842.html