Koenkk/zigbee2mqtt #23198: SLZB-06M : Huge amount of `Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".` errors with the 7.4.x ember firmware

MilesTEG1, 2024-06-28:
### What happened?

I flashed the SLZB-06M with the latest official firmware, witch is a 7.4.1 version.
I tried after the SMLIGHT support ask me to flash with https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/slzb-06m/ncp-uart-hw-v7.4.3.0-slzb-06m-115200.gbl firmware.

But for the two versions, I've got a huge amount of this error:
`Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".`

It is so much that all the zigbee network is very slow.

And after some times, the Z2M frontend doesn't respond:
[image]
But it continues to log the errors... the log.log file is about 1Mo after severals minutes only...
I have to restart Z2M to get the fontend to work again.

Is there a way to have my SLZB-06M working fine?

Thanks in advance.

### What did you expect to happen?

That the error doesn't appear, and my coordinator works fine.

### How to reproduce it (minimal and precise)

Use a SLZB-06M, 
Flash it with an ember Firmware (SMLIGH Official 20240510
[image]

Or with this one : https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/slzb-06m/ncp-uart-hw-v7.4.3.0-slzb-06m-115200.gbl

See the log and see the errors.

### Zigbee2MQTT version

1.38.0-1

### Adapter firmware version

7.4.3 [GA]

### Adapter

EmberZNet (SLZB-06M)

### Setup

Add-on on Home Assistant OS, on a NUC with Intel CPU

### Debug log

<details><summary>LOG (mode Info)</summary>
<p>
```log
[2024-06-28 20:24:58] info: 	z2m: Logging to console, file (filename: log.log)
[2024-06-28 20:24:58] info: 	z2m: Starting Zigbee2MQTT version 1.38.0 (commit #unknown)
[2024-06-28 20:24:58] info: 	z2m: Starting zigbee-herdsman (0.49.2)
[2024-06-28 20:24:58] info: 	zh:ember: Using default stack config.
[2024-06-28 20:24:58] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-06-28 20:24:58] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-06-28 20:24:58] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-06-28 20:24:58] info: 	zh:ember:uart:ash: Socket ready
[2024-06-28 20:24:58] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-06-28 20:24:59] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-06-28 20:24:59] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-06-28 20:24:59] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-06-28 20:24:59] info: 	zh:ember: [STACK STATUS] Network up.
[2024-06-28 20:24:59] info: 	zh:ember: [INIT TC] NCP network matches config.
[2024-06-28 20:24:59] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-06-28 20:24:59] info: 	zh:ember:queue: Request dispatching started.
[... 101 lines cut ...]
[2024-06-28 20:25:00] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/1221051039810110150109113116116_0x048727fffe1c53c0/network_map/config', payload 'null'
[2024-06-28 20:25:00] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/1221051039810110150109113116116_0x048727fffe1c53c0/permit_join_timeout/config', payload 'null'
[2024-06-28 20:25:00] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x048727fffe1c53c0/restart_required/config', payload 'null'
[2024-06-28 20:25:00] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/select/1221051039810110150109113116116_0x048727fffe1c53c0/log_level/config', payload 'null'
[2024-06-28 20:25:00] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/switch/1221051039810110150109113116116_0x048727fffe1c53c0/permit_join/config', payload 'null'
[2024-06-28 20:25:00] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/button/1221051039810110150109113116116_0x048727fffe1c53c0/restart/config', payload 'null'
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
[2024-06-28 20:25:00] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
```
</p>
</details> 

LOG (mode Debug)
[log.log](https://github.com/user-attachments/files/16035637/log.log)

Koenkk, 2024-06-29:
If you go to the z2m frontend, are there any devices where the network address (ieee address column) is 0x0000?

MilesTEG1, 2024-06-29:
> If you go to the z2m frontend, are there any devices where the network address (ieee address column) is 0x0000?

No `0x0000` ieee address in the frontend.

Koenkk, 2024-06-29:
@Nerivec have you seen this earlier? I suspect that there is some malfunctioning device on your network.

Nerivec, 2024-06-29:
Yes, https://github.com/Koenkk/zigbee2mqtt/issues/22682 https://github.com/Koenkk/zigbee2mqtt/issues/22271
It's not `ember` specific though, it's just hidden in `debug` for `ezsp`.

Clearing the NVM3 and/or re-creating the Z2M network appears to fix it for most.
Seems something goes wrong during a reboot...

MilesTEG1, 2024-06-29:
So what are you suggestion do ?

What is cleaning NVM3 ? How do I do it ?

MilesTEG1, 2024-07-03:
Hello,
Some news here.

Ive been in contact with SMLight support, and the thought that it was a Zigbee Network problem, and they told me to recreate it or to do another manipulation with https://github.com/tube0013/tubeszb_addons/blob/main/tzb-zigpy-cli-tools/DOCS.md .
I choose to recreate the Zigbee Network.

I must tell that my HAOS is a Proxmox VM. The SkyConnect was connected to the proxmox's host and passthrough to the HAOS VM.
Since I installed the SLZBV-06M, I disconnected the SkyConnect USB key, and remove its entries to the VM Config inside Proxmox.

To achieve the zigbee network reset, I do those changes:

1. Stop Z2M;
2. Backup the configuration files including the coordinator backup and;
3. remove the coordinator backup;
4. Change in the Z2M `configuration.yaml` the `network_key` or `pan_id` or `ext_pan_id` to « `GENERATE` » in order to let Z2M create those values;
5. Change the Zigbee channel from 11 to 15;
6. Set driver to ember.
7. Report some parameters inside the Z2M WebUI because of not my modifications in the Z2M `configuration.yaml` they seemed to be not applied…
8. start again Z2M;
9. and see the log.
10. Re-pair all my devices, starting by the routers, my smart plugs, finishing with end devices.
11. And… tadaaam ! No more errors ROUTE_ERROR_ADDRESS_CONFLICT.

It seems that there were a conflict somewhere with my old network from the SkyConnect…

And by chance, most of my devices were re-added without losing configuration, but for some I had to force delete from Z2M before re-pairing them.

So for now, it’s good!

I still have a startup problem: after a HAOS reboot, Z2M crashes dans frontend won't launch.
I have to reboot Z2M to get it back to work.
But I'll use another issue for that, an existent issue, I think.

Nerivec, 2024-07-04:
Sorry, missed your previous message.
For future reference, there's a post [here](https://github.com/darkxst/silabs-firmware-builder/issues/84#issuecomment-2094078288) that describes how to reset the NVM3 with bellows.
[ember-zli](https://github.com/Nerivec/ember-zli) also supports it (still new 😉).

Glad you could find a way to get this back on track. Thanks for the feedback!
I'll let you close this one, as soon as you consider the issue definitely solved.

closed 2024-07-21 by Nerivec

MilesTEG1, 2025-04-17:
Hello @Nerivec 
I flashed my ZLinky_TIC with limited firmware, and since then, I just get those errors back... `Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".`

I'd like to get rid of those errors, but I don't want to do a full repairing of my devices... I've got 71 !
And worse, some are in my wall (inside a wall mounting box) behind a swtich (for my covers), or behind an heavy heating tower dryer.

Is it possible to to so with reseting the NVM3?
I check the two links you gave me before, but on my mac (macOS) i'm not sure I can do what it's needed, and worse, how can I do it...

Nerivec, 2025-04-17:
If that ZLinky_TIC is the cause of the errors, unfortunately, it may not be fixable without a firmware fix on that device (if that device keeps reporting with address 0, which is not proper). If you have confirmed they come from that device, the only recourse until then is to take it out of the network.

---

Resetting the NVM3 will reset your adapter to factory defaults. Now, with a `coordinator_backup.json`, you should be able to get the network back on track after that, but **make sure you have one**, and be aware it may create some friction in the first few hours (until the network settles back down).

I haven't gotten much feedback from Mac use with ember-zli, but you can try one of the release tarballs (darwin arm64 or x64 depending on your machine):
https://github.com/Nerivec/ember-zli/wiki#from-release-tarballs
From there, you can go into the bootloader and [clear the nvm3](https://github.com/Nerivec/ember-zli/wiki/Bootloader#clear-nvm3):
- `ember-zli bootloader`
- follow prompt questions
- "Clear NVM3"
- follow prompt questions, answer `32768` when it asks for the `NVM3 Size`.

MilesTEG1, 2025-04-17:
> If that ZLinky_TIC is the cause of the errors, unfortunately, it may not be fixable without a firmware fix on that device (if that device keeps reporting with address 0, which is not proper). If you [...]
I indeed tried to remove it from Z2M, but unfortunately still having those errors...
I also tried to flash another zigbee FW on my SLZB-06M (the one from SMlight , latest version 8.0.2), but no changes.

I did another thing : unplug from my electricity provider's box (Linky) for more than 30s. And replug it and then try to re-pair it...

But doing so, I kept getting an error saying `Error: Interview failed because can't get node descriptor ('0x00158d0005d25e9d')` and a red warning on the device in Z2M...

Sox I force-deleted the ZLinky, then re-paired it after being disconnected for more than 30 seconds, but I still got this error...
I had to choose the coordinator instead of all the others for pairing to finally get a stable Zigbee network without these `ROUTE_ERROR_ADDRESS_CONFLICT` errors.

I was afraid I'il have to re-pair all my devices 😅

> ---
>
> Resetting the NVM3 will reset your adapter to factory defaults. Now, with a `coordinator_backup.json`, you should be able to get the network back on track after that, but **make sure you have [...]

Thanks for the method! 
I've got a MacBook Air M2.

For my information, I suppose I need to plug with usb the SLZB-06M to my Mac in order to use those commands?

Thanks for the quick reply 😇

Nerivec, 2025-04-17:
It's a little strange this behavior. Not sure what's going on behind the scene there... 👀

---

You can use USB or TCP (just point to it like you have in your `configuration.yaml` for Z2M). _It will try to discover, but depending on your network, that may or may not work._

Nerivec, 2025-04-18:
To get an address conflict on zero, technically you'd need a device that reported as "coordinator" (the only device supposed to have address zero in a ZigBee network). _Granted, there could be a bug in the status or the network address reported..._

You can usually force a self-heal (routing) of the network by powering it down for a while (10-15min), then power devices back up, and then power your coordinator and start Z2M... Bit annoying I know, but last resort... _Expect a few errors after that while it heals._

MilesTEG1, 2025-04-18:
> To get an address conflict on zero, technically you'd need a device that reported as "coordinator" (the only device supposed to have address zero in a ZigBee network). _Granted, there could be a bug [...]

That's a drastic thing to do 😆
I'll keep it in mind for the next time I got those errors.
For now, everything is alright.

jehy, 2025-05-07:
Okay, so it was my turn to encounter `ROUTE_ERROR_ADDRESS_CONFLICT`.  
Looks like I discovered something new!  

### How it happened:  
- Switched adapter hardware to **Sonoff v2**  
- Changed protocol from **EZSP** to **Ember**  
- Kept old network settings to avoid re-pairing devices  
- Result: Immediate spam of `ROUTE_ERROR_ADDRESS_CONFLICT` errors (5-7 at a time, every few seconds)  

### What didn't work:  
- Updating stick firmware  
- Upgrading Zigbee2MQTT (z2m)  
- Rebooting router devices and z2m  

### The fix:  
Noticed **one misbehaving device**:  
- My Zigbee lamp responded slowly/erratically  
- After **removing and re-pairing it**, errors disappeared completely  

### Key takeaways:  
✔ You *can* keep your old Zigbee network when changing hardware/protocol  
⚠ But some **router devices cache old settings**, causing conflicts  
🔧 Solution: Re-pair problematic devices  
❗ z2m logs won't point to specific faulty routers – watch for odd device behavior instead