Koenkk/zigbee2mqtt #23199: Sonoff TRVZB Schedule (composite) entity state is too long for HA

markxroberts, 2024-06-29:
### What happened?

Repeated errors in HA logs because the entity state is too long
```
2024-06-28 01:15:56.348 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Invalid state message '{'friday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'monday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'saturday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'sunday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'thursday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'tuesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'wednesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16'}' from 'zigbee2mqtt/Utility radiator'
```

### What did you expect to happen?

Perhaps split in to multiple entities

### How to reproduce it (minimal and precise)

Pair valve, state message rejected.

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

zStack3x0 20220219 and ConBee2/RaspBee2 0x26780700

### Adapter

Sonoff ZB-Dongle-P and ConBee2

### Setup

Container on Debian VM / AMD Ryzen processor

fir3drag0n, 2024-06-29:
Same problem here

michelbalzer, 2024-07-04:
Maybe related to #19788 #home-assistant/core#103141 and #16460?

I have this message:

```log
Invalid state message '{'friday': '00:00/17 07:30/21 09:00/17 14:00/20 18:00/17 18:00/17', 'monday': '00:00/17 07:30/21 09:00/17 14:00/20 18:00/17 18:00/17', 'saturday': '00:00/17 08:00/21 12:00/17 14:00/20 18:00/17 18:00/17', 'sunday': '00:00/17 08:00/21 12:00/17 14:00/20 18:00/17 18:00/17', 'thursday': '00:00/17 07:30/21 09:00/17 14:00/20 18:00/17 18:00/17', 'tuesday': '00:00/17 07:30/21 09:00/17 14:00/20 18:00/17 18:00/17', 'wednesday': '00:00/17 07:30/21 09:00/17 14:00/20 18:00/17 18:00/17'}' from 'zigbee2mqtt/Kinderzimmer Thermostat'```

bchhabra, 2024-07-04:
same problem on z2m 1.38.0.

Logger: homeassistant.components.mqtt.sensor
Source: components/mqtt/sensor.py:265
integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 7:44:21 PM (1 occurrences)
Last logged: 7:44:21 PM
Invalid state message '{'friday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'monday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'saturday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'sunday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'thursday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'tuesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'wednesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16'}' from 'zigbee2mqtt/Office Trv'

Cannot update state for entity sensor.office_trv_weekly_schedule after processing payload on topic zigbee2mqtt/Office Trv. The requested state (477) exceeds the maximum allowed length (255). Fall back to unknown, failed state: {'friday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'monday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'saturday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'sunday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'thursday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'tuesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'wednesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16'}

Koenkk, 2024-07-06:
Pushed a fix, can you check if it works?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

markxroberts, 2024-07-06:
Thanks for doing this.  I've tried the dev branch.  I think that this truncates the state length for a text entity.  Am I right in thinking this is separate to a sensor entity (this is how this field is currently represented)?

Koenkk, 2024-07-08:
Yes this should truncate the text, I'm not sure what exactly this error means, @jbouwh, what does the following error mean?

```
2024-06-28 01:15:56.348 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Invalid state message '{'friday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'monday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'saturday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'sunday': '00:00/16 07:00/19 10:00/19 10:00/19 10:00/19 23:00/16', 'thursday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'tuesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'wednesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16'}' from 'zigbee2mqtt/Utility radiator'
```

raarts, 2024-07-11:
@Koenkk Maybe because JSON is expected and the message uses single quotes?

Koenkk, 2024-07-11:
@raarts I don't think the JSON is interpreted because I guess this error is for the text sensor?

raarts, 2024-07-12:
I was just guessing, reading it back the tone was not how it intended, sorry about that.

Der-WU, 2024-07-13:
Can cofirm that the error still exist.
I am using 8 TRVZB from Sonoff and got 2 errors:

`Logger: homeassistant.components.mqtt.sensor
Quelle: components/mqtt/sensor.py:265
Integration: MQTT ([Dokumentation](https://www.home-assistant.io/integrations/mqtt), [Probleme](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
Erstmals aufgetreten: 10:01:01 (156 Vorkommnisse)
Zuletzt protokolliert: 11:01:46

Invalid state message '{'friday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'monday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'saturday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'sunday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'thursday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'tuesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'wednesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16'}' from 'zigbee2mqtt/Badezimmer Heizung '
Invalid state message '{'friday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'monday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'saturday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'sunday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'thursday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'tuesday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'wednesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16'}' from 'zigbee2mqtt/Badezimmer Heizung '
Invalid state message '{'friday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'monday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'saturday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'sunday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'thursday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'tuesday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'wednesday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20'}' from 'zigbee2mqtt/Badezimmer Heizung '
Invalid state message '{'friday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'monday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'saturday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'sunday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'thursday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'tuesday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'wednesday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20'}' from 'zigbee2mqtt/Schlafzimmer Heizung'
Invalid state message '{'friday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'monday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'saturday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'sunday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'thursday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'tuesday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'wednesday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20'}' from 'zigbee2mqtt/Svenja Heizung'`

`Logger: homeassistant.components.mqtt.sensor
Quelle: components/mqtt/util.py:375
Integration: MQTT ([Dokumentation](https://www.home-assistant.io/integrations/mqtt), [Probleme](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
Erstmals aufgetreten: 10:01:01 (156 Vorkommnisse)
Zuletzt protokolliert: 11:01:46

Cannot update state for entity sensor.badezimmer_heizung_weekly_schedule after processing payload on topic zigbee2mqtt/Badezimmer Heizung . The requested state (477) exceeds the maximum allowed length (255). Fall back to unknown, failed state: {'friday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'monday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'saturday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'sunday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'thursday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'tuesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16', 'wednesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16'}
Cannot update state for entity sensor.badezimmer_heizung_weekly_schedule after processing payload on topic zigbee2mqtt/Badezimmer Heizung . The requested state (477) exceeds the maximum allowed length (255). Fall back to unknown, failed state: {'friday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'monday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'saturday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'sunday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'thursday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'tuesday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'wednesday': '00:00/16 07:00/19 10:00/16 10:00/16 17:00/19 23:00/16'}
Cannot update state for entity sensor.badezimmer_heizung_weekly_schedule after processing payload on topic zigbee2mqtt/Badezimmer Heizung . The requested state (477) exceeds the maximum allowed length (255). Fall back to unknown, failed state: {'friday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'monday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'saturday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'sunday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'thursday': '00 [... cut ...] aturday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'sunday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'thursday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'tuesday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20', 'wednesday': '00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20'}`

i have try to set every day only to 00:00/20 but on safe it create 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 00:00/20 .
 sorry for my bad english.

Greatz

Der-WU, 2024-07-29:
Any update about this issue?

h3llrais3r, 2024-07-31:
Same problem here.

michelbalzer, 2024-08-02:
Seems fixed with 1.39.1?

Der-WU, 2024-08-02:
Still exist on 1.39.1. :(

h3llrais3r, 2024-08-05:
I'm also still seeing the same warning with latest 1.39.1 release.

Koenkk, 2024-08-05:
I cannot reproduce this issue with my own TRVZB on 1.39.1, can you try to remove the device via the z2m frontend (such that it disappears from HA) and re-pair it?

markxroberts, 2024-08-06:
I've tried deleting and re-pairing.  I'm sorry to say that it still gives me the same error.  If you don't use this particular entity, you can of course disable it, which has solved the problem for me!

Der-WU, 2024-08-12:
Confirm. Delete all trv. Repair them. Error comes back. Disable the enteiti Remove the errors. I Do Not use the sheudle. So it is Not a Problem die me. But it still exist. :) Great

fir3drag0n, 2024-09-11:
Problem still exists

ishbuggy, 2024-09-20:
Problem still exists in TRVZB version 1.1.5.

Koenkk, 2024-09-20:
Finally managed to find and fix the issue!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-20 by Koenkk

hilburn, 2024-10-25:
@Koenkk would it be possible to change the behaviour of the converter to instead split it out into separate attributes for each day:
```
schedule_monday:
schedule_tuesday:
```
etc rather than merging the value and truncating them?

mohoo, 2024-11-23:
@hilburn That would be really helpful. Currently the weekly_schedule attr is useless in home assistant because it is cut off.

nosektomas2004, 2025-01-29:
Hey, any update on this? Also I have the JSON string cut off so I can't use it 😕