Koenkk/zigbee2mqtt #23204: Updated IKEA E2001/E2002 to latest FW and now doesn't respond

smartroad, 2024-06-29:
### What happened?

I updated my IKEA E2001/E2002 controller over OTA and all seemed to be okay. Now that it has updated, it is no longer responding. It joins fine, but there is nothing when you press a button (action never changes). It also shows the battery as 200%

[image]

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Universal Zigbee 3.0 USB Dongle Plus ZBDongle E

### Setup

Addon Home Assistant

Kevincoenen, 2024-07-01:
I'm having similar issues with the latest firmware on the IKEA E2001/E2002. The buttons don't seem to work, or only long presses are recognized by the device. Reinserting the batteries in the device seems to fix the issue for a couple of days, but the problem always comes back. I have this issue with all of my IKEA E2001/E2002 devices (three of them). I even migrated from a SONOFF ZBDongle-P to an SLZB-06M adapter, but the problem persists on both of these adapters.

peternijssen, 2024-07-01:
Similar comments are reported here also; https://github.com/Koenkk/zigbee2mqtt/issues/22651

schafran, 2024-07-03:
Similar for me. Updated the devices to 2.4.11 and the arrow buttons stopped working. On and off buttons still doing fine.

martintilma, 2024-07-03:
I'm running: Z2M: 1.39.0 commit: 0326926, HA: 2024.6.4, Firmware: 2.4.11

When the buttons stops working I remove the battery and put them back. Then the buttons start working again. After a day they stop working again. 
How can I help debugging / fixing?

Nezz, 2024-07-10:
I have the same issue. Ever since the firmware update the remote has barely been working. It works after reinserting the battery for a day or two. The issue is still not fixed in 1.39.0. Is it possible to revert the firmware?

I tried pairing again and using the suggested binding, but it still does not work reliably.
[image]

[image]

For example pressing the right button publishes one of these two variations:
```
info 2024-07-10 22:52:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"arrow_right_release","action_duration":18.127,"battery":100,"last_seen":"2024-07-10T22:52:31+03:00","linkquality":131,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:52:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"","battery":100,"last_seen":"2024-07-10T22:52:31+03:00","linkquality":131,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:52:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote/action', payload 'arrow_right_release'
```

```
info 2024-07-10 22:52:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":null,"battery":100,"last_seen":"2024-07-10T22:52:58+03:00","linkquality":131,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:52:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":null,"battery":100,"last_seen":"2024-07-10T22:52:58+03:00","linkquality":134,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:52:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":null,"battery":100,"last_seen":"2024-07-10T22:52:59+03:00","linkquality":134,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:52:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"arrow_right_hold","battery":100,"last_seen":"2024-07-10T22:53:00+03:00","linkquality":134,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:52:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"","battery":100,"last_seen":"2024-07-10T22:53:00+03:00","linkquality":134,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:52:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote/action', payload 'arrow_right_hold'
```

Pressing left:
```
info 2024-07-10 22:54:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"arrow_left_release","action_duration":17.078,"battery":100,"last_seen":"2024-07-10T22:54:30+03:00","linkquality":127,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:54:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"","battery":100,"last_seen":"2024-07-10T22:54:30+03:00","linkquality":127,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:54:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote/action', payload 'arrow_left_release'
```

```
info 2024-07-10 22:54:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":null,"battery":100,"last_seen":"2024-07-10T22:54:46+03:00","linkquality":131,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:54:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":null,"battery":100,"last_seen":"2024-07-10T22:54:47+03:00","linkquality":127,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:54:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"arrow_left_hold","battery":100,"last_seen":"2024-07-10T22:54:48+03:00","linkquality":131,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:54:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"","battery":100,"last_seen":"2024-07-10T22:54:48+03:00","linkquality":131,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:54:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote/action', payload 'arrow_left_hold'
```

Pressing up and down does not do anything most of the time. When it does, this is published:
```
info 2024-07-10 22:55:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"brightness_stop","battery":100,"last_seen":"2024-07-10T22:55:59+03:00","linkquality":131,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:55:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"","battery":100,"last_seen":"2024-07-10T22:55:59+03:00","linkquality":131,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
[... cut ...] led_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:56:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"","battery":100,"last_seen":"2024-07-10T22:56:26+03:00","linkquality":134,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
info 2024-07-10 22:56:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote/action', payload 'brightness_move_down'
```

Nezz, 2024-07-10:
After reinserting the battery it starts to work normally.
Left:
```
[2024-07-10 23:44:11] debug: 	z2m: Received Zigbee message from 'Bedroom Remote', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":257,"value2":13}' from endpoint 1 with groupID 0
[2024-07-10 23:44:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"arrow_left_click","battery":100,"last_seen":"2024-07-10T23:44:11+03:00","linkquality":91,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
[2024-07-10 23:44:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"","battery":100,"last_seen":"2024-07-10T23:44:11+03:00","linkquality":91,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
[2024-07-10 23:44:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote/action', payload 'arrow_left_click'
```

Right:
```
[2024-07-10 23:44:06] debug: 	z2m: Received Zigbee message from 'Bedroom Remote', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":256,"value2":13}' from endpoint 1 with groupID 0
[2024-07-10 23:44:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"arrow_right_click","battery":100,"last_seen":"2024-07-10T23:44:06+03:00","linkquality":91,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
[2024-07-10 23:44:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"","battery":100,"last_seen":"2024-07-10T23:44:06+03:00","linkquality":91,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
[2024-07-10 23:44:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote/action', payload 'arrow_right_click'
```

Up:
```
[2024-07-10 23:44:14] debug: 	z2m: Received Zigbee message from 'Bedroom Remote', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
[2024-07-10 23:44:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"on","battery":100,"last_seen":"2024-07-10T23:44:14+03:00","linkquality":94,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
[2024-07-10 23:44:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"","battery":100,"last_seen":"2024-07-10T23:44:14+03:00","linkquality":94,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
[2024-07-10 23:44:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote/action', payload 'on'
```

Down:
```
[2024-07-10 23:44:18] debug: 	z2m: Received Zigbee message from 'Bedroom Remote', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
[2024-07-10 23:44:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"off","battery":100,"last_seen":"2024-07-10T23:44:18+03:00","linkquality":83,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
[2024-07-10 23:44:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote', payload '{"action":"","battery":100,"last_seen":"2024-07-10T23:44:18+03:00","linkquality":83,"update":{"installed_version":33816593,"latest_version":33816593,"state":"idle"},"update_available":null}'
[2024-07-10 23:44:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Remote/action', payload 'off'
```

Bkrickl, 2024-07-11:
I also have the exact same issue and did not find a solution yet. I am at 1.36.1.

EricV-hue, 2024-07-12:
Discussed this also in the forum topic [https://github.com/Koenkk/zigbee2mqtt/discussions/23243](url).

After experiencing the same things as mentioned above I currently use a Trädfri hub and one Styrbar remote + lightbulb, to check the remote. This combination is now already active for at least 2 days without interruption.
I now will try to install an older Z2M version as local add-on. See what happens. I will look for a version before 1.36.1.

DerOetzi, 2024-07-16:
Have same issue, maybe some kind of "timeout" I have one in my office room works fine the whole week, after 2.5 day no usage over weekend have to reinsert the battery

EricV-hue, 2024-07-16:
Quick update:
The remote on the Trädfi hub is still working as it should be.
My Z2M is now on 1.35.0. The remote connected to HA works fine, key presses are reported correctly, even for multiple days. Until the `left arrow` key is pressed. The response then gets unpredictable: sometimes i get `brightness_move_down`, `or brightness_stop`. Also the long keypress behave inconsistant. i think this behaviour was intended to be solved in 1.36.0?
At some point, after pressing the `left arrow` key a number of times the remote stopped. After a Soft reset (battery remove) the responses were normal.

Does this suggest that it is not the remote control update but the z2m software causing the problem?

Nezz, 2024-07-17:
The issue is resolved by a firmware downgrade, you can follow the instructions here:
https://github.com/Koenkk/zigbee2mqtt/discussions/23243#discussioncomment-10038494

EricV-hue, 2024-07-17:
Although downgrading is working, I think this is more a workaround than a solution. 
Downgrading the remote does not explain why the same remote combined with a Trådfri hub works without problem.

is there a handy way to do some sort of sniffing on mqtt messages?

Nezz, 2024-07-17:
It is indeed a workaround. I believe zigbee traffic sniffing would be needed:
https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

IKEA probably introduced something in the new firmware that causes the device to enter a bad state after a day or two. The Tradfri hub might send something to the remote once a day that keeps it in the working state. Or it might set up a reporting that's needed to not brick the remote.

EricV-hue, 2024-07-18:
Maybe. But that does not seem to explain why the remote keeps working with z2m 1.35.0. That is, as far as `brightness_move_up` and `brightness_move_down` are concerned.

martintilma, 2024-07-19:
I played with the Availability setting (from simple to advanced) but I didn't notice any difference. The last days the remotes are working fine again.

EricV-hue, 2024-07-19:
Which version of z2m and remote firmare are you using?

martintilma, 2024-07-20:
[1.39.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.0) commit: [0326926](https://github.com/Koenkk/zigbee2mqtt/commit/0326926) 

Remote firmware: [2.4.11](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) 
HA: 2024.7.2

victorhooi, 2024-07-25:
I also seem to be hitting this issue, and would love to see it resolved!

My information in case that helps:

Zigbee2Mqtt: [1.39.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.0) commit: [0326926](https://github.com/Koenkk/zigbee2mqtt/commit/0326926)
Adapter: [UZG Gateway (UZG-01)](https://uzg.zig-star.com/product/), running ZXG firmware ([240707](https://github.com/xyzroe/XZG/releases/tag/20240707)), and z-stack [240710](https://github.com/Koenkk/Z-Stack-firmware/discussions/505) (although I've also tried with [20230507](https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20230507))]
IKEA E2002 Styrbar Remote: Firmware 2.4.11

I do note that like others, taking out the battery, and putting it back in seems to make the remote work for about 1 day - but then it suddenly stops working.

I can try looking at Zigbee traffic sniffing, if people think that would help - is https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#_3-sniffing-traffic the best way to start?

I wonder if there's some way of doing it from the Zigstar adpater itself - or is it best to get a secondary Zigbee-capable device/adapter to do the sniffing? And I assume it would need to sniff for more than a 24-hour period?

EricV-hue, 2024-07-27:
Quick update: i now have 2 Styrbar remotes working with no issues for about 1 week.
Z2M version 1.35.0, Styrbar 2.4.11

Koenkk, 2024-07-27:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/3d766bc346c74cb985b75cbb13eb7c6e): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

EricV-hue, 2024-07-28:
Will do that. 
Updated z2m to 1.39.0 and now waiting for the issue to reoccur (or not?).

Nezz, 2024-07-28:
@Koenkk I tried to load the external converter but no luck:
```
z2m: Failed to load external converter file 'ext_converter.js' (Cannot find module '../lib/develco'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js)
```

After removing the unused imports:
```
z2m: Failed to load external converter file 'ext_converter.js' (Cannot read properties of undefined (reading 'options'))
```

Bkrickl, 2024-07-28:
> ```yaml
> ```yaml
> external_converters:
>   - ext_converter.js
> ```
> 
> 
>     
>       
>     
> 
>       
>     
> 
>     
>   
> ```

Same for me

Bkrickl, 2024-07-28:
Will also try it.
Had to adjust the code a little bit so it is working for me on 1.36.1.

Commented out the following:

// const {develcoModernExtend} = require('../lib/develco');
// const {battery} = require('../lib/modernExtend');
// if (utils.hasAlreadyProcessedMessage(msg, model)) return;

Since these things are not available for me

Then i added
const globalStore = require('zigbee-herdsman-converters/lib/store');

Will come back to you to tell if i have or don't have any issues.

Koenkk, 2024-07-29:
> @Koenkk I tried to load the external converter but no luck:
>
> ```
> z2m: Failed to load external converter file 'ext_converter.js' (Cannot find module '../lib/develco'
> Require stack:
> - [...]

Whoops, updated https://gist.github.com/Koenkk/3d766bc346c74cb985b75cbb13eb7c6e

Nezz, 2024-07-29:
I tried the updated one and @Bkrickl's suggestions too, but it's still failing to load with 1.39.0. I believe that's because in [1.36.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1) and later:
>Due to some internal changes, some external converters will break. To fix it simply remove const extend = require('zigbee-herdsman-converters/lib/extend'); from the external converter.

Nezz, 2024-07-29:
I managed to make Z2M work with it, here is what worked in the end:
https://gist.github.com/Nezz/d036fb0047b667e7c91a13ccd5a4b33c

Maverick5269-lab, 2024-07-31:
On the ikea thread says a new firmware with fixes are available 2.4.16 but i think is only available for ikea hub.

can someone know the download link manually?

Nezz, 2024-07-31:
Looks like they indeed released that update to fix these issues:
https://www.reddit.com/r/tradfri/comments/1e4lq3f/styrbar_firmware_version_2416/

Sadly they have not published it here:
http://fw.ota.homesmart.ikea.net/feed/version_info.json

Maverick5269-lab, 2024-07-31:
> Looks like they indeed released that update to fix these issues: https://www.reddit.com/r/tradfri/comments/1e4lq3f/styrbar_firmware_version_2416/
> 
> Sadly they have not published it here: http://fw.ota.homesmart.ikea.net/feed/version_info.json

I think this new firmware fixes the remote that enters a bugged state after a day or so.
Without it I think it will always enters in this state after sometime in standby. From the Ikea thread this happens not only in the home assistant but with Ikea hub too. I assume this is a firmware problem with device and not from Zigbee2Mqtt.

> Looks like they indeed released that update to fix these issues: https://www.reddit.com/r/tradfri/comments/1e4lq3f/styrbar_firmware_version_2416/
> 
> Sadly they have not published it here: http://fw.ota.homesmart.ikea.net/feed/version_info.json

Thanks I didn't knew were they were published.
But appears the IKEA Hub goes to other address getting it. 
From the thread the hub is DIRIGERA that applies the new version.

Bkrickl, 2024-08-01:
For me the same issue occurs even with the external converter.
So i think i have to wait for the updated to available for z2m.

Maybe it will work with this update.

Bkrickl, 2024-08-09:
Update is now available.
I already installed it for some of my devices.
All actions seems to work without any issues.

Will come back when tried it multiple days to see if the issue is now resolved

GermanDZ, 2024-08-09:
Release notes: https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

martintilma, 2024-08-10:
After our holiday 2 out of 3 remotes where working. Happy to see an update available. I will install them too to see if this solves the issue.

ElBalsero, 2024-08-10:
I am not able to bind to groups in this version. Was someone luckier?

EricV-hue, 2024-08-13:
Installed the updates in my 2 remotes. Working fine now. Should this change within a few days I will get back again.
Thanks for your comments and support

jschroeder624, 2024-08-13:
> Update is now available. I already installed it for some of my devices. All actions seem to work without any issues.
> 
> Will come back when tried it multiple days to see if the issue is now resolved

Can I ask how you installed the update? Every day I check with OTA via ZIGBEE2MQTT and 'No update available'

Nezz, 2024-08-13:
I must admit that I pressed "check for updated" many times before realizing that I enabled the OTA IKEA test url out of desperation a couple weeks ago.

jschroeder624, 2024-08-13:
> > > Update is now available. I already installed it for some of my devices. All actions seem to work without any issues.
> > 
> > 
> > > 
> > 
> > 
> > > Will come back when tried it multiple days to see if the issue is now resolved
> > 
> > 
> > Can I ask how you installed the update? Every day I check with OTA via ZIGBEE2MQTT and 'No update available'
> 
> I had to turn off OTA IKEA test url and reboot, then they showed up for my 5 remotes. I had to press a button on the remote while searching for update, to establish a connection

Oh, thank you, thank you, thank you! Many, many thanks as I forgot that I had enabled the test url similarly out of desperation just like @Nezz. How long this might have gone on? Possibly forever....

Colbyjdx, 2024-08-14:
> I am not able to bind to groups in this version. Was someone luckier?

Just upgraded one of my remotes and I have the exact same issue - bindings to individual lights work, but no longer for groups.

I didn't notice any issues with 2.4.11, and everything is still working fine on my other remotes still running that firmware.

jschroeder624, 2024-08-14:
> Just upgraded one of my remotes and I have the exact same issue - bindings to individual lights work, but no longer for groups.
>
> I didn't notice any issues with 2.4.11, and everything is still [...]

I am not sure what binding to groups does, but my switch does control a light group without this setting. At 2.4.16 and without 'Groups' checked, light groups do work. I think was a temporary workaround for something, but I'm not sure.

ElBalsero, 2024-08-14:
> I am not sure what binding to groups does, but my switch does control a light group without this setting. At 2.4.16 and without 'Groups' checked, light groups do work. I think was a temporary [...]

When you bind a switch to a group (or to a device, for what it matters...) what you do is to set the switch to communicate with that group directly without having to rely the communication through the hub.

This means that if your Zigbe2Mqtt goes down for whatever reason, that switch will continue to work and to control the bound group of lights.

jschroeder624, 2024-08-14:
> When you bind a switch to a group (or to a device, for what it matters...) what you do is to set the switch to communicate with that group directly without having to rely the communication through [...]

This sounds great - had no idea this was an option. Is there some (better) documentation that was is listed in the Zigbee2MQTT doc? For example, how do you control which buttons and what the buttons do when bound directly to a light or group?

ElBalsero, 2024-08-14:
Yeah… I can relate to what you say. Perhaps someone else can link a decent document. What I know comes out of pieces of information here and there.

Basically you can separately bind the clusters belonging to every endpoint of the switch.

There’s not much room for configuration besides that.

Some examples:

If you bind the OnOff cluster, the switch will be directly switching on and off the lights in the group.

If you bind the LevelCtrl cluster, your switch will control the intensity of the lights.

If you bind the Scenes cluster, it’ll control also the color or temperature.

There are other clusters that I have no idea what they do. Also you do not choose what parts of the switch do what. This particular Ikea switch, when bound, controls on/off with single presses of the on/off buttons, intensity with long presses of the same buttons and temperature or color with the arrows. That’s how it’s designed, and the functionality you get also depends on the lights you’re using.

And the last thing I know comes back to the endpoints I was mentioning. You can bind different clusters belonging to every endpoint. So, if you have a double switch with two endpoints, you can choose to bind the on/off of the first to a group and the on/off of the second to another different group. However, if your double switch has only one endpoint (like some of the ones made by Philips), if you use the binding functionality, you can only bind the only existing endpoint to one group and both buttons of the switch would then control the same group… which is a stupid waste. So, if you’re looking for double switches to use for binding, don’t make the same mistake a made and make sure you get one with two endpoints.

Bottom line: you do not have a lot of control, but when you get the right combination and things work, you get a home which continues to work even when you’re Zigbee2Mqtt server goes down, which is a big plus.

Only you have to be very careful with firmware updates like in this case. If I had updated all my switches without asking here before… all would be gone as this version seems to have broken the binding functionality.

Sorry for the long reply. I hope at least helps a bit.

jschroeder624, 2024-08-14:
This is great info, thanks. I use the Styrbars for a couple bedside switches for 2 lamps and a ceiling fan light group. The main switch when you walk into the room is a Rodret, so maybe I will tinker with that for binding directly to the light group. Appreciate the info - today I learned.

kevinBecker, 2024-08-16:
I’m trying to connect my styrbar (purchased ~2 weeks ago) to my zigbee network, but I’m not getting anything. I’m worried mine shipped with the broken firmware. I have no intentions of getting the ikea hub to update the remote. Should I just return it? Over 6+ hrs of trying, it’s half connected twice, but HA only received a single battery power update, and the button presses didn’t show in the logs.

Any ideas on what to do? I’m at a loss.

PS:   I’ve followed the instructions on the z2m website (press 4x rapidly until red breathing light. Press 10x rapidly if that fails. Also tried removing and replacing batteries/starting over).

EricV-hue, 2024-08-16:
Half connected? Did it show up in the z2m devices list?

elle2, 2024-09-03:
I've updated STYRBAR (E2001/2002) to 2.4.16 and now it is joined in Z2M but no action to keys press (up, down, left and right). 
In log there is no MQTT message sent and no error message.
If I try to re-configure in Z2M the device the log report this:

`[2024-09-03 15:24:54] error:    z2m: Failed to configure 'dimmer ale', attempt 1 (Error: Bind 0x4c5bb3fffe20450f/1 genOnOff from '0x00124b001938448b/1' failed (AREQ - ZDO - bindRsp after 10000ms)`

Z2M: 1.40
HA: 2024.8.2
Remote Firmware: 2.4.16

ElBalsero, 2024-09-04:
> I've updated STYRBAR (E2001/2002) to 2.4.16 and now it is joined in Z2M but no action to keys press (up, down, left and right). In log there is no MQTT message sent and no error message. If I try to re-configure in Z2M the device the log report this:
> 
> `[2024-09-03 15:24:54] error: z2m: Failed to configure 'dimmer ale', attempt 1 (Error: Bind 0x4c5bb3fffe20450f/1 genOnOff from '0x00124b001938448b/1' failed (AREQ - ZDO - bindRsp after 10000ms)`
> 
> Z2M: 1.40 HA: 2024.8.2 Remote Firmware: 2.4.16
> 
> Update: magically now there are messages related to keys press. The battery level in Z2M web GUI is shown as red % with no value (and is unknown in HA)

Hey @elle2 ,
That error message looks like an error when trying to establish a bind  the on/off cluster.
As discussed a few messages earlier, this version really looks like it’s breaking the binding. So, if you’re trying en bind the Styrbar to a group, chances are that it’s not going to work.

macluxHD, 2024-11-07:
I updated one of my STYBAR switches to 2.4.16, I was able to bind my TRADFRI lights with the Clusters (LevelCtrl, OnOff, Scenes) turning the lights on/off and dimming works but not changing colors with the arrows.

The logs do show that I am pressing the left and right arrows but nothing happens.
```
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/light switch', payload '{"action":"arrow_left_click","battery":null,"linkquality":30,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/light switch', payload '{"action":"","battery":null,"linkquality":30,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/light switch/action', payload 'arrow_left_click'
```
Also it does not recognize the battery level anymore either.

ElBalsero, 2024-11-07:
Hi @macluxHD 

When you bind two devices, what you are actually doing is to make them talk to each other straight away bypassing the hub.

Roughly, when you press a button, and you have bound the devices, two things happen:

1. The button press information is still sent to the hub. That is what you see in your logger. The  hub, in your case, whatever platform where you are running your MQTT broker (Home Assistant?) will do nothing with it unless you have set an automation to handle it.
2. A direct Zigbee message is sent from the switch to the lights. Straight away, without using your broker. That message is hardcoded into the switch, and the way that the light will respond to it is hardcoded into the light. 

This second message you will not be able to trace it in any logger in your platform (Home Assistant?). To be able to trace that message you need a Zigbee sniffer and Wireshark.

You said that you got to bind the STYRBAR with your TRADFRI lights and on/off is working. I had a brief moment in the past when I also thought I had succeeded to do that with STYRBAR 2.4.16. However I was wrong.

Would you make a small test to double check? Shut down your Zigbee2MQTT module. Make sure that is down by trying other lights. They should of course not respond. Then try again this STYRBAR that you bound. Is the TRADFRI still responding to On/Off? Then you indeed succeed to bind that one. It is not responding anymore? Then you are in the same situation I briefly was. I thought I had successfully bound them, but actually the On/Off messages were handled by Zigbee2MQTT and not via the binding.

Regards,

ElBalsero

macluxHD, 2024-11-07:
Hi @ElBalsero

Thanks for your message

So I shut down my raspberry pi running z2m, mosquitto and Home assistant and turning the lights on/off and dimming them is still working. So I did actually bind the devices directly to each other. 

Ideally I want the switch to work like it did before, so that I can change colors without needing a home assistant automation, maybe I am binding them incorrectly but as far as I understand if I bind the Scene Cluster it should bind the color changing. 

I haven't done that before so I am not sure, with the old firmware I was able to bind them by just holding the pair button for 10 seconds which does not seem to work anymore, I had to do it through the z2m binding UI.

ElBalsero, 2024-11-07:
Oh… those are good news then.
I have never been able to do the binding with the 10 second thing. I’ve always done it via z2m.
I’ll give it another try then.
Are you binding an individual light or a group? I mean a group of lights grouped in z2m UI with its own name. Not a group in the sense of binding first light and then repeating the binding with the next one.
I am not sure of I’m being clear…

macluxHD, 2024-11-07:
I didn't use a group.

At first I tried putting all the lights into one group and then binding that group, but that didn't work at all.
To get it to work with at least on/off and dimming I created a binding for each bulb separately.

Just to be sure I removed every device from z2m then factory reset them, so for the bulbs turning them off and on in quick succession and pressing the bind button 4 times on the switch. After z2m added all of them I just went into the binding UI and while pressing a button on the switch to keep it awake I added a binding for each light bulb.