Koenkk/zigbee2mqtt #23205: CO2 sensor TS0601 _TZE204_ogkdpgy2 defined but shows as not supported

zlobnic, 2024-06-29:
### What happened?

I have CO2 censor with ZigBee support (https://aliexpress.ru/item/1005003446514631.html). 

Zigbee2MQTT version 1.38.0 (HA 2024.6.4) can see it and also adds. But there's no support. Don't understand what to do... tried EDGE version, no luck.

[image]
[image]
[image]
[image]

Another ZigBee device works fine (Xiaomi Button)

### What did you expect to happen?

CO2 data...

In this video same sensor works fine: https://youtube.com/watch?v=4aQye8tArmM

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20201128

### Adapter

Wireless Zigbee CC2531 CC2540 Sniffer

### Setup

Addon @ H96Max

zlobnic, 2024-06-30:
UPD: this tip worked for me: https://github.com/Koenkk/zigbee2mqtt/issues/23183#issuecomment-2198652112

closed 2024-07-01 by zlobnic

malexmnt, 2024-07-18:
for more understanding:
1.stop z2m
2. you need to change _TZE204_ogkdpgy2 to _TZE200_ogkdpgy2 in the text file /homeassistant/zigbee2mqtt/database.db
3.start z2m, and check if the device works.

rokiden, 2024-09-07:
I have fixed flood bug of this device with [custom firmware](https://github.com/rokiden/Tuya_CO2_STC8/blob/tuya_co2/user/README.md).

heisenberg2980, 2024-10-16:
> for more understanding: 1.stop z2m 2. you need to change _TZE204_ogkdpgy2 to _TZE200_ogkdpgy2 in the text file /homeassistant/zigbee2mqtt/database.db 3.start z2m, and check if the device works.

@malexmnt do you know if this workaround is still needed? I thought by now this device would have been added to z2m already

vgdh, 2024-11-03:
> > for more understanding: 1.stop z2m 2. you need to change _TZE204_ogkdpgy2 to _TZE200_ogkdpgy2 in the text file /homeassistant/zigbee2mqtt/database.db 3.start z2m, and check if the device works.
> 
> @malexmnt do you know if this workaround is still needed? I thought by now this device would have been added to z2m already, but I just tried and z2m says the device is unsupported

I've just now tested the method and it works