Koenkk/zigbee2mqtt #23206: New device:  Moes , Thermoestatic Valve

egarijo, 2024-06-29:
### Link

{"id":21,"type":"Router","ieeeAddr":"0xa4c138b618017288","nwkAddr":4156,"manufId":4417,"manufName":"_TZE204_qasjif9e","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0006\u0000\u0000\u0000e\u0007\u0000\u0000\u0000\u0012\u0006q�-\u0013\u0006q�-e\u0007q�-\u0012\u0005\u0000\u0000\u0000e\u0006\u0000\u0000\u0000\u0012","65506":56,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE204_qasjif9e","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1718784631633}

### Database entry

{"id":21,"type":"Router","ieeeAddr":"0xa4c138b618017288","nwkAddr":4156,"manufId":4417,"manufName":"_TZE204_qasjif9e","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0006\u0000\u0000\u0000e\u0007\u0000\u0000\u0000\u0012\u0006q�-\u0013\u0006q�-e\u0007q�-\u0012\u0005\u0000\u0000\u0000e\u0006\u0000\u0000\u0000\u0012","65506":56,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE204_qasjif9e","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1718784631633}

### Comments

I created the file T0601.js file following the instructions:
the file is the following:

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TRV FSZPJYDC', // Update this with the real model of the device (written on the device itself or product page)
    vendor: 'Moes', // Update this with the real vendor of the device (written on the device itself or product page)
    description: 'THERMOSTATE VALVE', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: [],
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.valve_possition(), e.state()] // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

After that I have added the following lines on the configuration.yml

advanced:
  log_level: debug
external_converters:
  - TS0601.js
But still the same. Nothing happens.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_9mjy74mp',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

LintHart, 2024-07-27:
Hi @egarijo , did you ever get anywhere with these? I unfortunately bought a bunch of them without realising they're not supported. Also having trouble creating an external definition

egarijo, 2024-07-27:
Hi. At the moment I can't do anything with that. I'm Waiting if with the
time, somebody solves the problem. I have time to the next winter.
Regards

LintHart, 2024-07-30:
I had some luck with learning how to work with DDFs in Phoscon as it has a builder and managed to adjust a template to read and control the temperature. I did figure out how to start an external definition but my skills aren't there yet to finish it. 

I bought a Tuya hub and extracted all of the DPIDs in hope that we might be able to get some assistance :)

`
"1":"Mode",
"2":"Target temperature",
"3":"Current temperature",
"6":"Working status",
"7":"Window status",
"8":"Open window",
"12":"Child lock",
"13":"Battery",
"14":"Fault alarm",
"15":"Min. limit temperature",
"16":"Max. limit temperature",
"17":"Heating schedule",
"18":"Week program Tuesday",
"19":"Week program Wednesday",
"20":"Week program Thursday",
"21":"Week program Friday",
"22":"Week program Saturday",
"23":"Week program Sunday",
"101":"Room sensor calibration",
"108":"Valve",
"109":"Models",
"110":"Motor thrust",
"111":"Display brightness",
"112":"Software version",
"113":"Screen orientation",
"114":"System mode",
"115":"Switch deviation (energy-saving mode only)",
"116":"Motor data"
`

micheltol, 2024-09-17:
@egarijo @LintHart any luck? I'm in the same boat and pretty much a newbee in zigbee2mqtt-space 😦

hideintheclouds, 2024-10-01:
I think that the device is a Tuya type, so a different converter should be created, according to the documentation here:
https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html

I had the same issue with a [Moes TRV](https://www.aliexpress.com/item/1005006068601747.html?spm=a2g0o.order_list.order_list_main.5.187518022PIULc) (_TZE204_9mjy74mp TRV801Z) and was able to make it work by following the guide above.

Inspired by @LintHart extracted DPIDs, and looking at tuya.js, I was able to find a similar config [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/13ca5a04abfb6e38a03aee0d3eb4737a18e33071/src/devices/tuya.ts#L4933C1-L4934C1)

The converter js should be placed next to configuration.yaml and can be referenced either via the config or the Home Assistant Zigbee2MQTT side addon.

I ended up with something like this:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');                                                                                          
const tz = require('zigbee-herdsman-converters/converters/toZigbee');                                                                                            
const exposes = require('zigbee-herdsman-converters/lib/exposes');                                                                                               
const reporting = require('zigbee-herdsman-converters/lib/reporting');                                                                                           
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');                                                                                     
const e = exposes.presets;                                                                                                                                       
const ea = exposes.access;                                                                                                                                       
const tuya = require('zigbee-herdsman-converters/lib/tuya');                                                                                                     
                                                                                                                                                                 
const definition = {                                                                                                                                             
    // Since a lot of Tuya devices use the same modelID, but use different datapoints                                                                            
    // it's necessary to provide a fingerprint instead of a zigbeeModel                                                                                          
    fingerprint: [                                                                                                                                               
        {                                                                                                                                                        
            // The model ID from: Device with modelID 'TS0601' is not supported                                                                                  
            // You may need to add \u0000 at the end of the name in some cases                                                                                   
            modelID: 'TS0601',                                                                                                                                   
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.                                                                        
            manufacturerName: '_TZE204_9mjy74mp',                                                                                                                
        },                                                                                                                                                       
    ],                                                                                                                                                           
    model: 'TRV801Z',                                                                                                                                            
    vendor: 'Moes',                                                                                                                                              
    description: 'Moes old TRV/zigbee model',                                                                                                                    
    fromZigbee: [tuya.fz.datapoints],                                                                                                                            
    toZigbee: [tuya.tz.datapoints],                                                                                                                              
    onEvent: tuya.onEventSetLocalTime,                                                                                                                           
    configure: tuya.configureMagicPacket,                                                                                                                        
[... cut ...]
                [101, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
                [108, 'position', tuya.valueConverter.divideBy10],
                [114, 'mode', tuya.valueConverterBasic.lookup({comfort: tuya.enum(0), eco: tuya.enum(1)})],
            ],
        },

    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;

```

Now this converter is probably not 100% reliable, but it seems to be working fine in Home Assistant.
This might be useful for you.

egarijo, 2024-10-02:
Thanks for your info.
I am not an expert on that matter, but I will try to test it.
For your informatin, I bougth the valve to Moes. But it seams must be
manufacturing by Tuya.

Thanks again.

egarijo, 2024-10-02:
Hi, at the moment I have not chance.
I will try to test the propossal by @Adrian M and test it. But I will need
time.
Regards

monteship, 2024-10-04:
> I think that the device is a Tuya type, so a different converter should be created, according to the documentation here: [...]

Confirm as working solution

egarijo, 2024-10-04:
Hi, I can't confirm anything at the moment.

egarijo, 2024-10-09:
Hi Adrian.
Today I had time to test.
Now the device appears as "supported".
[image: image.png]

But if I check the values exposed, all of them appears without any
information.

[image: image.png]

I follow all the steps, but someone must be missing, and I can't find it.
¿Do you have any suggestion?
Regards

Joshi120, 2024-10-21:
Heya, i am very confused. i added the code from above next to the Config file and referenced it but it still says not supported. I own the exact TRV mentioned above the code

egarijo, 2024-10-22:
Hi,

I don't know what is happened, but I can explain you what I did.

I copied the code on this file: TS0601_u000.js

[image: image.png]

After that, on the configuration.yml I added the following the external
converters.
[image: image.png]

And that's all. As you can see in my homeassistant, appears as suported.

[image: image.png]

Regards

PedroKTFC, 2024-10-25:
> I think that the device is a Tuya type, so a different converter should be created, according to the documentation here: [...]

Looking at the code you ended up with above and the code you reference just before it as your inspiration, I can't see any significant difference. Excuse my ignorance but can you explain what you've changed. I'm using ZHA in HA and I'm getting some basic functionality but I'd like to get more if I can work out how. Thanks.

hideintheclouds, 2024-10-26:
@PedroKTFC The difference is: value conversion/lookup for the alarm_switch, as the existing config was throwing an exception when deserializing the received payload from the TRV:
                [14, 'alarm_switch', tuya.valueConverterBasic.lookup({ON: 1, OFF: 0})],
vs. the initial one:
                [14, 'alarm_switch', tuya.valueConverter.onOff],

LAD47, 2024-10-26:
@hideintheclouds The same thermostat from Moes comes with manufacturerName: "_TZE200_rtrmfadk". It is automatically installed by zigbee2mqtt. However, it has the same problem you explained with the 'alarm_switch'. I have tested '_TZE200_rtrmfadk' and it works without error messages when I use your 'alarm_switch' definitions. It would probably be wise to fix '_TZE200_rtrmfadk' as well.

egarijo, 2024-10-29:
Hi.

I have testedthe same. The thermostat from Moes comes with differences
inside. As you can see on the picture, one is discovered as TRV602 (Tuya)
and the same thermostat is discovered as TRV801Z.  This difference also
appears on the reference of the valve. The first one is discovered
automatically by z2mqtt and for the other I have to include an external
connector. Also there are a small differences on the values expossed.
[image: image.png]

The alarm switch, is not working on the first one.
[image: image.png]

I never tested with ZHA.

Regards

canope-25, 2024-10-30:
> @PedroKTFC The difference is: value conversion/lookup for the alarm_switch, as the existing config was throwing an exception when deserializing the received payload from the TRV: [...]

Hi ,

First of all thank you for the job done 😄.

I would like to share the fact that some useful settings are still not yet supported by the converters proposed there.

It will be useful to also have the ability to define :
- the eco mode hysteresis 
- the display orientation
- the light level of of the display

The associated three options are available through Tuya / gw / app:

[image]

All the other settings are working with the converter beside the three previous ones: 

[image]
[image]
[image]

Thank you in advance for your help!

PedroKTFC, 2024-10-30:
This is what I get in ZHA. Clearly, the mapping for the equivalent ZHA quirk needs further work. Unfortunately the quirk is complicated so working out what needs to change will take some time.

[image]

canope-25, 2024-10-31:
I'm trying to extend the Tuya Ideveloper OT Core license hoping to be able to access to more details regarding the device but seems difficult .
What can I do in order to help improve the features coverage with zb2mqtt ?

canope-25, 2024-10-31:
Hi ,

I will try ( this evening ) the js template available at this link :

https://github.com/DonaldChung-HK/z2m-moes-trv-801-converter/blob/main/moes_valve_ZTRV801.js

Seems to be more complete

closed 2024-12-31 by egarijo

PedroKTFC, 2024-12-31:
I'm afraid I've moved over to Zigbee2MQTT for this device where support is more up to date (but for a completely different reason).