Koenkk/zigbee2mqtt #23215: IKEA Firmware for E1757 FYRTUR roller blinds (24.4.13)

tda90, 2024-06-30:
### What happened?

Hello team,
After upgrading to the firmware 24.4.13 of the FYRTUR Blinds E1757 it have stopped to work with ZIgbee2MQTT.
If I repair the blinds - it works for 1 time, and then it stops.
I've already tried as suggested to set the genPollCtrl to 172800 but without luck.

I'm not using the signal repeater - because before the upgrade it was working without problems.
Also I didn't fully understand how to pair it (https://www.zigbee2mqtt.io/devices/E1757.html) The guide is a little bit unincomprible to me.
Thanks a lot for the help.

### How to reproduce it (minimal and precise)

Upgrade to 24.4.13

### Zigbee2MQTT version

1.38.0 

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Docker Container

### Debug log

[2024-06-30 21:13:31] error:    z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'TendaCamera' (0x6c5cb1fffe716be7) failed: Error: Interview failed because can not get active endpoints ('0x6c5cb1fffe716be7')'
[2024-06-30 21:13:31] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{},"error":"interview of 'TendaCamera' (0x6c5cb1fffe716be7) failed: Error: Interview failed because can not get active endpoints ('0x6c5cb1fffe716be7')","status":"error","transaction":"tw57q-6"}

closed 2024-07-01 by tda90

frolswe, 2024-07-06:
I arrived here when googling for pairing issues with this device. How did you solve your problem?

alex3305, 2024-08-25:
@tda90 Could you provide your solution?