Koenkk/zigbee2mqtt #23217: TS0601 _TZE204_laokfqwu shows as not supported

evergreen0057, 2024-07-01:
### What happened?

Problem description

I bought the following device (Wall Mount ZigBee version), it shows up in HA but shows as unsupported. 
https://www.aliexpress.com/item/1005004705672138.html?spm=a2g0o.order_list.order_list_main.23.3fa318028sysPI#nav-review

Device type = Router
Zigbee Model = TS0601
Zigbee Manufacturer =  _TZE204_laokfqwu
Support status = Null [(how_to_add_support)](https://www.zigbee2mqtt.io/advanced/support-new devices/01_support_new_devices.html)

I have tried a few different versions of the following js files but none of them seem to work with this mmWave sensor. 
https://github.com/Koenkk/zigbee2mqtt/issues/17483
https://github.com/Koenkk/zigbee2mqtt/issues/18677

### What did you expect to happen?

The device with the new manufacturer ID workes loke the others.

### How to reproduce it (minimal and precise)

Occurs while adding the device

### Zigbee2MQTT version

1.38.0-1

### Adapter firmware version

20230529113024

### Adapter

usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230529113024-if00

### Setup

HAOS

AleksandKorchagin, 2024-07-09:
TS0601 _TZE204_laokfqwu shows as not supported. I bought this sensor. The sensor is not detected. Help me solve the problem with the sensor. 
Thank you.

evergreen0057, 2024-07-10:
Hey @AleksandKorchagin, I reached out to the seller that I bought my device from and they pointed me to this Github link (https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt/blob/main/wenzhi_tuya_M100_240704.js) and I imported it and now the sensor works as expected

AleksandKorchagin, 2024-07-10:
Thanks, I'll try it tonight. I’ll write later whether it works for me.

Puebllo, 2024-07-10:
It work for mine sensor too (TS0601 _TZE204_laokfqwu)

AleksandKorchagin, 2024-07-10:
2024-07-10 22:56:02zh:controller: Interview for '0xa4c138c6fa77f431' started
Info 2024-07-10 22:56:02z2m: Device '0xa4c138c6fa77f431' joined
Info 2024-07-10 22:56:02z2m: Starting interview of '0xa4c138c6fa77f431'
Info 2024-07-10 22:56:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138c6fa77f431","ieee_address":"0xa4c138c6fa77f431"},"type":"device_joined"}'
Info 2024-07-10 22:56:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138c6fa77f431","ieee_address":"0xa4c138c6fa77f431","status":"started"},"type":"device_interview"}'
Info 2024-07-10 22:56:02zh:controller: Succesfully interviewed '0xa4c138c6fa77f431'
Info 2024-07-10 22:56:02z2m: Successfully interviewed '0xa4c138c6fa77f431', device has successfully been paired
Warning 2024-07-10 22:56:03z2m: Device '0xa4c138c6fa77f431' with Zigbee model 'TS0601' and manufacturer name '_TZE204_laokfqwu' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2024-07-10 22:56:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138c6fa77f431","ieee_address":"0xa4c138c6fa77f431","status":"successful","supported":null},"type":"device_interview"}'

It does not work for me ((( What could be the problem ?

AleksandKorchagin, 2024-07-10:
> Hey @AleksandKorchagin, I reached out to the seller that I bought my device from and they pointed me to this Github link (https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt/blob/main/wenzhi_tuya_M100_240704.js) and I imported it and now the sensor works as expected

Are you correct in the code?

Puebllo, 2024-07-11:
did you added to z2m configuration external converter ? Here is example from me. In the same directory as z2m configuration you should add your file with converter and restart add-on

```
external_converters:
  - ext_converter.js
```

AleksandKorchagin, 2024-07-11:
> ext_converter.js

I add the file ext_converter.js to the folder with the 2mqtt application. I am adding an external convector ext_converter.js to the 2mqtt application. I receive a message from 2mqtt that an external convector has been added. I'm pre-launching HA. 

The sensor is not defined in 2mqtt ((((

Puebllo, 2024-07-11:
In z2m is showing your sensor as supported ? after applying external converter it should look similiar to mine

[image]

AleksandKorchagin, 2024-07-11:
Earned thanks. Tell me the motion sensor is added to the HA as sensor or binary.sensor ?

Koenkk, 2024-07-12:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/22713

closed 2024-07-12 by Koenkk