Koenkk/zigbee2mqtt #23226: Z2M network unstable and offline device randomly

samuele2723, 2024-07-01:
### What happened?

Hello, since a week or so, i'm seeing random device unavailability in home assistant through MQTT integration of Z2M.
I think at the same time, i updated the original firmware original on the sonoff pen 2021ish with the latest 2023 Koenkk version

SETUP: Z2M on HAOS with MQTT
[image]

When going to troubleshoot, seems that device is also offline in Z2M and get a lot of "ping error" message, nevertheless when i try to use them from Z2m web interface device is initially unresponsive, but then start works and gets back online in HA.

It seems that anyway, there is some "lag" for this responsiveness from Z2M web interface.

--> tried restart Z2M
--> tried reboot HA instance
--> tried to change my Sonoff dongle (i have two same and same firmware latest from Koenkk)

It also seems that LQI signal is lower than usual, for some reason, still i did not physically change anything in my setup since long time abut position of devices etc.

Quick look on the mesh, doesn't look weak to me

[image]

still, some of the problematic devices seems connected with "none of the above" not sure this helps troubleshooting
[image]

here some example of device messages:

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20230507

### Adapter

Zigbee 3.0 USB Dongle Plus

### Setup

Home assistant OS with MQTT and Z2m Addon

### Debug log

ERROR NO LQI

`2024-07-01 20:54:50Failed to execute LQI for 'Termostato Bagno'
2024-07-01 20:54:50Failed to execute LQI for 'Fari Giardino'
2024-07-01 20:54:50Failed to execute LQI for 'GU10 Sterlizia Giardino'
2024-07-01 20:54:50Failed to execute LQI for 'GU10 4D Perimetrale'
2024-07-01 20:54:50EventBus error 'OTAUpdate/deviceMessage': CommandResponse 0x60a423fffe025210/11 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":2720,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))
2024-07-01 20:54:50Failed to execute LQI for 'Controller Bagno Specchio'
2024-07-01 20:54:50Failed to execute LQI for 'Controller Lavanderia Vasca'
2024-07-01 20:54:50Failed to execute LQI for 'GU10 Mansarda 8'
2024-07-01 20:54:50Failed to execute LQI for 'GU10 Bagno 5'
2024-07-01 20:54:50Failed to execute LQI for 'GU10 Bagno 3'
2024-07-01 20:54:50Failed to execute LQI for 'Controller Cameretta Armadio'
2024-07-01 20:54:50Failed to execute LQI for 'Switch Cameretta Elicottero'
2024-07-01 20:54:50Failed to execute LQI for 'Controller Mansarda Scrivania'
2024-07-01 20:54:50Failed to execute LQI for 'GU10 Mansarda 4'
2024-07-01 20:55:22Failed to execute LQI for 'GU10 Bagno 2'`

WARNING
WARNING NO PING ROUTE
`2024-07-01 20:35:36Failed to ping 'Fari Soppalco' (attempt 2/2, ZCL command 0x00124b0024c5d551/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
2024-07-01 20:38:34Failed to ping 'GU10 Aromatiche Giardino' (attempt 1/1, ZCL command 0xbc33acfffe8b91f5/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
2024-07-01 20:38:39Failed to ping 'Dimmer Soppalco' (attempt 1/1, ZCL command 0x086bd7fffe5cf9a6/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
2024-07-01 20:38:46Failed to ping 'Presence Tuya' (attempt 1/1, ZCL command 0xa4c138e5b7df5157/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
2024-07-01 20:38:52Failed to ping 'Presa Renne' (attempt 1/1, ZCL command 0x842e14fffe161e2c/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))`

WARING PING TIMEOUT

`2024-07-01 20:54:50Failed to ping 'GU10 Mansarda 3' (attempt 1/2, ZCL command 0x680ae2fffe3b25f6/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49427 - 1 - 228 - 0 - 1 after 10000ms))
2024-07-01 20:54:50Failed to ping 'Fari Soppalco' (attempt 1/2, ZCL command 0x00124b0024c5d551/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 1011 - 1 - 232 - 0 - 1 after 10000ms))
2024-07-01 20:54:50Failed to ping 'GU10 Mansarda 8' (attempt 1/2, ZCL command 0x50325ffffe454113/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 44502 - 1 - 236 - 0 - 1 after 10000ms))
2024-07-01 20:54:50Failed to ping 'GU10 Mansarda 4' (attempt 1/2, ZCL command 0xbc33acfffe755f15/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 18037 - 1 - 2 - 0 - 1 after 10000ms))
2024-07-01 20:54:50Failed to ping 'GU10 Bagno 5' (attempt 1/2, ZCL command 0xbc33acfffe82954a/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 47745 - 1 - 8 - 0 - 1 after 10000ms))
2024-07-01 20:54:50Failed to ping 'GU10 Mansarda 2' (attempt 1/2, ZCL command 0x588e81fffe674205/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 62921 - 1 - 23 - 0 - 1 after 10000ms))
2024-07-01 20:54:50Failed to ping 'GU10 Sterlizia Giardino' (attempt 1/2, ZCL command 0xa4c13816572cf0cc/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33206 - 1 - 30 - 0 - 1 after 10000ms))
2024-07-01 20:54:50Failed to ping 'GU10 Mansarda 7' (attempt 1/2, ZCL command 0x842e14fffe64c5c3/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 7738 - 1 - 50 - 0 - 1 after 10000ms))
2024-07-01 20:54:50Failed to ping 'GU10 Mansarda 5' (attempt 1/2, ZCL command 0x588e81fffe69265b/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction" [... cut ...] nceNumber":null,"writeUndiv":false}) failed (Timeout - 32065 - 1 - 214 - 0 - 1 after 10000ms))
2024-07-01 21:03:45Failed to ping 'Presa Forno' (attempt 1/1, ZCL command 0x842e14fffe0fc31c/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33208 - 1 - 227 - 0 - 1 after 10000ms))`

howardtopher, 2024-07-02:
I had the same issue today.  Same setup, same dongle, but dongle has the older firmware version.  I've been using this setup for a while and never had any problems until today I got a text from my wife, "The house is broken."

I tried a few things.
1. Restarted zigbee2mqtt addon - didn't fix it.
2. Installed waiting upgrade to 1.39.0 - didn't fix it.
3. Rebooted HAOS - didn't fix it.
4. Turned off HAOS, unplugged the dongle for 15 minutes - seems ok for now

I've lost my logs since I rebooted and it's been ok for now, but I was receiving the same error messages as above

Failed to execute LQI for <device>
No route to device

----------------

Zigbee2MQTT version
Started with 1.38.0, upgraded to 1.39.0

Adapter firmware version
20220219

Adapter
Zigbee 3.0 USB Dongle Plus

Setup
Home assistant OS with MQTT and Z2m Addon

samuele2723, 2024-07-02:
Thank you for the reply Howard. Yes definitely weird to me as well because looks like network or interference but i can't explain

I just tried hard reboot as you suggested, but i have two dongles also with different firmware and same behaviour and similar devices shows offline. Can't be hardware or dongle related in my opinion.

HA version? I have 2024.6.3 

Next step for testing will be move temporary the hardware to see if different position in the houses i get changes in the result, but is more to troubleshoot than a solution.

I have no experience to analyze Z2M log maybe let's see if someone else reply and knows where to look

samuele2723, 2024-07-03:
hello, i moved my Z2M machine but seems not solving. Still i get ping error and no-route errors even from devices very close to my bridge

samuele2723, 2024-07-06:
Update, i solved totally the issue doing following:

- firmware 20221226
- downgrade z2m to version 1.36
- removing Awox lamps (i had 3'of them recently)

I hope this helps someone else, feeling is that problem were the 3 "EGLO" lamps which are reported as Awox manufacturer

closed 2024-12-28 by github-actions[bot]