Koenkk/zigbee2mqtt #23229: Deprecated package warnings

thargy, 2024-07-02:
### What happened?

When running `update.sh` I get the following warnings:

```bash
npm warn deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
npm warn deprecated @humanwhocodes/config-array@0.11.14: Use @eslint/config-array instead
npm warn deprecated rimraf@3.0.2: Rimraf versions prior to v4 are no longer supported
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated @humanwhocodes/object-schema@2.0.3: Use @eslint/object-schema instead
```

The first of which seems pretty important, as it suggests a potential memory leak.

### What did you expect to happen?

No warnings on updating.

### How to reproduce it (minimal and precise)

Run `update.sh`

### Zigbee2MQTT version

 1.39.0 commit: 03269267

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (CC2652P)

### Setup

Installed manually on Rasberry Pi

ghoz, 2024-07-05:
Same as https://github.com/Koenkk/zigbee2mqtt/issues/23238
These are indirect dependencies, last time I checked they dev / build only : eslint and jest...
slightly less worse I guess

thargy, 2024-12-20:
Ping to keep open.

Koenkk, 2024-12-21:
Do you still get these with the current dev branch?

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2025-03-01 by github-actions[bot]