Koenkk/zigbee2mqtt #23230: [Wrong device]: Moes ZB-LZD10-RCW detected as Tuya TS0505B_1

wraeuk, 2024-07-02:
### Link

https://www.aliexpress.com/item/1005004382991985.html

### Model

ZB-LZD10-RCW

### Description

10W RGB+CCT Smart Downlight

### Vendor

Moes

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/ZB-LZD10-RCW.png

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0xa4c1387c2a278a83","nwkAddr":4512,"manufId":4107,"manufName":"_TZ3210_jjqdqxfq","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,6280,4,5,6,8,768,10,4096,65518],"outClusterList":[25,4096,65518],"clusters":{"genLevelCtrl":{"attributes":{"61440":1000}},"genBasic":{"attributes":{"hwVersion":0}},"lightingColorCtrl":{"attributes":{"57344":1000,"57345":[1000,65536113],"61453":1,"colorCapabilities":8,"colorTempPhysicalMin":143,"colorTempPhysicalMax":500,"tuyaRgbMode":0}},"genOnOff":{"attributes":{"61440":0,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"z.1.0","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1719886938850}

### Notes

Curiously I had one of the 15 lights correctly detect, despite all being bought at the same time (and looking physically identical.) Removing and repairing any of the incorrectly detected ones goes back to TS0505B_1, I haven't been game to re-pair the correctly detected one, but it's definitely detected incorrectly in the past.

The correctly detected device has the following db entry
{"id":11,"type":"Router","ieeeAddr":"0xa4c1386bac72e5b4","nwkAddr":22013,"manufId":4417,"manufName":"_TZ3210_s9lumfhn","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_s9lumfhn","powerSource":1,"zclVersion":3,"appVersion":101,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":2,"currentX":8880,"currentY":2615,"colorTemperature":168,"currentHue":170,"currentSaturation":254}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":39}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":101,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1719886887988}

Koenkk, 2024-07-02:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-02 by Koenkk

Koenkk, 2024-07-03:
> The "correct" device has working brightness adjustment in color_xy whereas the others do not

I don't understand fully, can you explain in more detail what the issue is?

wraeuk, 2024-07-03:
Yep, sorry.

The devices in question are physically identical.
The device with "manufName":"_TZ3210_s9lumfhn" is working as expected
The device with "manufName":"_TZ3210_jjqdqxfq" mostly works except when changing to color_xy, where the brightness adjustment doesn't do anything (stays at 100%, or goes to off at 0%). Brightness adjustment works fine when using color_temp.

I logged a new device support issue for this ages ago and never got around to looking at it harder (it aged out) https://github.com/Koenkk/zigbee2mqtt/issues/17570 which has some info from a tuya gateway if that helps make sense of it.

If this helps, the generated external converters for each

**_TZ3210_s9lumfhn**
const {identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0505B'],
    model: 'TS0505B',
    vendor: '_TZ3210_s9lumfhn',
    description: 'Automatically generated definition',
    extend: [identify(), light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"]}})],
    meta: {},
};

module.exports = definition;

**_TZ3210_jjqdqxfq**
const {identify} = require('zigbee-herdsman-converters/lib/modernExtend');
const {philipsLight} = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['TS0505B'],
    model: 'TS0505B',
    vendor: '_TZ3210_jjqdqxfq',
    description: 'Automatically generated definition',
    extend: [identify(), philipsLight({"color":true})],
    meta: {},
};

module.exports = definition;

Koenkk, 2024-07-04:
with the tuya gateway, did the brightness change when color_xy was used?

wraeuk, 2024-07-05:
Yep, sure did.

Koenkk, 2024-07-06:
From what I understand:
- `_TZ3210_s9lumfhn` -> uses standard commands
- `_TZ3210_jjqdqxfq` -> is controlled through datapoints?

wraeuk, 2024-07-07:
That does seem to be the case. Here's a full listing of the datapoints from the tuya portal if that helps any

{
    "1": "ON/OFF",
    "2": "Mode",
    "209": "Cycle Timing",
    "210": "Vacation Timing",
    "3": "Brightness",
    "30": "Rhythms",
    "31": "Light to sleep",
    "32": "Light to wake",
    "33": "Power Off Memory",
    "34": "Do not Disturb",
    "35": "ON/OFF Gradient",
    "4": "Color Temp",
    "5": "Colorful",
    "6": "Scene",
    "7": "Timer",
    "8": "Music",
    "81": "zigbee light link threshold",
    "9": "Adjust"
}

The "Brightness" data point does *not* change when adjusting brightness when using the "Color" mode (dp2)

When set to the "white" mode (dp2) the Color Temp (dp4) and Brightness (dp2) points work, and are values from 0-1000

I'm not sure if this is standard, as I haven't looked at _TZ3210_s9lumfhn  on the tuya gateway, but I can if that would help.

Koenkk, 2024-07-08:
Can you try to write a simple external converter using datapoints with e.g. just the `state`? You can find an example here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/1fe26bb1ed52f10ab7c7a7b219d500476f852958/src/devices/tuya.ts#L2001

eerison, 2024-08-23:
@Koenkk 

I saw that you fixed some wrong model detection changing this "magical number/reference", how do I know what is the correct one for my device? I am stuck on this 🥲🥲.

husnainali09, 2025-10-21:
These lights work very inconsistently with Z2M. Is this a known issue ?

husnainali09, 2025-10-21:
@wraeuk Did you manage to get these lights to work properly ?

wraeuk, 2025-10-21:
I never got around to making colour + brightness work, but I don't have any issues with temp + brightness which is all I really use it for. If you're having other issues with standard operation I would guess it's another revision of the light which changes something inexplicably.

pranjal-joshi, 2026-01-30:
Ordered bunch of these ZB-LZD10-RCW. Curious to know if they act as a zigbee router as well because they're always mains powered. 
@wraeuk @FVerhamme can you guys shed light on this?