Koenkk/zigbee2mqtt #23232: after updated to 1.39.01 the bug #22258 Fix Entity 'homeassistant/sensor' is unknown (@Koenkk) seems to be still present

arripeo, 2024-07-02:
### What happened?

zigbee2mqtt try to start ricursively then stop the log say that the Mqtt base topic could not be the same of homeassintant.discovery.
At now all the two value are "homeassistant"

### How to reproduce it (minimal and precise)

simply update new version on HA 2024.6.4 or on Hassio with OS 12.3

### Zigbee2MQTT version

1.39.01

### Adapter firmware version

20211210

### Adapter

Slaesh Cc2652rb-stick

### Setup

Add-on on Homeassistant

### Debug log

09:34:55] INFO: Preparing to start...
[09:34:55] INFO: Socat not enabled
[09:34:56] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-07-02 09:34:59] info: z2m: Logging to console, file (filename: log.log)
Error: 'homeassistant.discovery_topic' cannot not be equal to the 'mqtt.base_topic' (got 'homeassistant')
at new HomeAssistant (/app/lib/extension/homeassistant.ts:191:19)
at new Controller (/app/lib/controller.ts:124:45)
at start (/app/index.js:152:18)
[09:34:59] INFO: Preparing to start...
[09:35:00] INFO: Socat not enabled
[09:35:00] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-07-02 09:35:03] info: z2m: Logging to console, file (filename: log.log)
Error: 'homeassistant.discovery_topic' cannot not be equal to the 'mqtt.base_topic' (got 'homeassistant')
at new HomeAssistant (/app/lib/extension/homeassistant.ts:191:19)
at new Controller (/app/lib/controller.ts:124:45)
at start (/app/index.js:152:18)
[09:35:04] INFO: Preparing to start...
[09:35:04] INFO: Socat not enabled
[09:35:05] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-07-02 09:35:07] info: z2m: Logging to console, file (filename: log.log)
Error: 'homeassistant.discovery_topic' cannot not be equal to the 'mqtt.base_topic' (got 'homeassistant')
at new HomeAssistant (/app/lib/extension/homeassistant.ts:191:19)
at new Controller (/app/lib/controller.ts:124:45)
at start (/app/index.js:152:18)

arripeo, 2024-07-02:
Solved in add-on site

closed 2024-07-02 by arripeo

lemanjo, 2024-07-03:
What does "Solved in add-on site" exactly mean? How did you solve this?
I'm having the same issue.

lemanjo, 2024-07-03:
Yeah, I restored the older version and update it later on.

arripeo, 2024-07-03:
Solved in add-site mean that in dedicated GitHub for hassio add-on I've solved. Anyway I changed the mqtt.base_topic to value "zigbee2mqtt" and I've maintained "homeassistant" value in homeassistant.discover field. Then I've restarted the z2m add-on. Pay attention that you will loose the mqtt devices and entity in HA. So after the restart of add-on if the add-on works make also a full reboot of your host then all the entities and devices will be re-descovered again

akeck, 2024-07-03:
I have now the problem, that 1.39 gets installed without my doing and ofc stops working again. A way to ignore the update is not available, so I have to restore 1.38 a few times a day...
Please release 1.39.1...

Koenkk, 2024-07-03:
To fix this, remove `base_topic: homeassistant` from the addon configuration page

whizzoo, 2024-09-09:
> To fix this, remove `base_topic: homeassistant` from the addon configuration page

This does not fix the issue for me, I get the same error message in the log. The only thing that works is changing the base_topic, but this causes new entities to be added for most of my devices. Is there a way to avoid this?

martijn1247812, 2024-09-18:
> This does not fix the issue for me, I get the same error message in the log. The only thing that works is changing the base_topic, but this causes new entities to be added for most of my devices. Is [...]

In my case I run  both HA and zigbee2mqtt in docker; I removed 'base_topic: homeassistant' from configuration.yaml in my zigbee2mqtt docker (data folder). I used the '#'