Koenkk/zigbee2mqtt #23233: Develco HESZB-120 high battery drain

klickfisch, 2024-07-02:
### What happened?

I'm using a HESZB-120 in my kitchen. The device itself work pretty solid, but the battery drain is high. The estimated lifetime should be 5 years, but mine will last only 3-4 months.
The device has a high lqi with over 150 and also enough router nearby.

I also contacted the Develco support:
"We received several similar observations regarding this device on the Zigbee2MQTT driver.
 I would suspect the driver to be pulling data from the device too often, which can cause increased battery usage.
 Another reason here could be a network – as if the device loses connection to the gateway – it will constantly search for a network to join.
 Such behavior will naturally, and significantly increase power consumption.
 You could try to move the device closer to the gateway for the test."

### What did you expect to happen?

Not to change the battery every 3-4 months.

### How to reproduce it (minimal and precise)

Removed the device, updated the firmware, rebooted all.
I'm using also Develco SMSZB-120 wito no problems.

### Zigbee2MQTT version

1.39.0 and earlier

### Adapter firmware version

20230507

### Adapter

CC2652 - HamGeek POE Zigbee 3.0

### Setup

HA OS on Synology VM

sjorge, 2024-07-04:
Mine drained the battery like crazy after doing the OTA last month. Removing and repairing fixed that for me.
It seemed based on my wireshard dump that they somehow lost the TC key and would continuesly drop, rejoin the network. After a remove, restart z2m, join that seems to have fixed it.

klickfisch, 2024-07-05:
I repaired after the last battery change, will keep an eye what happend now with a new battery.

JanDetrez, 2024-09-03:
I have exactly the same issue. recording the logs, I see that the  heat sensor is logging a lot more often than the smoke sensor I have from the same brand. Will also try the repairing suggested here...

See "Hittemelder" (= Heat sensor) reporting every 30 s, while the "rookmelder" (= Smoke sensor) reporting every 5 min with same reporting settings

> info 2024-09-03 14:22:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Rookmelder verdiep', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":159,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":23.93,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":3000,"zone_id":23}'
info 2024-09-03 14:22:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Hittemelder berging "},"type":"device_announced"}'
info 2024-09-03 14:22:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hittemelder berging ","ieee_address":"0x0015bc0034001b27"},"type":"device_announce"}'
info 2024-09-03 14:22:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hittemelder berging ', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":66,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":24.43,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":3100,"zone_id":23}'
info 2024-09-03 14:23:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Hittemelder berging "},"type":"device_announced"}'
info 2024-09-03 14:23:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hittemelder berging ","ieee_address":"0x0015bc0034001b27"},"type":"device_announce"}'
info 2024-09-03 14:23:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hittemelder berging ', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":66,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":24.43,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":3100,"zone_id":23}'
info 2024-09-03 14:23:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Hittemelder berging "},"type":"device_announced"}'
info 2024-09-03 14:23:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hittemelder berging ","ieee_address":"0x0015bc0034001b27"},"type":"device_announce"}'
info 2024-09-03 14:23:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Hittemelder berging "},"type":"device_announced"}'
info 2024-09-03 14:23:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hittemelder berging ","ieee_address":"0x0015bc0034001b27"},"type":"device_announce"}'
info 2024-09-03 14:23:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hittemelder berging ', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":63,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":24.43,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":3100,"zone_id":23}'
info 2024-09-03 14:24:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Hittemelder berging "},"type":"device_announced"}'
info 2024-09-03 14:24:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hittemelder berging ","ieee_address":"0x0015bc0034001b27"},"type":"device_announce"}'
info 2024-09-03 14:24:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hittemelder berging ', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":63,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":24.43,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":3100,"zone_id":23}'
info 2024-09-03 14:24:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Hittemelder berging "},"type":"device_announced"}'
info 2024-09-03 14:24:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hittemelder berging ","ieee_address":"0x0015bc0034001b27"},"type":"device_announce"}'
info 2024-09-03 14:24:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hittemelder berging ', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":63 [... cut ...]
info 2024-09-03 14:27:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hittemelder berging ', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":63,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":24.43,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":3100,"zone_id":23}'
[image]
[image]

JanDetrez, 2024-09-03:
After trying to re-pair I keep getting an interview error for some reason. Not sure if this was the case before, I supposed the interview completed last time
```
2024-09-03 14:59:32zh:controller: Interview failed for '0x0015bc0034001b27 with error 'Error: Interview failed because of failed IAS enroll (zoneState didn't change ('0x0015bc0034001b27')'
2024-09-03 14:59:32z2m: Failed to interview 'Hittemelder berging', device has not successfully been paired
```

JanDetrez, 2024-09-03:
So after a factory reset, repairing, the interview completed, and the update frequency decreased, hopefully solving the draining issue... let's see how it goes.

```

info 2024-09-03 16:01:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hittemelder berging', payload '{"alarm":null,"battery":null,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":69,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":25.31,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":null,"zone_id":23}'
info 2024-09-03 16:01:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Rookmelder zolder', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":48,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":25.43,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":3000,"zone_id":23}'
info 2024-09-03 16:02:16z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Rookmelder verdiep', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":48,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":24.06,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":3000,"zone_id":23}'
info 2024-09-03 16:06:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hittemelder berging', payload '{"alarm":null,"battery":null,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":66,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":25.31,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":null,"zone_id":23}'
info 2024-09-03 16:06:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Rookmelder zolder', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":39,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":25.43,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":3000,"zone_id":23}'
info 2024-09-03 16:07:05z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Rookmelder verdiep', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":159,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":24.06,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":3000,"zone_id":23}'
```

ok-67, 2024-09-25:
I have the same problem. The battery drains from 100  - 60 % in 1 - 2 months. And then the unit starts giving short beeps to indicate empty battery, but the battery is reported to be at 60 %. I changed the battery just a few days ago and now it is down to 80 %

The unit is located at my holliday house, will try to repair it next time I am there and see what happens.

rommess, 2024-10-07:
repairing / interview / reconfigure and so on
same issue, high battery drain

Rescla, 2024-10-18:
I had the same issue. One of the behaviors I observed was that when I tried to reconfigure the temperature reporting interval it would create a new record, instead of modifying the existing record. I also saw the device pushing updates about every minute.

After resetting (remove from z2m, factory reset, re-pair) I can reconfigure the reporting entries and the reporting interval is around 5 minutes. Perfect!

[image]

However, as you can see in the screenshot some values are not reported correctly, including battery percentage and latest/installed version. When I try to manually add the temperature on endpoint 35 I get the following error:

`Bind 0x0015bc0034001876/35 msTemperatureMeasurement from '0x00124b002e0e1373/1' failed (Status 'NOT_SUPPORTED')
z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0x0015bc0034001876/35 msTemperatureMeasurement from '0x00124b002e0e1373/1' failed (Status 'NOT_SUPPORTED')'`

JanDetrez, 2024-10-18:
I also recognise the battery not being returned as a percentage, but rather it says this: 

'{"alarm":null,"battery":null,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b0025799787","linkquality":27,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":22.31,"test":false,"update":{"installed_version":262152,"latest_version":262152,"state":"idle"},"update_available":false,"voltage":null,"zone_id":23}'

[image]

rindlerblabla, 2024-10-20:
I am experiencing the same. However my device is spamming with announce messages.

```
info 2024-10-20 17:23:14z2m: Accepting joining not in blocklist device '0x0015bc0034000eaa'

info 2024-10-20 17:23:14z2m: Accepting joining not in blocklist device '0x0015bc0034000eaa'

info 2024-10-20 17:23:17z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"brandvarnare/pannrummet","ieee_address":"0x0015bc0034000eaa"},"type":"device_announce"}'

info 2024-10-20 17:23:46z2m: Accepting joining not in blocklist device '0x0015bc0034000eaa'

info 2024-10-20 17:23:46z2m: Accepting joining not in blocklist device '0x0015bc0034000eaa'

info 2024-10-20 17:23:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"brandvarnare/pannrummet","ieee_address":"0x0015bc0034000eaa"},"type":"device_announce"}'
```

closed 2025-02-15 by github-actions[bot]

rommess, 2026-01-14:
agree with @fsedarkalex , about 6 month since battery replaced and still reporting 100% (alerte of the device itself around 63%)

rommess, 2026-01-14:
expect this firmware 

[image]