Koenkk/zigbee2mqtt #23235: All devices are lost after Z2M update

paavokurkela, 2024-07-02:
### What happened?

After add-on update on Home Assistant, all the devices are lost from Z2M and Homa Assistant (no devices listed under MQTT instance)

Restoring a backup restores devices on Home Assistant  but not on Z2M. Same issue happened to me also with earlier update, unfortunately I don't have any details on that. Log is literally empty.

### How to reproduce it (minimal and precise)

Update Z2M

### Zigbee2MQTT version

1.38.0-1

### Adapter firmware version

20230507

### Adapter

Slaesh 

### Setup

Add-on under HA on Intel Nuc

kligithu, 2024-07-02:
The same here

**What did you expect to happen?**
No response

**How to reproduce it (minimal and precise)**
Update Z2M

**Zigbee2MQTT version**
1.38.0-1 to 1.39.0-1

**Adapter firmware version**
core v2.3.6 : Zigbee 20240315

**Adapter**
SMLIGHT SLZB-06

**Setup**
Add-on under HA on Synology VMM

**Debug log**
No response

raffoul, 2024-07-02:
Same. Rollback done.

Webmonsteril, 2024-07-02:
Have same issue rolled back also

Najihel, 2024-07-02:
Hello,

Pretty same issue here, devices are not missing but all entities are duplicated. The original one is broken and Z2M create duplicate suffixed by `_2`

[image]

[image]

jmcollin78, 2024-07-02:
Same here ! Rollback running. What is strange is that the duplication of entities occurs after a full restart of HA and not just after the update to 1.39.0.

Najihel, 2024-07-03:
> Same here ! Full rollback running. What is strange is that the duplication of entities occurs after a full restart of HA and not just after the update to 1.39.0.
> 
> This is a severe issue and 1.39.0 deployment should be aborted if you want my opinion.
> 
> Edit: the full restor have restored the entities.

I confirm the triger is the restart of HA on my side.

gijsh070, 2024-07-03:
Same issue here, double sensors en entities. They even remain double after restoring.....

countrr, 2024-07-03:
Same here, had to roll back to 1.38.0. Will not update again until this is solved.

Zver5, 2024-07-03:
Same here, had to roll back to 1.38.0. Will not update again until this is solved.

raphmur, 2024-07-04:
Had same issue updating today to 1.39.0. Rolling back to 1.38.0
What kind of log would you need? How to support?

kefrens, 2024-07-04:
I have same issue; How do you rollback zigbee2mqtt?

raphmur, 2024-07-04:
@kefrens in home assistant settings> system> backups
Then select the backup that was normally done at z2m update and backup. Then reboot and you should be good

Vidjer, 2024-07-05:
Hello have same issue. rollback 1.38 ok

paavokurkela, 2024-07-05:
In my case rollback to 1.38 did not restore devices to Z2M.

raphmur, 2024-07-05:
@paavokurkela , did you restart HA?

jmcollin78, 2024-07-05:
> In my case rollback to 1.38 did not restore devices to Z2M.

I have to do a full restore and not only a rollback on Z2M 1.38

gijsh070, 2024-07-05:
> > In my case rollback to 1.38 did not restore devices to Z2M.
> 
> 
> 
> I have to do a full restore and not only a rollback on Z2M 1.38

Same here, full restore needed

paavokurkela, 2024-07-05:
@raphmur Yes I did.  I also tried full restore from back up. That restored entities on HA but Z2M devices did not restore.

Koenkk, 2024-07-07:
Could you:

- Provide your z2m `configuration.yaml` and if applicable, screenshot of the Z2M addon config page
- The debug log when starting z2m 1.39.0
- The version of HA that you are using

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

paavokurkela, 2024-07-07:
@Koenkk thank you for looking on this.

As I restored my system I currently have 1.38 installed.
I was able to reproduce the problem by pairing two Ikea lights and upgrade from 1.38 to 1.39.

Please note that this very same problem existed already  earlier. I believe it was from 1.37 to 1.38 and it was first time this problem happened to me.

Please find infos requested below:

### **Couple devices paired to 1.38**
[image]

### **Devices gone 1.39**
[image]

### **HA version:**
[image]

### **configuration.yaml (with debugs enabled)**

> data_path: /homeassistant/zigbee2mqtt/
> socat:
>   enabled: false
>   master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
>   slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
>   options: "-d -d"
>   log: true
> mqtt:
>   server: mqtt://core-mosquitto:1883
>   user: ****
>   password: ****
>   client_id: Z2M
> serial:
>   port: /dev/ttyUSB0
> zigbee_herdsman_debug: false
> homeassistant: true
> advanced:
>   channel: 14

  
### **Add-on config screen (credentials  hidden)** 
  [image]
  
### **log (1.38 with two Ikea lights paired)**
  [12:40:58] INFO: Preparing to start...
[12:40:58] INFO: Socat not enabled
[12:40:59] INFO: Starting Zigbee2MQTT...
[2024-07-07 12:41:02] info: 	z2m: Logging to console, file (filename: log.log)
[2024-07-07 12:41:02] info: 	z2m: Starting Zigbee2MQTT version 1.38.0 (commit #unknown)
[2024-07-07 12:41:02] info: 	z2m: Starting zigbee-herdsman (0.49.2)
[2024-07-07 12:41:02] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-07-07 12:41:02] info: 	zh:zstack:znp: Serialport opened
[2024-07-07 12:41:02] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-07-07 12:41:03] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-07-07 12:41:27] info: 	zh:controller: Wrote coordinator backup to '/homeassistant/zigbee2mqtt/coordinator_backup.json'
[2024-07-07 12:41:27] info: 	z2m: zigbee-herdsman started (reset)
[2024-07-07 12:41:27] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20230507,"transportrev":2},"type":"zStack3x0"}'
[2024-07-07 12:41:27] info: 	z2m: Currently 0 devices are joined:
[2024-07-07 12:41:27] info: 	z2m: Zigbee: disabling joining new devices.
[2024-07-07 12:41:27] info: 	z2m:mqtt: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-07-07 12:41:27] info: 	z2m:mqtt: Connected to MQTT server
[2024-07-07 12:41:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-07-07 12:41:28] info: 	z2m: Started frontend on port 8099
[2024-07-07 12:41:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-07-07 12:41:28] info: 	z2m: Zigbee2MQTT started!
[2024-07-07 12:41:33] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x00124b00259bb8b4/connection_state/config', payload '{"device":{"hw_version":"zStack3x0 20230507","identifiers":["zigbee2mqtt_bridge_0x00124b00259bb8b4"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.38.0"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x00124b00259bb8b4_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'

### **log (startup during upgrade process to 1.39 and all devices lost)**
[13:05:07] INFO: Preparing to start...
[13:05:08] INFO: Socat not enabled
[... 26 lines cut ...]
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/select/0x90395efffee6d2b7/identify/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/select/0x287681fffe9cb162/effect/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/select/0x287681fffe9cb162/power_on_behavior/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/select/0x287681fffe9cb162/identify/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x90395efffee6d2b7/power_on_behavior/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x90395efffee6d2b7/color_options/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x90395efffee6d2b7/linkquality/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x90395efffee6d2b7/update_state/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x287681fffe9cb162/power_on_behavior/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x287681fffe9cb162/color_options/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x287681fffe9cb162/linkquality/config', payload 'null'
[2024-07-07 13:05:37] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x287681fffe9cb162/update_state/config', payload 'null'
[... cut ...] 4b00259bb8b4"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.39.0"},"icon":"mdi:human-greeting-proximity","name":"Permit join","object_id":"zigbee2mqtt_bridge_permit_join","origin":{"name":"Zigbee2MQTT","sw":"1.39.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0x00124b00259bb8b4_permit_join_zigbee2mqtt","value_template":"{{ value_json.permit_join | lower }}"}'

lecourtb, 2024-07-09:
Hello, I opened a related issue that may help.
#23294

dcyou, 2024-07-09:
hi guys I had same issue 😞  too but I manage myself by taking time to remove the old entities (without _2) then renaming all the entities by removing the "_2" (+/- 10 devices with 4 entity each) 💪  But I struggle with how to remove the zigbee2mqtt_bridge entities, if someone how it is possible? 🆘

jmcollin78, 2024-07-09:
> Could you:
> 
> * Provide your z2m `configuration.yaml` and if applicable, screenshot of the Z2M addon config page
> * The debug log when starting z2m 1.39.0
> * The version of HA that you are using
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Hello @Koenkk ,

Is there any chance to have a fix in the 1.39.1 release (https://github.com/Koenkk/zigbee2mqtt/pull/23221) ?
It is not listed in the release note.

Koenkk, 2024-07-09:
@paavokurkela in the 1.38.0 log I see `[2024-07-07 12:41:27] info: 	z2m: Currently 0 devices are joined:`. Could you provide the **debug** log of:

- Start z2m 1.38.0
- Pair 2 devices
- Restart z2m 1.38.0
- Upgrade to z2m 1.39.0

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

paavokurkela, 2024-07-10:
@Koenkk thank you for your request.

While gathering logs you requested I found out that devices are gone every time I just simply restart Z2M, no even update needed. Please see log I was able to catch for you below:

Something I want to point out is that while I tried to store log.txt under /homeassistant/zigbee2mqtt/log/%TIMESTAMP% nothing was stored there. Also as log below says: 
`[2024-07-10 08:41:39] info: 	zh:controller: Wrote coordinator backup to '/homeassistant/zigbee2mqtt/coordinator_backup.json'`
the file exists but it's timestamp is for Jan 14th which I believe is date I have installed the system at first place.

Hope this leads us to right direction.

```
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:writer: --> frame [254,5,33,50,1,7,0,2,0,18]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: <-- [254,4,97,50,12,0,0,0,91]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,4,97,50,12,0,0,0,91]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [12,0,0,0] - 91
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvLength - {"len":12}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SREQ: --> SYS - nvRead - {"sysid":1,"itemid":7,"subid":2,"offset":0,"len":12}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:writer: --> frame [254,8,33,51,1,7,0,2,0,0,0,12,18]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: <-- [254,14,97,51,0,12,0,0,0,0,46,65,18,57,193,108,55,244,122]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,14,97,51,0,12,0,0,0,0,46,65,18,57,193,108,55,244,122]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --> parsed 14 - 3 - 1 - 51 - [0,12,0,0,0,0,46,65,18,57,193,108,55,244] - 122
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvRead - {"status":0,"len":12,"value":{"type":"Buffer","data":[0,0,0,0,46,65,18,57,193,108,55,244]}}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SREQ: --> SYS - nvLength - {"sysid":1,"itemid":7,"subid":3}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:writer: --> frame [254,5,33,50,1,7,0,3,0,19]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: <-- [254,4,97,50,12,0,0,0,91]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,4,97,50,12,0,0,0,91]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [12,0,0,0] - 91
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvLength - {"len":12}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SREQ: --> SYS - nvRead - {"sysid":1,"itemid":7,"subid":3,"offset":0,"len":12}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:writer: --> frame [254,8,33,51,1,7,0,3,0,0,0,12,19]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: <-- [254,14,97,51,0,12,0,0,0,0,248,166,248,169,17,156,175,132,249]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,14,97,51,0,12,0,0,0,0,248,166,248,169,17,156,175,132,249]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --> parsed 14 - 3 - 1 - 51 - [0,12,0,0,0,0,248,166,248,169,17,156,175,132] - 249
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvRead - {"status":0,"len":12,"value":{"type":"Buffer","data":[0,0,0,0,248,166,248,169,17,156,175,132]}}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SREQ: --> SYS - nvLength - {"sysid":1,"itemid":7,"subid":4}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:writer: --> frame [254,5,33,50,1,7,0,4,0,20]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: <-- [254,4,97,50,12,0,0,0,91]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,4,97,50,12,0,0,0,91]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [12,0,0,0] - 91
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvLength - {"len":12}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SREQ: --> SYS - nvRead - {"sysid":1,"itemid":7,"subid":4,"offset":0,"len":12}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:writer: --> frame [254,8,33,51,1,7,0,4,0,0,0,12,20]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: <-- [254,14,97,51,0,12,158,144,20,0,255,255,255,255,255,255,255,255,74]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,14,97,51,0,12,158,144,20,0,255,255,255,255,255,255,255,255,74]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --> parsed 14 - 3 - 1 - 51 - [0,12,158,144,20,0,255,255,255,255,255,255,255,255] - 74
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvRead - {"status":0,"len":12,"value":{"type":"Buffer","data":[158,144,20,0,255,255,255,255,255,255,255,255]}}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-10 08:41:39] debug: 	zh:zstack:znp: SREQ: --> SYS - nvLength - {"sysid":1,"itemid":7,"subid":5}
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:writer: --> frame [254,5,33,50,1,7,0,5,0,21]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: <-- [254,4,97,50,0,0,0,0,87]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --- parseNext [254,4,97,50,0,0,0,0,87]
[2024-07-10 08:41:39] debug: 	zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [0,0,0,0] - 87
[... cut ...] ","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x00124b00259bb8b4_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
[2024-07-10 08:43:02] info: 	z2m: Successfully changed options
[2024-07-10 08:43:02] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"fsygo-1"}'
```

lecourtb, 2024-07-11:
Same issue.

I noticed that in Z2M MQTT parameters, the base topic which was "zigbee2mqtt" before upgrade was renamed to "zigbee2mqttHA" after upgrade.

I reproduced two times the problem (thanks to ha backups) and give up try to upgrade !

Later, I also tried a third upgrade with base_topic: zigbee2mqtt in configuration.yaml but without success.

Hope it will help

Koenkk, 2024-07-11:
@lecourtb probably you have set `base_topic: zigbee2mqttHA ` in the HA addon configuration page, this should be set correctly (`zigbee2mqtt` in your case)

Aseg, 2024-07-12:
I have the same issue, I cant update to 1.39 without break my devices and automations because devices duplicate with suffix '_2'

lecourtb, 2024-07-17:
That was effectively the case (`base_topic: zigbee2mqttHA` set in the HA addon configuration page)
I will try to upgrade Z2M as soon as possible after the change.

dystechnic, 2024-07-19:
Same issue here.
Restoring a backup didn't help. I ended up deleting and renaming devices. I'm afraid to reboot HA...

dermotos, 2024-08-02:
Has anyone with this issue upgraded to `1.39.1` yet? If so did this issue persist? I don't see anything in the release notes that suggests this issue has been fixed yet.

thomasknuchel, 2024-08-02:
Tested and issue is still persisting

Wendigogo, 2024-08-02:
You should take a look at #23294 .
I'm pretty sure the given solution will work here.

Enjoy!

Najihel, 2024-08-02:
> You should take a look at #23294 .
> I'm pretty sure the given solution will work here.
> 
> Enjoy!

Yes it is ! Thanks

[image]

[image]

raffoul, 2024-08-04:
Rename base topic to zigbee2mqtt works for me. Thank you.

paavokurkela, 2024-08-04:
I got my system back to working state.

I deleted z2m completely as well all files and folders.
I reinstalled z2m (1.39.1) and used my old configuration. Z2M did not create any folders or files, not even zigbee2mqtt which I created then manually. On my old config I had` /homeassistant/zigbee2mqtt` which I changed to `/config/zigbee2mqtt` which seemed to make a trick and Z2M begun to write files. This is of course very odd as /config is simply a symlink to /homeassistant folder.

This or that my system works again.

jmcollin78, 2024-08-05:
Renaming base_topic to zigbee2mqtt and installing yhe 1.19.1 resolves the issue for me.

Koenkk, 2024-08-05:
Removing the `base_topic` from the Z2M HA addon should fix this indeed.

closed 2024-08-05 by Koenkk

kefrens, 2024-09-05:
For me doesn't work as well.
In the addon config I have:
`base_topic: core-mosquitto`

In the config of Zigbee2MQTT I have as well:
`base_topic: core-mosquitto`

But it still creates all entities with a '_2'

kefrens, 2024-09-06:
> For me doesn't work as well. In the addon config I have: `base_topic: core-mosquitto`
> 
> In the config of Zigbee2MQTT I have as well: `base_topic: core-mosquitto`
> 
> But it still creates all entities with a '_2'

Responding to myself.
1.  I had to delete in the the addon configuration.yaml the base topic.
2.  make sure base topic in the addon and on the zigbee2mqtt website is the same
3.  reboot the system (stop/start wasn't enough)

gijsh070, 2024-09-07:
Thanks a lot @kefrens 

This solved my issue finally. They need to match the addon base_topic and mqtt site.

gijsh070, 2024-09-07:
Anyone noticing that triggers somehow are named differently after the latest update? I get a lot of uknown triggers suddenly after updating.