Koenkk/zigbee2mqtt #23237: [New device support]: It`s a TUYA cover motor,the model is _TZE200_xtrnjaoz

my5123, 2024-07-02:
### Link

https://solution.tuya.com/cn/projects/CMa6rbnjhitahd

### Database entry

don`t know how to find 

### Comments

I tried some external definition, but it doesn`t work.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_xtrnjaoz',
    description: '5',
    fromZigbee: [legacy.fromZigbee.tuya_cover, fz.ignore_basic_report],
    toZigbee: [legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            e.composite('options', 'options', ea.STATE_SET)
                .withFeature(e.numeric('motor_speed', ea.STATE_SET)
                    .withValueMin(0).withValueMax(255).withDescription('Motor speed'))
                .withFeature(e.binary('reverse_direction', ea.STATE_SET, true, false)
                    .withDescription('Reverse the motor direction'))],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(2), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(0)})],
            [2, 'position', tuya.valueConverter.coverPositionInverted],
            [3, 'position', tuya.valueConverter.coverPositionInverted],
            [4, 'opening_mode', tuya.valueConverterBasic.lookup({'tilt': tuya.enum(0), 'lift': tuya.enum(1)})],
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'learning': tuya.enum(2)})],
            [13, 'battery', tuya.valueConverter.raw],
            [101, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],
            [107, 'factory_reset', tuya.valueConverter.setLimit],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

my5123, 2024-07-11:
I`ve got all the DPid from curtain motor maunufatory,cant try the code below.
Now I can get the position pecentage / state / reverse_direction worked, but other funtions still not working.

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: 'Tuya',
    description: '涂鸦控制器+中菱管状电机GM24TF/Z 7',
    whiteLabel: [
        tuya.whitelabel('Shenzhen Golden Security Technology', 'GM46D', 'Curtain motor', ['_TZE200_gnw1rril']),
    ],
    fromZigbee: [tuya.fz.datapoints, legacy.fromZigbee.tuya_cover, fz.ignore_basic_report],
    toZigbee: [tuya.tz.datapoints, legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options],
    exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            e.enum('reverse_direction', ea.STATE_SET, ['forward', 'back']).withDescription('Reverse the motor direction'),
            e.binary('motor_fault', ea.STATE, true, false).withDescription('Motor Fault'),
            e.binary('set_limit', ea.STATE_SET, true, false).withDescription('Set Limit'),
            //e.binary('auto_power', ea.STATE_SET, true, false).withDescription('Auto Power'),
            //exposes.numeric('motor_speed', ea.STATE_SET).withValueMin(0).withValueMax(255).withDescription('Motor speed'),
            ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CONTINUE': tuya.enum(3), 'CLOSE': tuya.enum(2), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(0)})],
            //[6, 'auto_power', tuya.valueConverter.trueFalse1],
            [9, 'position', tuya.valueConverter.coverPosition],
            [8, 'position', tuya.valueConverter.coverPosition],
            [10, 'motor_fault', tuya.valueConverter.trueFalse1],
            [11, 'reverse_direction', tuya.valueConverterBasic.lookup({forward: tuya.enum(0), back: tuya.enum(1)})],
            [101, 'set_limit', tuya.valueConverter.trueFalse1],
            //[105, 'motor_speed', tuya.valueConverter.raw],

        ],
    },
};

module.exports = definition;`
-- | -- | -- | -- | -- | --
1 | 设备控制 | control | 可下发可上报（rw） | enum | 枚举值: open, stop, close, continue
6 | 自启动开关 | auto_power | 可下发可上报（rw） | bool |  
8 | 开启百分比状态 | percent_state | 只上报（ro） | value | 数值范围: 0-100,                   间距: 1, 倍数: 0,                   单位: %
9 | 百分比控制 | percent_control | 可下发可上报（rw） | value | 数值范围: 0-100,                   间距: 1, 倍数: 0,                   单位: %
10 | 故障告警 | fault | 只上报（ro） | fault | 故障值: motor_fault
11 | 电机反转 | control_back | 可下发可上报（rw） | enum | 枚举值: forward, back
101 | 行程设置 | setting | 可下发可上报（rw） | bool |

my5123, 2024-07-13:
[image]

Koenkk, 2024-07-13:
Despite not all features working yet, in case you want, you can already make a PR to add initial support for this device. This can be done by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)

closed 2024-12-28 by github-actions[bot]

antonba, 2025-02-06:
Hey @my5123 did you manage to get the remaining functions working? I have the same motor and want to make it work. Are you willing to share your latest code so I can continue from it?

my5123, 2025-02-07:
> Hey [@my5123](https://github.com/my5123) did you manage to get the remaining functions working? I have the same motor and want to make it work. Are you willing to share your latest code so I can continue from it?

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_xtrnjaoz'},        
    ],
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    
    vendor: 'Tuya',
    description: '涂鸦控制器+中菱管状电机GM24TF/Z 34',
    fromZigbee: [tuya.fz.datapoints, legacy.fromZigbee.tuya_cover, fz.ignore_basic_report],
    toZigbee: [tuya.tz.datapoints, legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options],
    exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            e.enum('reverse_direction', ea.STATE_SET, ['forward', 'back']).withDescription('Reverse the motor direction'),
            e.binary('motor_fault', ea.STATE, true, false).withDescription('Motor Fault'),
            e.binary('auto_power', ea.STATE_SET, true, false).withDescription('Auto Power'),
            //e.numeric('motor_speed', ea.STATE_SET).withValueMin(34).withValueMax(45).withDescription('Motor speed').withUnit('rpm'),
            exposes.binary('set_upper_limit', ea.STATE_SET, true, false).withDescription('Set the limit'),
            e.enum('state', ea.STATE, ['CLOSE', 'STOP', 'OPEN', 'CONTINUE']).withDescription('State of the valve'),
            ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CONTINUE': tuya.enum(3), 'CLOSE': tuya.enum(0), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(2)})],
            [6, 'auto_power', tuya.valueConverter.raw],
            [9, 'position', tuya.valueConverter.coverPosition],
            [8, 'position', tuya.valueConverter.coverPosition],
            [10, 'motor_fault', tuya.valueConverter.trueFalse1],
            [11, 'reverse_direction', tuya.valueConverterBasic.lookup({forward: tuya.enum(0), back: tuya.enum(1)})],
            
            //[102, 'motor_speed', tuya.valueConverter.raw],
            [101, 'set_upper_limit', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;`

antonba, 2025-02-09:
Thank you @my5123 . Can you share the full list of DPIDs from the manufacturer? I tried sending in 102-107 as booleans, none worked.

For limits I think the motor should support:
- set upper
- set lower
- remove upper
- remove lower
- remove all

I'd need to at least be able to set upper and lower to fully set it up.

antonba, 2025-02-09:
Oh I just saw you actually posted what you got from manufacturer above. I wonder if that's truly the full list. I think the Tuya IoT Platform site can provide the list too, but I wasn't able to add mine directly to Tuya for some reason, only Zigbee.

antonba, 2025-02-09:
@my5123 do you have the Tuya Zigbee Gateway? If so, you could get DPIDs via that using the Tuya IoT Platform:
https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

If you don't have it, I'll buy one and try to get the datapoints via that.

my5123, 2025-02-09:
> @my5123 do you have the Tuya Zigbee Gateway? If so, you could get DPIDs via that using the Tuya IoT Platform:
> https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html
> 
> If you don't have it, I'll buy one and try to get the datapoints via that.

i don't have a tuya getway. but I had connected tuya develop team and post all the dps before.

antonba, 2025-02-09:
Ok, I ordered a gateway. Will take 7-10 days to arrive.

If you look at this video you can see that all the above limit-related functions should be available:
https://www.youtube.com/shorts/Kzzq60Ene8I

Hope I can get them using the IoT Platform site. And thanks for your help!

antonba, 2025-03-06:
Actually I managed to figure out how reset the motor and set a new limit. Since this isn't documented anywhere I will put it here:

**Reset:**
Long press reset button for 5-10 seconds until you hear a longer beep (IIRC)

**Set limit:**
- Turn on dp_setting (Set limit) mode
- Then press Open (Move Up) and when the blinds reach the desired upper limit, press Stop
- Then press Close (Move down) and when the blinds reach the desired lower limit press Stop
- The limit will now be set and Set limit mode will have turned itself off

my5123, 2025-06-26:
> Actually I managed to figure out how reset the motor and set a new limit. Since this isn't documented anywhere I will put it here:
> 
> **Reset:**
> Long press motor button for 5-10 seconds until you hear a longer beep (IIRC).
> This is not the same as pairing mode, which is a few short presses of the button.
> 
> **Set limit:**
> - Turn on dp_setting (Set Limit) mode
> - Then press Open (Move Up) and when the blinds reach the desired upper limit, press Stop
> - Then press Close (Move Down) and when the blinds reach the desired lower limit press Stop
> - The limit will now be set and Set limit mode will have turned itself off
> 

From your describe, I see that the limit setting could be done from the device reset button instate from zigbee2mqtt. 

I can do that with a remoter aswell, just want to know if I can do that within z2m. Now it seems impossble.

antonba, 2025-06-28:
> From your describe, I see that the limit setting could be done from the device reset button instate from zigbee2mqtt.
> 
> I can do that with a remoter aswell, just want to know if I can do that within z2m. Now it seems impossble.

It's been a while since I had to reset, but I think pressing the button on the motor (or remote) was required. I did not find a way using **only** Zigbee.