Koenkk/zigbee2mqtt #23238: Z2M uses outdated NPM dependencies

habitats-tech, 2024-07-02:
### What happened?

A project running on sound foundations has advantages. I suggest NPM dependencies are fixed.

### What did you expect to happen?

Not see any warnings on modules being outdated.

### How to reproduce it (minimal and precise)

git clone --depth 1 https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
npm ci

[image]

### Zigbee2MQTT version

1.38.0 and 1.39.0

### Adapter firmware version

N/A

### Adapter

N/A

### Setup

x86-64 Debian 12 based system container (LXC) under Proxmox 8.x (no Docker)

### Debug log

`npm ci`
npm warn deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated @humanwhocodes/object-schema@2.0.3: Use @eslint/object-schema instead
npm warn deprecated @humanwhocodes/config-array@0.11.14: Use @eslint/config-array instead
npm warn deprecated rimraf@3.0.2: Rimraf versions prior to v4 are no longer supported

added 799 packages, and audited 800 packages in 10s

87 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
npm notice
npm notice New minor version of npm available! 10.7.0 -> 10.8.1
npm notice Changelog: https://github.com/npm/cli/releases/tag/v10.8.1
npm notice To update run: npm install -g npm@10.8.1
npm notice

Koenkk, 2024-07-02:
This is caused by indirect dependencies, so this cannot be directly solved by z2m. Direct z2m dependencies are updated weekly.

mrmaximas, 2024-07-05:
@habitats-tech have you found a solution?

habitats-tech, 2024-07-06:
> @habitats-tech have you found a solution?

No time to investigate yet. Hopefully this coming week.

closed 2024-12-28 by github-actions[bot]