Koenkk/zigbee2mqtt #23239: Failure to boot on update

BradleyFord, 2024-07-02:
### What happened?

I have just updated to the latest release and Z2M seems to be stuck in a state where it is half started.

I can see on my SMLight SLZB-06M that there is a socket opened
I can see in the logs that Z2M seems to be running ok

But I am no longer able to access the web interface, and also I can see that HQ is also not receiving any information about the devices.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

2.3.6

### Adapter

SLZB-06M

### Setup

Proxmox / https://tteck.github.io/Proxmox/

### Debug log

[log.log](https://github.com/user-attachments/files/16068744/log.log)

`homeassistant:
  legacy_entity_attributes: true
  legacy_triggers: true
permit_join: false
mqtt:
  base_topic: z2m-master
  server: mqtt://192.168.1.xxx:1883
  user: xxxxxxxx
  password: xxxxxxxx
  keepalive: 60
  reject_unauthorized: true
  version: 4
  client_id: Z2M-master
  include_device_information: true
advanced:
  channel: 22
  ext_pan_id:
    - xxxxxxxx
  network_key:
    - xxxxxxxx
  transmit_power: 20
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: info 
  log_directory: data/log/%TIMESTAMP%
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 514
    protocol: udp4
    type: '5424'
  last_seen: ISO_8601_local
ota:
  zigbee_ota_override_index_location: ota.json
  ikea_ota_use_test_url: false
  disable_automatic_update_check: true
serial:
  port: tcp://192.168.1.xxxxxxxx:6638
  adapter: ember
  disable_led: true
  baudrate: 115200
frontend:
  port: 8081
  host: 192.168.1.xxxxxxxx
  url: http://192.168.1.xxxxxxxx:8081/
devices:
  '0xa4c138e0af310db0':
    friendly_name: L2_MasterWIRlight1
  '0xa4c13880d1a478da':
    friendly_name: L2_MasterWIRlight2
    color_sync: true
    state_action: true
  '0x54ef4410002a5bdb':
    friendly_name: L2_Master_AQARAtemp
    humidity_precision: 2
  '0xa4c13832b414ba96':
    friendly_name: L2_Masterlight7
  '0xa4c13871667fa53c':
    friendly_name: L2_Masterlight6
  '0xa4c138f345a9f622':
    friendly_name: L2_Masterlight3
  '0xa4c13802eff454a6':
    friendly_name: L2_Masterlight2
  '0xa4c138e019421a16':
    friendly_name: L2_Masterlight5
  '0xa4c1380e035ee759':
    friendly_name: L2_Masterlight4
  '0xa4c1387f92b30c7b':
    friendly_name: L2_Masterlight_1
  '0xa4c13886f4d912ca':
    friendly_name: L2_MasterbathLight1
  '0xa4c1386ae6fd55de':
    friendly_name: L2_MasterbathLight2
  '0xa4c1381d90cd413e':
    friendly_name: L2_MasterbathLight4
  '0xa4c138ec788c9343':
    friendly_name: L2_MasterbathLight3
  '0xa4c138c6a81e8c46':
    friendly_name: L1_ThaiKitchen_1
  '0xa4c138e53b817c32':
    friendly_name: L1_ThaiKitchen_2
  '0xa4c1383921ce39db':
    friendly_name: L1_Kitchen8
  '0xa4c1388de4958c10':
    friendly_name: L1_Kitchen10
  '0xa4c1381992478b86':
    friendly_name: L1_Kitchen7
  '0xa4c138b28cabbfe2':
    friendly_name: L1_Kitchen9
  '0xa4c138123f3f9171':
    friendly_name: L1_Kitchen11
  '0xa4c13809d9d7a13a':
    friendly_name: L1_Kitchen3
  '0xa4c138a06a5136a2':
    friendly_name: L1_Kitchen6
  '0xa4c1383f741c7e76':
    friendly_name: L1_Kitchen5
  '0xa4c138d679f764b3':
    friendly_name: L1_Kitchen1
  '0xa4c138204232631d':
    friendly_name: L1_Kitchen4
  '0xa4c138639fc609de':
    friendly_name: L1_Kitchen2
  '0x00158d000af33c76':
    friendly_name: L1_Letterbox
    vibration_timeout: 60
    x_calibration: ''
    device_temperature_calibration: ''
  '0x00158d000ae466de':
    friendly_name: L1_Pantry_AqaraTemp
  '0x00158d000ae4674b':
    friendly_name: L1_Living_AqaraTemp
  '0xbc026efffea7d485':
    friendly_name: L1_KitchenCooktop_Power
    power_precision: 2
    voltage_precision: 2
    energy_precision: 1
device_options:
  legacy: false
groups:
  '1':
    friendly_name: L2_Masterlight
    devices:
      - 0xa4c13802eff454a6/11
      - 0xa4c138f345a9f622/11
      - 0xa4c1380e035ee759/11
      - 0xa4c138e019421a16/11
      - 0xa4c13871667fa53c/11
      - 0xa4c13832b414ba96/11
  '2':
    friendly_name: L2_MasterbathLight
    devices:
      - 0xa4c13886f4d912ca/11
      - 0xa4c1386ae6fd55de/11
      - 0xa4c138ec788c9343/11
      - 0xa4c1381d90cd413e/11
  '3':
    friendly_name: L1_KitchenLight
    devices:
      - 0xa4c138d679f764b3/11
      - 0xa4c138639fc609de/11
      - 0xa4c13809d9d7a13a/11
      - 0xa4c138204232631d/11
      - 0xa4c1383f741c7e76/11
      - 0xa4c138a06a5136a2/11
      - 0xa4c1381992478b86/11
      - 0xa4c1383921ce39db/11
      - 0xa4c138b28cabbfe2/11
      - 0xa4c1388de4958c10/11
      - 0xa4c138123f3f9171/11
  '4':
    friendly_name: L2_MasterWIRLight
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
`

ov1d1u, 2024-07-02:
Does it starts if you (temporarily) remove groups from the configuration file?

BradleyFord, 2024-07-02:
> Does it starts if you (temporarily) remove the [groups](https://www.zigbee2mqtt.io/guide/usage/groups.html) section from the configuration file?

Just tried deleting that, it seems to not make any difference.

It's really strange that I cannot see it hanging or failing 
```

[2024-07-02 21:13:48] info: 	z2m: Logging to console, file (filename: log.log)
[2024-07-02 21:13:48] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #03269267)
[2024-07-02 21:13:48] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-07-02 21:13:49] info: 	zh:ember: Using default stack config.
[2024-07-02 21:13:49] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-07-02 21:13:49] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-07-02 21:13:49] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-02 21:13:49] info: 	zh:ember:uart:ash: Socket ready
[2024-07-02 21:13:49] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-02 21:13:49] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-07-02 21:13:49] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-02 21:13:49] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-02 21:13:49] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-02 21:13:49] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-02 21:13:50] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-07-02 21:13:50] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-07-02 21:13:50] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-07-02 21:13:51] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-07-02 21:13:51] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-02 21:13:51] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-02 21:13:51] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-02 21:13:51] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-07-02 21:13:51] info: 	zh:ember:queue: Request dispatching stopped; queue=0 priorityQueue=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Total frames: RX=2, TX=3
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Cancelled   : RX=1, TX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   DATA frames : RX=0, TX=1
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   DATA bytes  : RX=0, TX=4
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=1
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-07-02 21:13:51] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-07-02 21:13:51] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-07-02 21:13:51] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-07-02 21:13:51] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash: Socket ready
[2024-07-02 21:13:51] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-02 21:13:52] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-07-02 21:13:52] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-07-02 21:13:52] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-07-02 21:13:53] info: 	zh:ember: [STACK STATUS] Network up.
[2024-07-02 21:13:53] info: 	zh:ember: [INIT TC] NCP network matches config.
[2024-07-02 21:13:53] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-07-02 21:13:53] info: 	zh:ember:queue: Request dispatching started.
[2024-07-02 21:13:54] info: 	zh:ember:ezsp: <=== [ZDO clusterId=32824 sender=63234] Support not implemented upstream.
[2024-07-02 21:14:20] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "63234".
[2024-07-02 21:14:20] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "63234".
[2024-07-02 21:14:20] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "63234".
```