Koenkk/zigbee2mqtt #23242: Moes Smart Siding Windows Pusher

moshiko2312, 2024-07-02:
### Link

https://he.aliexpress.com/item/1005006907018715.html

### Database entry

{"id":55,"type":"EndDevice","ieeeAddr":"0x94b216fffee60a47","nwkAddr":50377,"manufId":4098,"manufName":"_TZ3210_5rta89nj","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0601","manufacturerName":"_TZ3210_5rta89nj","powerSource":3,"zclVersion":3,"appVersion":74,"stackVersion":1,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff075d59","zoneState":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1719935540970}

### Comments

Moes Smart Siding Windows Pusher 

### External definition

```shell
const {identify, battery, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');  const definition = {     zigbeeModel: ['TS0601'],     model: 'TS0601',     vendor: '_TZ3210_5rta89nj',     description: 'Automatically generated definition',     extend: [identify(), battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],     meta: {}, };  module.exports = definition;
```

moshiko2312, 2024-07-06:
If you need one i will buy for you @Koenkk

ryancasler, 2024-07-07:
Looks like Linptech is bringing it out under their brand name too.
https://www.aliexpress.com/i/3256805825642371.html

moshiko2312, 2024-07-18:
yes but this original from moes its working very good i ordering  8 pics

VikeDragon, 2024-12-20:
Greetings! This is relevant and has not been resolved yet. It would be great if device support appeared in z2m.

closed 2025-05-07 by github-actions[bot]