Koenkk/zigbee2mqtt #23245: All Parameters in  Parameters can't stay

cddu33, 2024-07-02:
### What happened?

When i click on "Retain" for exemple , 1s after, "Retain" is disable

[image]

### What did you expect to happen?

Parameters stay active

### How to reproduce it (minimal and precise)

Select device, Select Parameters, click on Retain

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

 20230507

### Adapter

 zStack3x0 

### Setup

vm debian 11

WarC0zes, 2024-07-03:
Hi,
I noticed the same behavior, when wanting to add an attribute filter.
[image]

Koenkk, 2024-07-03:
Can you check if this is solved by disabling permit join?

cddu33, 2024-07-03:
Hi , Permit Join is not active

WarC0zes, 2024-07-04:
Hi,
Same, Permit join is set to false.
[image]

closed 2024-12-28 by github-actions[bot]