Koenkk/zigbee2mqtt #23249: [New device support]:

oroGithub, 2024-07-03:
### Link

https://de.aliexpress.com/item/1005007050686755.html?spm=a2g0o.order_list.order_list_main.291.430518020D83P3

### Database entry

{"id":56,"type":"Router","ieeeAddr":"0xa4c13824fb7652a2","nwkAddr":38162,"manufId":4417,"manufName":"_TZE204_dapwryy7","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"(N\u0018.i(N\u0018.i*N\u0018.i,N\u0018.i","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_dapwryy7","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1720022889064}

### Comments

Hi,
I have created a new external converter and it seems to work fine.
I have taken the keywords from the Tuya Standard Instruction Set. Because none of these keys were defined in lib/tuya.ts, I added them there.

> [..., 'none_delay_time', 'indicator_led', 'none_delay_time_min', 'none_delay_time_sec', 'm_detection_distance_max', 'm_detection_distance_min', 'b_detection_distance_max', 'b_detection_distance_min', 'sm_detection_distance_max', 'sm_detection_distance_min', 'm_sensitivity', 'sm_sensitivity', 'b_sensitivity', ]

Should I try to use already defined keys?
I would like to contribute further, but have no idea how to use github.

### External definition

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_dapwryy7']),
    model: 'TS0601_new',
    vendor: 'Tuya',
    description: 'Mini human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.enum('presence_state', ea.STATE, ['none', 'presence', 'peaceful', 'small movement', 'large movement']).withDescription('Presence state'),
        e.illuminance_lux(),
        e.binary('indicator_led', ea.STATE_SET, 'ON', 'OFF').withDescription('LED Indicator'),
        e.numeric('target_distance', ea.STATE).withValueMin(0).withValueMax(1000).withValueStep(1).withUnit('cm').withDescription('Target distance'),
        e.numeric('none_delay_time', ea.STATE_SET).withValueMin(0).withValueMax(28800).withValueStep(1).withUnit('Sec').withDescription('Hold delay'),
        e.numeric('none_delay_time_min', ea.STATE_SET).withValueMin(0).withValueMax(479).withValueStep(1).withUnit('Min').withDescription('None delay time (Minutes)'),
        e.numeric('none_delay_time_sec', ea.STATE_SET).withValueMin(0).withValueMax(59).withValueStep(1).withUnit('Sec').withDescription('None delay time (Seconds)'),
        e.numeric('m_detection_distance_max', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.01).withUnit('m').withDescription('Move detection max distance'),
        e.numeric('m_detection_distance_min', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.01).withUnit('m').withDescription('Move detection min distance'),
        e.numeric('b_detection_distance_max', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m').withDescription('Breath detection max distance'),
        e.numeric('b_detection_distance_min', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m').withDescription('Breath detection min distance'),
        e.numeric('sm_detection_distance_max', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m').withDescription('Small detection Move max distance'),
        e.numeric('sm_detection_distance_min', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m').withDescription('Small detection Move min distance'),
        e.numeric('sm_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withDescription('Small Move sensitivity'),
        e.numeric('b_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withDescription('Breath sensitivity'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'presence': tuya.enum(1), 'peaceful': tuya.enum(2), 'small movement': tuya.enum(3), 'large movement': tuya.enum(4),})],
            [101, 'target_distance', tuya.valueConverter.raw],
            [102, 'illuminance_lux', tuya.valueConverter.raw],
            [103, 'none_delay_time', tuya.valueConverter.raw],
            [104, 'indicator_led', tuya.valueConverter.onOff],
            [105, 'none_delay_time_min', tuya.valueConverter.raw],
            [106, 'none_delay_time_sec', tuya.valueConverter.raw],
            [107, 'm_detection_distance_max', tuya.valueConverter.divideBy100],
            [108, 'm_detection_distance_min', tuya.valueConverter.divideBy100],
            [109, 'b_detection_distance_max', tuya.valueConverter.divideBy100],
            [110, 'b_detection_distance_min', tuya.valueConverter.divideBy100],
            [114, 'sm_detection_distance_max', tuya.valueConverter.divideBy100],
            [115, 'sm_detection_distance_min', tuya.valueConverter.divideBy100],
            [116, 'm_sensitivity', tuya.valueConverter.raw],
            [117, 'sm_sensitivity', tuya.valueConverter.raw],
[... 36 lines cut ...]
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'presence': tuya.enum(1), 'peaceful': tuya.enum(2), 'small movement': tuya.enum(3), 'large movement': tuya.enum(4),})],
            [101, 'target_distance', tuya.valueConverter.raw],
            [102, 'illuminance_lux', tuya.valueConverter.raw],
            [103, 'none_delay_time', tuya.valueConverter.raw],
            [104, 'indicator_led', tuya.valueConverter.onOff],
            [105, 'none_delay_time_min', tuya.valueConverter.raw],
            [106, 'none_delay_time_sec', tuya.valueConverter.raw],
            [107, 'm_detection_distance_max', tuya.valueConverter.divideBy100],
            [108, 'm_detection_distance_min', tuya.valueConverter.divideBy100],
            [109, 'b_detection_distance_max', tuya.valueConverter.divideBy100],
            [110, 'b_detection_distance_min', tuya.valueConverter.divideBy100],
            [114, 'sm_detection_distance_max', tuya.valueConverter.divideBy100],
            [115, 'sm_detection_distance_min', tuya.valueConverter.divideBy100],
            [116, 'm_sensitivity', tuya.valueConverter.raw],
            [117, 'sm_sensitivity', tuya.valueConverter.raw],
            [118, 'b_sensitivity', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

oroGithub, 2024-07-04:
Edit: It seemed contradictory to use an external converter and then also manipulate a source file to add the undefined keys. This now also happens in the external converter.

SgtStr4nger, 2024-07-06:
Thanks a lot for your work, it worked instantly for me! 

There are a few things I do not understand though: 
Why are there multiple None delay times? There are: time (sec), time min, time sec
I think the first one (None delay time (Hold delay(seconds)) is supposed to be Hold delay time (sec)?

Do None delay time sec & min add together?

oroGithub, 2024-07-07:
Hey, I was wondering about exactly the same thing. And I think you're right with your guess. The first value (28800 seconds) is the same as the other two converted to seconds (479 * 60 + 60). Maybe the makers of the sensor didn't know exactly which format was the right one in the end and they exposed both to the outside.
I haven't had time to look into the sensor any further.

weidis, 2024-07-09:
I can find the device and everything shows up but I am not able to find presence as a trigger when creating a flow.

SgtStr4nger, 2024-07-09:
I used detection of change in the presence-state as a trigger and it’s working just fine for me. Just search for the device name and it should be part of the quite long selection of available entities.

weidis, 2024-07-09:
> I used detection of change in the presence-state as a trigger and it’s working just fine for me. Just search for the device name and it should be part of the quite long selection of available entities.

Thanks, I have tried that but the only things I can place as triggers from that entity is "Unknown" "Unavailable" or "Any state"

SgtStr4nger, 2024-07-09:
> Thanks, I have tried that but the only things I can place as triggers from that entity is "Unknown" "Unavailable" or "Any state"

You can simply enter „large movement“ yourself and it works just fine ^^

oroGithub, 2024-07-09:
This device does not have a Boolean (true/false) status output. Instead, it uses the five states "none", "presence", "peaceful", "small_move" and "large_move". Of these states, however, I have only observed "none", "peaceful" and "large_move" in both the Tuya Smart app and in z2m.

weidis, 2024-07-09:
> You can simply enter „large movement“ yourself and it works just fine ^^

Ah thanks mate!

Koenkk, 2024-07-09:
Could you make a pull request to add out-of-the-box support for this device?

This can be done by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)

TraceoFox, 2024-07-10:
I would like to add your converter tho I'll get the following error when I wanna save it:

`[2024-07-10 18:48:48] error: 	z2m: Request 'zigbee2mqtt/bridge/request/extension/save' failed with error: 'ConstructorClass is not a constructor'
[2024-07-10 18:48:48] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/extension/save', payload '{"data":{},"error":"ConstructorClass is not a constructor","status":"error","transaction":"xmxdo-23"}'`

cyberdie, 2024-07-22:
Any update on this?

oroGithub, 2024-07-22:
Yes :smiley: 
[https://github.com/Koenkk/zigbee-herdsman-converters/blob/9ede78a950d703dac4e60271e08dabf7808661e6/src/devices/tuya.ts#L10736](url)

closed 2024-07-23 by Koenkk

JaCoLJcL, 2025-03-14:
2025-03-14 13:29:16z2m: Device 'XXX' with Zigbee model 'TS0601' and manufacturer name '_TZE284_iadro9bf' is NOT supported

I have this problem but i cant find _TZE284_iadro9bf (this is a 24 GHz human presence sensor) - can I use some of others converters to fix this ^ ?