Koenkk/zigbee2mqtt #23250: PA-44Z missing data point "silence"[Wrong device]:

elzershark, 2024-07-03:
### Link

https://www.zigbee2mqtt.io/devices/PA-44Z.html

### Model

PA-44Z

### Description

Smoke Sensor

### Vendor

TuYa

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/PA-44Z.png

### Database entry

??

### Notes

Please insert this (silence)
here is the complete code:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_m9skfctm', '_TZE200_rccxox8p']),
        model: 'PA-44Z',
        vendor: 'Tuya',
        description: 'Photoelectric smoke detector',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.smoke(),
            e.battery(),
	    tuya.exposes.silence(),
            e.test(),
            e.numeric('smoke_concentration', ea.STATE).withUnit('ppm').withDescription('Parts per million of smoke detected'),
            e.binary('device_fault', ea.STATE, true, false).withDescription('Indicates a fault with the device'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'smoke', tuya.valueConverter.trueFalse0],
                [2, 'smoke_concentration', tuya.valueConverter.divideBy10],
                [11, 'device_fault', tuya.valueConverter.raw],
                [15, 'battery', tuya.valueConverter.raw],
		[16, 'silence', tuya.valueConverter.raw],
                [101, 'test', tuya.valueConverter.raw],
            ],
        },
    };

module.exports = definition;
```

thx

Koenkk, 2024-07-04:
Could you make a pull request to fix it by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

elzershark, 2024-07-05:
Yes, i have that

Koenkk, 2024-07-06:
Thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-06 by Koenkk