Koenkk/zigbee2mqtt #23254: Device not supported ECODIM 5.0 with SLZB-06 as router.

xsienix, 2024-07-03:
### What happened?

I cant fix an unsupported device even i know it is supported. 
The dimmer is a duo-dimmer. 
Device shows up with all sliders etc, but i can only use one site of the dimmer.  

### What did you expect to happen?

A out of the box working solution, what is supposed to, because the thing is supported i'v red.

### How to reproduce it (minimal and precise)

I follow all the advise on Github. Making a extender etc. In the log, the z2m trows an error cause there is soms syntex failure, even do i past it from Koens advise in an earlier post.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20240315

### Adapter

SLZB-06 via network cable.

### Setup

Add-on on Home Assistant.

### Debug log

[log.log](https://github.com/user-attachments/files/16091261/log.log)

xsienix, 2024-07-05:
After some time it turned out that i have a new Ecodim.05.**zigbee**version, where zigbee is another dimmer then the 0.5 meaning that it is obviouse that it won't work. I need to make an extender. 
Issue closed.

xsienix, 2024-07-10:
As mentioned above it seems that i got an exotic Ecodim device. 
After some testing i notice that when i push "on" in the frontend of Z2M, the "on" button of the other channel goes on to.except the lamp will not go on. Also the brightness slider of channel 1 side, makes the slider on channel 2 moving. It seems that channel 2 is a copy of channel 1. Even though i follow the documentations, i can not fix it.
In the database the dimmer is mentioned, but only channel 1. Also turning the dimmer at channel 2 using the fysiek button on the dimmer, nothing will be mentioned in the log or database. 

There must be a slightly difference in firmware between the Ecodim05 and the Ecodim05 Zigbee dimmer. 

Requesting some help.
Thank you.

ozanmakes, 2024-07-19:
Hi @xsienix have you figured this one out? I am experiencing the same issue. I got the device to be recognized with the following device config, but I am not able to reliably set any states:

```
const {deviceEndpoints, identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Eco-Dim.05 Zigbee'],
    model: 'Eco-Dim.05 Zigbee',
    vendor: 'EcoDim BV',
    description: 'LED dimmer duo 2x 0-100W',
    extend: [
      deviceEndpoints({endpoints: {left: 2, right: 1}}),
      light({effect: false, configureReporting: true, endpointNames: ['left', 'right']}),
    ],
};

module.exports = definition;
```

xsienix, 2024-07-20:
Hi @ozanmakes ,

No, 
For some uknown reason i can only operate 1 dimmer.

Timieh, 2024-12-05:
Also having some issues and trying to debug. 
But nothing seems to work at all. 
Maybe we can help each other? 
You can find me on Discord (NL) timieh92

It shows unsupported and creating external def. doesn't really affect anything. I'm a bit stuck for days already. 
Also has the EcoDim05 2 channel DUO Dimmer Zigbee variant installed.

ozanmakes, 2024-12-05:
I am still using overrides I posted on this thread: https://github.com/Koenkk/zigbee2mqtt/issues/23375

It works well 90% of the time, well enough that I stopped thinking about it.

ozanmakes, 2024-12-05:
What is the 'extend part'? how does your external converter look like now?

xsienix, 2024-12-20:
@Timieh timieh92,

Meanwhile the dimmer is operating. There is an annoying thing and thats the status updates of the channels right and left. When you push the buttons on the dimmer, there are no status updates in homeassistant. I wonder if you have those status updates.

Many thanks.

xsienix, 2024-12-21:
Finaly, after a half a year orso i fixet it.

Right after i posted this message, i did a search, that what i did a half year ago, but never found answers. 
During search is stumbled into this [[topic](https://www.reddit.com/r/Zigbee2MQTT/comments/173a9sm/aqara_switch_does_not_update_the_status_when_one/)]
When you read it, you can see, that there is a picture as well, that explains everything. 

Meaning, you need to fill in some things in the reporting area of Z2M otherwise the dimmer reports nothing. 
Manuals are not clear about this, and in forums you can hardly find it.

I did this, and everything is working correctly.

Timieh, 2024-12-22:
[image]
I made it like this. This is working fine! 
(small note, I cannot fully test channel 1/left becuase that is having some strange issues. But expect to be fine.

closed 2025-03-01 by github-actions[bot]