Koenkk/zigbee2mqtt #23259: Attempt to access memory outside buffer bounds

perosb, 2024-07-04:
### What happened?

```
[2024-07-03 12:58:23] error:     zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds
at new NodeError (node:internal/errors:405:5)
at boundsError (node:internal/buffer:86:11)  
at Buffer.readUInt16LE (node:internal/buffer:245:5)
at Buffer.readUIntLE (node:internal/buffer:182:17) 
at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) 
at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54)    
at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23
at Array.every (<anonymous>)│
at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15) 
at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35) 
```

### Zigbee2MQTT version

1.39

### Adapter firmware version

6.7.8.0 build 373

### Adapter

EZSP v8

### Setup

docker/k8s

Koenkk, 2024-07-04:
You are using the old `ezsp` driver, it's recommended to use the new `ember` drivers ([info](https://github.com/Koenkk/zigbee2mqtt/discussions/21462))

perosb, 2024-07-04:
Yes, I dont think the new one works with this adapter and/or firmware.
Haven't tried it for a few releases tho, will test again tomorrow.

azalin1975, 2024-09-06:
I also keep getting the error message:
`Frame changeSourceRouteHandler parsing error: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds`
But only when running an OTA of one of the devices. I am wondering if this could brick the device it is updating or if this is not going to affect the device and the update will continue regardless?

In other words, should I not update until I have investigated if my dongle can be flashed to a newer firmware and until I have moved over to the ember driver? 

The thing is, I have several devices that are either hard ot reach or difficult to (re)pair. So I have been putting off updating the firmware on the dongle since I also do not know which version of the dongle I have.

stefandxm, 2024-10-08:
I have this issue with IKEA Parasoll, cannot interview them. Also cannot use ember  as host not supports it.

DagNygren, 2024-12-01:
Have the same here. Ember is not an option as the reprogrammed Lidl Zigbee gateway doen't have a newer version

closed 2025-02-07 by github-actions[bot]

0xSabonis, 2025-02-19:
that is in case you are usuing dev net on solana

Ambly, 2025-02-24:
For those reading this, the discussion https://github.com/Koenkk/zigbee2mqtt/discussions/22520
contains a nice solution (a 7.4.5.0 firmware for the Lidl gateway)