Koenkk/zigbee2mqtt #23260: [New device support]:  _TZE284_sgabhwa6 Smart Soil Sensor

syphernl, 2024-07-04:
### Link

https://www.aliexpress.com/item/1005007044173080.html

### Database entry

{"id":51,"type":"EndDevice","ieeeAddr":"0xa4c1384dd08b7e57","nwkAddr":60986,"manufId":4417,"manufName":"_TZE284_sgabhwa6","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"\u0004\u0000\u0000\u0000e\u0006\u0000\u0000\u0000\u0012�Q\u0019.\u0019�Q\u0019.\u0007\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE284_sgabhwa6","powerSource":3,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1720095939380}

### Comments

In debug mode, pushing the button results in the following Datapoints to be sent:

```
[2024-07-04 14:28:16] debug:    zhc:tuya: Datapoint 3 not defined for '_TZE284_sgabhwa6' with value 0
[2024-07-04 14:28:16] debug:    zhc:tuya: Datapoint 5 not defined for '_TZE284_sgabhwa6' with value 213
[2024-07-04 14:28:16] debug:    zhc:tuya: Datapoint 110 not defined for '_TZE284_sgabhwa6' with value 703
```

All DP's increase once I touch the prongs of the device and click the button on it. 
DP3 might the be temperature as it went to ~22 once held but I'm not sure what other datapoints are and how to convert them to correct values. Neither seem to be battery as it increases when held and decreases when not.

I expect this device to report:
- Temperature
- Soil Moisture
- Battery Percentage

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE284_sgabhwa6']),
    zigbeeModel: ['TS0601'],
    model: 'TS0601_soil_outdoor',
    vendor: 'TuYa',
    description: 'Outdoor Soil Sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.battery(), e.temperature(), e.soil_moisture()],                             
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [3, 'temperature', tuya.valueConverter.raw]
        ],                                            
    },    
};    
  
module.exports = definition;
```

syphernl, 2024-07-05:
Seems it is already supported, but not fingerprinted properly. I have opened a PR to add this.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7741

syphernl, 2024-07-08:
Reopening this as this device appears to be different from the `_TZE284_g2e6cpnw` one.

The `battery_state` is only reporting on 1 of the 3 devices I have (and is reporting `low`). 
The `battery` is being reported by all 3 but their graphs are **identical** to the one of the temperature, except there are different values. 

It appears that DP `110` is the temperature, but in Fahrenheit..
DP3 (divide by 10) = 20,5  °C, DP110 (divide by 10) is "68,9%". 
According to an [online fahrenheid-to-celcius conversion](https://www.rapidtables.com/convert/temperature/fahrenheit-to-celsius.html?x=68.9) this matches the celcius temperature

That leaves me with 2 questions:
* Is there a way to make this user-configurable which temperature unit is being used (celcius vs fahrenheit)?
* What would be the DP for battery, if it is not 110.  here is nothing in the logs thusfar about unknown DP's being reported by the devices ..

nikito7, 2024-07-08:
Tuya app

[image]

Maybe sensitivity can be changed on device

setfire2, 2024-07-13:
Hello @syphernl
Thanks for sharing your external definition. My TS0601 Soil Sensor identifies as _TZE284_sgabhwa6 as well. I copied your external definition to the same directory as z2m configuration.yaml and named it `TS0601_soil_2.js`
In configuration.yaml I added
```
external_converters:
  - TS0601_soil_2.js
```

After restarting z2m and pairing the device, z2m still says the device is not supported. Am I missing anything? It's my first time using external converters.

steenlarsen, 2024-07-19:
Regarding the above Tyua app information. I read the manual that came with my sensor and from that I assume the following : 
1) Temp and moisture sensitivity defines after what amount of change the sensor sends the new values to the Zigbee gateway. (to save battery)
2) The 10 minutes time defines how often the sensor reports it status to the Zigbee gateway even if the sensitivity levels are not triggered.
3) The max and minimum temp and humidity seems to be linked to a kind of alarm feature which will trigger scenes that can be used to turn on watering systems.

Some text from the manual : 
  
  
 

>  3. Detection logic

>   (1) Logic for reporting soil temperature and humidity. The device will trigger a report based on the detection time, temperature sensitivity and humidity sensitivity.

  

>  If the detection time is set to 20 minutes, the temperature sensitivity is 0.5 degrees Celcius and the humidity sensitivity is 10%. The device will detect every 20 minutes. If the soil temperature changes by more than 0.5 or the soil humidity changes by more than 10% the main interface of the APP will update the soil temperature and humidity values.

>  (2) Soil temperature and humidity alarm logic.

> The device will trigger an alarm based on whether it is above the upper limit or below the lower limit of temperature and humidity. For example, if the device is set to a maximum humidity of 90% and the soil humidity exceeds 90%, the device will trigger a high humidity alarm. 

  

>   Other function

> (1) Scene linkage

> Eg: When the soil temperature and humidity are detected to be below 50%, the scene linkage will automatically turn on the watering equipment and count down to 5 minutes for watering.

Hope this is useful for you.

jvrodrisan, 2024-07-29:
Hi, it seems this device will be covered under this new definition [https://github.com/Koenkk/zigbee-herdsman-converters/pull/7803](url).

I am testing this as an external definition, but some things such as the humidity are not working yet... we will wait for this PR to be merged in dev and will further test it.

In the meantime, I have debugged the tuya DPs, here they are for this device should someone need them:

```
{"3":"Moisture",
"5":"Temperature",
"9":"Temp Unit ",
"14":"Battery Level ",
"15":"Battery level",
"101":"Temp Alarm",
"102":"Moisture Alarm",
"103":"Maxtemp Set",
"104":"Minitemp Set",
"105":"Maxhum Set",
"106":"Minihum Set",
"107":"Temp Sensitivity",
"108":"Moisture Sensitivity",
"109":"Detection time",
"110":"Temperature",
"111":"Temp Sensitivity",
"112":"Maxtemp Set",
"113":"Minitemp Set"}
```

goberhammer, 2024-07-29:
So I gave it a try and tried to somewhat merge the external definition suggested in a previous comment https://github.com/Koenkk/zigbee2mqtt/issues/23260#issue-2390785832 with this PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/7710/files
Note that is my first ever external definition and I'm basically doing copy and paste (C developer here, knows nothing about Java/TypeScript :-) )
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =   {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE284_sgabhwa6']),
    model: 'TS0601_soil_2',
    vendor: 'Tuya',
    description: 'Soil sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.temperature(),
        e.soil_moisture(),
        e.battery(),
        tuya.exposes.batteryState(),
        tuya.exposes.temperatureUnit(),
        e.detection_interval(),
    ],
    meta: {
        tuyaDatapoints: [
            [3, 'soil_moisture', tuya.valueConverter.raw],
            [5, 'soil_temperature_celsius', tuya.valueConverter.divideBy10],
            [9, 'temperature_unit', tuya.valueConverter.temperatureUnitEnum],
            [14, 'battery_state', tuya.valueConverter.batteryState],
            [15, 'battery', tuya.valueConverter.raw],
            [101, 'soil_temperature_alarm', tuya.valueConverter.raw],
            [102, 'soil_moisture_alarm', tuya.valueConverter.raw],
            [103, 'max_soil_temperature_set_celsius', tuya.valueConverter.raw],
            [104, 'min_soil_temperature_set_celsius', tuya.valueConverter.raw],
            [105, 'max_soil_moisture_set', tuya.valueConverter.raw],
            [106, 'min_soil_moisture_set', tuya.valueConverter.raw],
            [107, 'soil_temperature_sensitivity_celsius', tuya.valueConverter.raw],
            [108, 'soil_moisture_sensitivity', tuya.valueConverter.raw],
            [109, 'detection_interval', tuya.valueConverter.raw],
            [110, 'soil_temperature_fahrenheit', tuya.valueConverter.divideBy10],
            [111, 'soil_temperature_sensitivity_fahrenheit', tuya.valueConverter.raw],
            [112, 'max_soil_temperature_set_fahrenheit', tuya.valueConverter.raw],
            [113, 'min_soil_temperature_set_fahrenheit', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

With the above code, I get this for Exposes:
[image]

And this for State:
```
{
    "battery": null,
    "linkquality": 156,
    "soil_moisture": 0,
    "temperature": null,
    "battery_state": null,
    "detection_interval": null,
    "soil_temperature_celsius": 31,
    "soil_temperature_fahrenheit": 87.8,
    "temperature_unit": "celsius",
    "soil_moisture_alarm": 0
}
```

Which do not seem ok/correct.
As I said, I'm out of my deeps here, any help/correction would be appreciated :-)

For @setfire2 : to use this:
- disable auto joining in Z2M ui (otherwise the sensor will try to re-add itself as soon as it is removed
- remove current instance
- use the code above in your extension file
- IMPORTANT: restart Z2M
- enable joining

rare-magma, 2024-07-29:
@goberhammer probably better to give this one a try https://github.com/Koenkk/zigbee-herdsman-converters/pull/7803/files it seems to have been created by the manufacturer

setfire2, 2024-07-29:
@goberhammer: thanks for your commitment and trying to help me out. I'll try out both versions, yours and the one from the manufacturer.

goberhammer, 2024-07-29:
@rare-magma thanks for pointing me to the PR.
I've quickly updated my external definition from that, here it is:
```
import fz from '../converters/fromZigbee';
import * as exposes from '../lib/exposes';
import * as legacy from '../lib/legacy';
const e = exposes.presets;
const ea = exposes.access;
import * as tuya from '../lib/tuya';
import {Definition} from '../lib/types';

const definition =   {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE284_sgabhwa6']),
    model: 'NAS-STH01B2',
    vendor: 'NEO',
    description: 'Soil moisture and temperature',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.numeric('humidity', ea.STATE).withUnit('%').withValueMin(0).withValueMax(100).withDescription('Soil humidity'),
        e.numeric('temperature', ea.STATE).withUnit('°C').withValueMin(-10).withValueMax(60).withDescription('Soil temperature'),
        e.numeric('temperature_f', ea.STATE).withUnit('°F').withValueMin(14).withValueMax(140).withDescription('Soil temperature'),
        e
            .numeric('temperature_sensitivity', ea.STATE_SET)
            .withUnit('°C')
            .withValueMin(0.3)
            .withValueMax(1)
            .withValueStep(0.1)
            .withDescription('Upper temperature limit'),
        e.numeric('humidity_sensitivity', ea.STATE_SET).withUnit('%').withValueMin(1).withValueMax(5).withDescription('Upper temperature limit'),
        e.enum('temperature_alarm', ea.STATE, ['lower_alarm', 'upper_alarm', 'cancel']).withDescription('Temperature alarm state'),
        e.enum('humidity_alarm', ea.STATE, ['lower_alarm', 'upper_alarm', 'cancel']).withDescription('Humidity alarm state'),
        e
            .numeric('max_temperature_alarm', ea.STATE_SET)
            .withUnit('°C')
            .withValueMin(0)
            .withValueMax(60)
            .withDescription('Upper temperature limit'),
        e
            .numeric('min_temperature_alarm', ea.STATE_SET)
            .withUnit('°C')
            .withValueMin(0)
            .withValueMax(60)
            .withDescription('Lower temperature limit'),
        e.numeric('max_humidity_alarm', ea.STATE_SET).withUnit('%').withValueMin(0).withValueMax(100).withDescription('Upper humidity limit'),
        e.numeric('min_humidity_alarm', ea.STATE_SET).withUnit('%').withValueMin(0).withValueMax(100).withDescription('Lower humidity limit'),
        e.numeric('schedule_periodic', ea.STATE_SET).withUnit('min').withValueMin(5).withValueMax(60).withDescription('Report sensitivity'),
        e.numeric('battery_percentage', ea.STATE).withUnit('%').withValueMin(0).withValueMax(100).withDescription('Battery percentage'),
    ],
    meta: {
        tuyaDatapoints: [
            [
                101,
                'temperature_alarm',
                tuya.valueConverterBasic.lookup({
                    lower_alarm: tuya.enum(0),
                    upper_alarm: tuya.enum(1),
                    cancel: tuya.enum(2),
                }),
            ],
            [
                102,
                'humidity_alarm',
                tuya.valueConverterBasic.lookup({
                    lower_alarm: tuya.enum(0),
                    upper_alarm: tuya.enum(1),
                    cancel: tuya.enum(2),
                }),
            ],
            [3, 'humidity', tuya.valueConverter.divideBy10],
            [5, 'temperature', tuya.valueConverter.divideBy10],
            [110, 'temperature_f', tuya.valueConverter.divideBy10],
            [107, 'temperature_sensitivity', tuya.valueConverter.divideBy10],
            [108, 'humidity_sensitivity', tuya.valueConverter.raw],
            [103, 'max_temperature_alarm', tuya.valueConverter.divideBy10],
            [104, 'min_temperature_alarm', tuya.valueConverter.divideBy10],
            [105, 'max_humidity_alarm', tuya.valueConverter.raw],
            [106, 'min_humidity_alarm', tuya.valueConverter.raw],
            [109, 'schedule_periodic', tuya.valueConverter.raw],
            [15, 'battery_percentage', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

But this seems not to work, and strangely enough the sensor is still recognized with the old external definition even if I've removed it from config and deleted from disk. Does Z2M cache the definitions of found devices somewhere ?

setfire2, 2024-07-29:
From https://github.com/Koenkk/zigbee-herdsman-converters/pull/7803/files I added lines 291-363 to my neo,js and removed the definitions from tuya.js and seems to work quite well. I see much more values and configuration options now. 
@baggiowu: Only the following description seems wrong:
[image]

setfire2, 2024-07-29:
@baggiowu: please notice my comment on line 350 of neo.ts. Also I get no value for the battery level.

goberhammer, 2024-07-31:
So this device is driving me a little insane... I've grepped though all the sources and confirmed that there is no other tuya fingerprint of device type `_TZE284_sgabhwa6` and the only definition is in my added external definition from a previous comment ( https://github.com/Koenkk/zigbee2mqtt/issues/23260#issuecomment-2256635278 )
Yet if I remove the device, restart Z2M and re-add it, it still is wrongly identified as another device (see image):
[image]
As you can see vendor is not **NEO** and description is different from the one in my external definition (soil sensor instad of Soil moisture and temperature). How could this be ? It seems that Z2M remembers some old definitions that is doesn't exist anymore and use that instead of the new one....
I'm lost...

goberhammer, 2024-08-05:
Just an update: I've removed the device, updated Z2M to the latest version 1.39.1 and now the sensor works perfectly. Not all the settings available in Tuya app are visible, but at least temperature, humidity, battery and battery state seems to be correct.

setfire2, 2024-08-05:
@goberhammer: unfortunately not. What you see as betery level in reality is the temperature in fahrenheit. Theres already a really good integration for this device checked in to the repo but it seems like it didn't make it into 1.39.1: 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/7816/commits/14406fce7cecb7b94c18a91c5e9f1b79e33bf7ac
See also this conversation: https://github.com/Koenkk/zigbee-herdsman-converters/pull/7816

goberhammer, 2024-08-06:
Thanks for the heads-up  :-D
I was able to patch my tuya.js (into the Docker container) directly from the tuya.ts file in the patch you linked and now it works a lot better.

setfire2, 2024-08-06:
You're welcome!

setfire2, 2024-09-20:
Temperature is fine for me as well. Only the battery level shows nonsense.

nprez83, 2024-09-22:
Any update on this issue @baggiowu? I have 2 of these devices and neither reports battery percentage.

baggiowu, 2024-09-23:
OK. Thanks for your feedback. I'll confirm this issue.

baggiowu, 2024-09-23:
> it seems latests update did not solve the issue. I am on version:
> zigbee-herdsman-converters version
> 20.12.1
> zigbee-herdsman version
> 0.57.3
>
> however, battery is not ok, temperature is not ok
>
> {
> "battery": -126.4,
> "battery_state": "low",
> "humidity_sensitivity": 1,
> "linkquality": 51,
> "max_temperature_alarm": 3,
> "schedule_periodic": 15,
> "soil_moisture": 23,
> "temperature": -88,
> "temperature_f": -126.4,
> "temperature_sensitivity": 0.5,
> "humidity_alarm": null,
> "max_humidity_alarm": null,
> "min_humidity_alarm": null,
> "min_temperature_alarm": null,
> "temperature_alarm": null
> }
>
> humidity sensitivity is now working, so I get update every 1% as requested, so thats great.
>
> any chances of fixing temperature and battery?

OK. Thanks for your feedback. I'll confirm this issue.

syphernl, 2024-09-23:
I have 3 of these devices, two seem to work fine (in regards to temperature + humidity). One seems to be stuck at reporting an old temperature and is reporting 0% humidity.

State of affected one:

```json
{
    "humidity": 0,
    "last_seen": "2024-09-23T08:30:45+02:00",
    "linkquality": 105,
    "temperature": 27.6,
    "temperature_f": 81.6
}
```

```json
{
    "battery": 68.9,
    "humidity": 19,
    "last_seen": "2024-09-23T08:26:11+02:00",
    "linkquality": 109,
    "soil_moisture": 19,
    "soil_temperature_fahrenheit": 77.1,
    "temperature": 16,
    "temperature_alarm": "cancel",
    "temperature_f": 60.8,
    "temperature_unit": "celsius"
}
```

I have already tried (force-)unpairing and pairing it again to no avail.

setfire2, 2024-09-23:
@baggiowu: thanks for checking the issue(s) once again.
@syphernl: are you talking about the same device? -> https://www.zigbee2mqtt.io/devices/TS0601_soil_2.html#tuya-ts0601_soil_2
It does not expose humidity. Only soil_moisture.

nprez83, 2024-09-23:
One more piece of info @baggiowu, to make sure everyone's on the same page as there are several PRs and threads related to this device (_TZE284_sgabhwa6). I think at some point you had include this device in the [neo.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/neo.ts) converter, but it seems the current version of the herdsman has it under the [tuya.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts). Figured I'd mention it.

syphernl, 2024-09-24:
> @baggiowu: thanks for checking the issue(s) once again. @syphernl: are you talking about the same device? -> https://www.zigbee2mqtt.io/devices/TS0601_soil_2.html#tuya-ts0601_soil_2 It does not expose humidity. Only soil_moisture.

Yes, same device. But I was using a different converter. 
Removed that custom one for now, but it still acts the same, it reports `humidity` not `soil_moisture`. 1 device fails to report any changes (stuck on very old values), others seem to work fine (except for battery).

It indeed reports as Tuya, not Neo.

baggiowu, 2024-09-25:
I have fixed the issues, and create a new pull request.

baggiowu, 2024-09-25:
The device(_TZE284_sgabhwa6) is added by the others, so I delete it from neo.ts when pull request.

Now I fix the issue. I have updated the device  model, and tuya DPID.

setfire2, 2024-09-25:
Well, I tried it out and for me it seems to be worse, to be honest. The battery state still shows the same value and humidty is now 11% instead of soil_moisture 16% as before. From my perspective it would be more helpful to revert that change.

nprez83, 2024-09-27:
@baggiowu, is DP 15 the correct DP for battery? I tried your edits in [65101a3](https://github.com/baggiowu/zigbee-herdsman-converters/commit/65101a392bea4803b7c48b21a96d93b73a07675d), made no changes to the battery DP, so I still have no battery reading. I see the NAS-STH02B2 device has DP 15 for battery, but this does not seem to be the case for this device.

nprez83, 2024-10-12:
@baggiowu, please see [my post](https://github.com/Koenkk/zigbee2mqtt/issues/23729#issuecomment-2395179393). The device just simply does not report DP 15 at all. Is this a firmware issue?

baggiowu, 2024-10-21:
When the device is woken up five times, it will detect the battery level. When the change is greater than 5%, DP 15 is reported.

If DP 15 is never reported, it is issues of MCU FW.

Can you tell me the version of MCU FW?

setfire2, 2024-10-21:
Hi @baggiowu. I'd really like to give you the information. I just don't know how to find out. Any help would be appreciated.

shift-del1, 2024-10-25:
@baggiowu: my soil sensor is connected to Tuya, here are the version numbers:

[image]

nprez83, 2024-10-27:
@baggiowu my sensors are unfortunately not connected to the SmartLife app as I don't have a Tuya hub, so they're only connected to HA via a Sonoff stick. Not sure if I can find out the MCU FW in that setting.

nprez83, 2024-12-27:
@baggiowu, this device still doesn't report battery, any updates?

nprez83, 2025-02-27:
Any updates on this issue @baggiowu?

GrumpyMeow, 2025-03-18:
I've just received my TS0601_soil_2 (`_TZE284_sgabhwa6`) and it works rightaway without a problem.
Maybe these screenshots help to investigate the issues...

[image]

[image]

[image]

nprez83, 2025-03-19:
Very interesting @GrumpyMeow. My soil sensor still doesn't report battery level or battery state, and my Z2M, herdman, and converter versions are actually more up to date than yours. I wonder if the company made a firmware version, but without adding the device thorough the Tuya hub directly, I'm not sure how we can find out what the built in firmware version is. 

[image]

[image]

GrumpyMeow, 2025-03-19:
On my device it never reported the Battery State. It took a few hours before it reported BatteryLevel

GrumpyMeow, 2025-03-24:
The battery level does change over time. Quite quickly actually.

[image]

setfire2, 2025-03-24:
Thanks again for the Info, @GrumpyMeow. Looks like you picked a good one...
Could you maybe post a link where you bought it?
I'm sure the difference lies in the firmware.

rare-magma, 2025-03-25:
I can confirm the same behaviour on 2 of my sensors:
1. `_TZE284_sgabhwa6` Battery state N/A, Battery level updates as expected: https://www.aliexpress.com/item/1005007016626219.html (Bought in 2024-05-26 from https://www.aliexpress.com/store/5747295, no longer available)
2. `_TZE284_g2e6cpnw` Battery state low, Batter level stuck on 77.7%: https://www.aliexpress.com/item/1005007109402012.html (Bought in 2024-07-25)

setfire2, 2025-03-25:
Thanks @rare-magma.  Too bad that the working one is no longer available. What's also interesting: I have a `TZE284_sgabhwa6` as well (bought on 2024-07-05 from https://de.aliexpress.com/item/1005007109402012.html) and the battery level is stuck there. Might it be that you mixed up the device IDs?
Probably they broke something in the firmware at some point.
@baggiowu: maybe this information helps you to investigate further?

rare-magma, 2025-03-25:
@setfire2 I've confirmed that the device IDs and state samples are correct above so not sure if the ID has any correlation, perhaps there's some other ID we can check?

nprez83, 2025-03-25:
I agree it must be a firmware issue. I bought two devices back in September 2024, and none of them report battery level at all. I even played around with the converter and could not get it to report, only to realize the data point is just not broadcasted by the devices period. Here are the Alliexpress links.

https://a.aliexpress.com/_mtTn7CT
https://a.aliexpress.com/_mNuq2R9

setfire2, 2025-04-08:
Well, to my surprise the value for battery level changed for the first time three weeks ago. It raised from 63.6 to 95 percent and a week later it dropped to 90 percent:
[image]

It still looks odd to me, somehow.

nprez83, 2025-04-08:
> Well, to my surprise the value for battery level changed for the first time three weeks ago. It raised from 63.6 to 95 percent and a week later it dropped to 90 percent:
> [image]
> 
> It still looks odd to me, somehow.

Is that value different than your humidity value?

setfire2, 2025-04-20:
Absolutely. This is what the humidity looks like.

[image]

nprez83, 2025-04-24:
That's great. It does look like yours is working well. My 2 still don't advertise battery level, I must have bought a bad batch.

leraystep95, 2025-06-24:
What a pity that this compatibility problem remains following! I acquired a very similar module with solar charging system ... and I did not manage to integrate it into Zigbee2mqtt!

https://fr.aliexpress.com/item/1005008023933361.html?spm=a2g0o.order_list.order_list_main.5.61615e5bevgCNN&gatewayAdapt=glo2fra

closed 2025-09-01 by github-actions[bot]