Koenkk/zigbee2mqtt #23262: Innr SP 120 stops reporting state - only manually refreshable via z2m GUI

Soukyuu, 2024-07-04:
### What happened?

I have read through some previous issues, i.e. #3576 and tried re-configuring as well as re-adding the devices in question and they seem to work for a few seconds after which the updates stop and you have to press on the blue arrow symbol in the z2m GUI to actively update the value. The logs contain intermittent entries of this kind:

`Error 2024-07-04 16:47:46z2m: Publish 'get' 'power' to 'weihnachtsbeleuchtung_terrase' failed: 'Error: ZCL command 0x00158d0003883f96/1 haElectricalMeasurement.read(["activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (168))'`

This started happening over the last month, maybe two months without me actively changing anything about my zigbee setup or network.

I have tried putting my spare tuya plug at the same spot and it updates its power value without issues.

### What did you expect to happen?

The state for power, current and energy to update automatically as it did previously.

### How to reproduce it (minimal and precise)

- add the plug to your network
- plug in a device which draws power
- go to plug's "exposes" page on z2m
- observe the values are not being updated

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Setup

Add-on on HomeAssistant OS, x86_64

Soukyuu, 2024-07-04:
It seems it stopped working at around June 6th, but there were no updates done at that day. The update that is close to that date was done on June 2nd, updating to 1.38.0

I am currently trying to rollback to 1.37.1 to see if it fixes things.

Soukyuu, 2024-07-04:
Downgrading to up to 1.34.0 has no effect on the problem, but the blue refresh icon beside power has now disappeared.

I can get the updates going automatically for a while if I go to binding and press "bind" again. It then continues updating for about 3 minutes and then abruptly stops. LQI is at 255, so it's either a router suddenly not forwarding the messages even though I haven't re-positioned anything or something else is broken.

Soukyuu, 2024-07-04:
Just attempted to re-configure and got this:
`Error 2024-07-04 18:02:45Failed to configure 'arbeitszimmer_pc', attempt 1 (Error: ConfigureReporting 0x00158d000362cbb2/1 haElectricalMeasurement([{"attribute":"rmsCurrent","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (175)) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:81:20) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:759:28) at Object.rmsCurrent (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:216:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/innr.ts:594:13) at Configure.configure (/app/lib/extension/configure.ts:117:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))`

w00dst0ck, 2024-09-20:
Same problem with my INNR SP 120 plugs.

w00dst0ck, 2024-09-20:
https://github.com/zigpy/zha-device-handlers/issues/3243#issuecomment-2337805352

@Soukyuu this fixed the problem for me.

Soukyuu, 2024-09-20:
Thanks, will see if it works. However I already see that if I change the value, then the reporting entry is duplicated, with the incorrect remaining on top and my changed ones on the bottom. If I then attempt to remove the incorrect one, it removes my edit one and keeps the incorrect one!

[image]

Soukyuu, 2024-09-20:
By the way, that is also what happens if I reconfigure the device: duplicated entries that cannot be removed.

@Koenkk: is there maybe a more generic issue of how reporting configuration is set up for the devices?

Soukyuu, 2024-09-20:
@w00dst0ck: are you sure it fixed the problem for you? The values do update for some time, but then stop after a while. It's still happening here even after applying the settings, probably because the default entries are overriding the edited ones.

w00dst0ck, 2024-09-20:
I‘ll check the values tomorrow and give an update.

closed 2024-12-28 by github-actions[bot]