Koenkk/zigbee2mqtt #23264: Error in Watchdog Documentation

ScumbagSteve, 2024-07-04:
### What happened?

There is an error in the Watchdog documentation:
https://www.zigbee2mqtt.io/guide/installation/15_watchdog.html

It is mentioned to set the ENV variable like that:
Z2M_WATCHDOG=0.5,3,6,15 npm start

But it only works like described in the issue [here](https://github.com/Koenkk/zigbee2mqtt/pull/23043#issuecomment-2174236940)
Z2M_WATCHDOG=1,2,3

I am using docker environment.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

/

### Adapter

SLZB 06M

### Setup

Docker

Koenkk, 2024-07-06:
`Z2M_WATCHDOG=0.5,3,6,15 npm start` should work, what doesn't work and could you provide the STDOUT when starting z2m?

ScumbagSteve, 2024-07-15:
It might have to do with the UNRAID way to run the docker and that it sets it in high-commas. But it definitely didn't boot the other way for me. I tried at least half an hour, until I found this solution.

```
docker run
  -d
  --name='zigbee2mqtt'
  --net='br0'
  --ip='192.168.1.230'
  -e TZ="Europe/Berlin"
  -e HOST_OS="Unraid"
  -e HOST_HOSTNAME="UNRAID"
  -e HOST_CONTAINERNAME="zigbee2mqtt"
  -e 'TCP_PORT_9442'='9442'
  -e 'Z2M_WATCHDOG'='0.5,3,6,15'
  -l net.unraid.docker.managed=dockerman
  -l net.unraid.docker.webui='http://192.168.1.230:9442/'
  -l net.unraid.docker.icon='https://raw.githubusercontent.com/digiblur/unraid-docker-templates/master/images/zigbee2mqtt.png'
  -v '/mnt/user/appdata/zigbee2mqtt':'/app/data':'rw'
  --restart=unless-stopped
  --mac-address 02:42:c0:a8:01:e6 'koenkk/zigbee2mqtt' 
```

closed 2024-12-28 by github-actions[bot]