Koenkk/zigbee2mqtt #23266: No access to frontend since 1.39.0 upgrade

eekamouze, 2024-07-05:
### What happened?

Web UI is not accessible since 1.39.0. OK after rollback to 1.38.0

### What did you expect to happen?

Frontend access on the same url as usual

### How to reproduce it (minimal and precise)

Upgrade from 1.38.0 to 1.39.0 --> no frontend
Rollback to 1.38.0 --> frontend OK

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12

### Setup

Docker (RPI4)

Akusky, 2024-07-05:
same problem, i resolved setting "availability: false"

Koenkk, 2024-07-06:
- Does z2m itself start? (e.g. can you control Zigbee devices through Home Assistant)
- Could you provide the debug log when starting z2m?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

eekamouze, 2024-07-06:
Yes, z2m starts and seems to work as expected, only the frontend is unavailable. I don't see any difference in log_level info at startup between 1.38.0 and 1.39.0.

Here is the debug attached
[debug.txt](https://github.com/user-attachments/files/16115580/debug.txt)

Spb2005, 2024-07-06:
Had the same issue, fixed it by adding frontend: true in the configuartion.yaml
But i still have the issue, that all my devices are lost (https://github.com/Koenkk/zigbee2mqtt/issues/23235).
I am thinking about a downgrading to 1.38

ArieGT, 2024-07-06:
Having the same issue. Went back to version 1.37.1-1, no issue with that version.

Bigtweety, 2024-07-06:
> same problem, i resolved setting "availability: false"

+1 same issue
fixed by changing custom availability to false as proposed.

Thanks @ArieGT.

Hopefully this gets fixed, I do like having control over device timeout.

BigT

jerrm, 2024-07-06:
May not be related, I had a similar issue in 1.38 (haven't upgraded to 1.39) .  The problem for me were the `icon:` entries in devices.yaml.  After removing all `icon:` entries, everthything came back.

Seems like the yaml processing for the frontend might be buggy somewhere.

romkey, 2024-07-06:
Same problem with 1.39.0. Removing `availability: true` from `configuration.yaml` and restarting got the web UI back. Nothing of interest in the logs.

eqasxar, 2024-07-08:
z2m stopped working properly for me too. All entities received an additional suffix '_2' . If before the update there was switch.room_light, then after the update it is switch.room_light_2 and thus all more than 150 entities. In this case, all scripts and all automations need to be rewritten. It's a really big job. Now I have disabled auto update and keep zigbee2mqtt old version. The future is risky.

Koenkk, 2024-07-09:
@romkey could you provide the debug log when starting with both availability true and false?

Tirguy, 2024-07-09:
Same problem since update this evening; here are some informations & debug LOG :

[z2m_availability-false.txt](https://github.com/user-attachments/files/16149740/z2m_availability-false.txt)
[z2m_availability-true.txt](https://github.com/user-attachments/files/16149742/z2m_availability-true.txt)

Here are the 2 options with the DEBUG mode level...

Thank's again for all

Thierry

Koenkk, 2024-07-11:
Should be fixed now! Let me know if it works.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

romkey, 2024-07-11:
@Koenkk My apologies for not getting you logs in time to be helpful.

The dev branch works for me, the web interface is now working properly with `availability: true`.

Thanks for the quick fix and for all the work you put into this!

Koenkk, 2024-07-12:
Great, I guess this can be closed now, thanks all!

closed 2024-07-12 by Koenkk