Koenkk/zigbee2mqtt #23268: [New device support]:

serot23, 2024-07-05:
### Link

https://www.zigbee2mqtt.io/devices/TW-S1.html

### Database entry

{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d0002f87ccb","nwkAddr":52353,"manufId":4628,"manufName":"iHorn","powerSource":"Battery","modelId":"LH05121","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"LH05121","manufacturerName":"iHorn","powerSource":3,"zclVersion":1,"appVersion":15,"stackVersion":2,"hwVersion":17,"dateCode":"20170414"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0029de0963","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":15,"stackVersion":2,"hwVersion":17,"dateCode":"20170414","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1720169981439}

### Comments

i have several konke smoke detectors, where 1 is supported by z2m and identify as:

Apparaat type
EndDevice
Zigbee model
3AFE010104020028
Zigbee Fabrikant
Konke
Omschrijving
Photoelectric smoke detector

Fabrikant
[Konke](https://www.zigbee2mqtt.io/supported-devices/#v=Konke)
Model
[TW-S1](https://www.zigbee2mqtt.io/devices/TW-S1.html#konke-tw-s1)

and several others which are not supported, i bought these first and the voice is in chinese, the internals and exterior looks identical but this identifies as:

Apparaat type
EndDevice
Zigbee model
LH05121
Zigbee Fabrikant
iHorn
Ondersteuningsstatus
Niet ondersteund [(how_to_add_support)](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)

Firmware build date
20170414

but the exposed stuff isn't the same as the english spoken (supported one). external converter worked until version 1.36 after the updates, i can't get it to work like the supported one.

when i press generate they generate the same external converter,

but the supported one works like intended like my old converter:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
zigbeeModel: ['LH05121'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
model: 'LH05121', // Vendor model number, look on the device for a model number
vendor: 'Konke', // Vendor of the device (only used for documentation and startup logging)
description: 'Konke smoke detector', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
fromZigbee: [fz.ias_smoke_alarm_1],
toZigbee: [],
exposes: [e.smoke(), e.battery_low()], // <-- this will define which fields will be exposed in the definition message to configure a front end (e.g. the z2m frontend, Home Assistant, Domoticz)
};

module.exports = definition;

I tried so many converters i could but failed and don't understand why the same device from the same seller has differences. I tried the original bridge to ota, but that didn't work either:

https://zigbee.blakadder.com/Konke_KK-SA-H03.html

### External definition

```shell
const {identify, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['LH05121'],
    model: 'LH05121',
    vendor: 'iHorn',
    description: 'Automatically generated definition',
    extend: [identify(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;
```

serot23, 2024-07-05:
this is the external definition of the supported one:

const {identify, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['3AFE010104020028'],
    model: '3AFE010104020028',
    vendor: 'Konke',
    description: 'Automatically generated definition',
    extend: [identify(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;

serot23, 2024-07-05:
[image]

serot23, 2024-07-05:
[image]

Koenkk, 2024-07-07:
With the generated external definition, does either `alarm_1` or `alarm_2` change to `true` when detecting smoke?

serot23, 2024-07-07:
> With the generated external definition, does either `alarm_1` or `alarm_2` change to `true` when detecting smoke?

Thank you for your quick response, alarm_1 changes to true when smoke is detected

Koenkk, 2024-07-09:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-09 by Koenkk