Koenkk/zigbee2mqtt #23270: Livarno (Lidl) lights color_power_on_behavior typo gives an error in the log

yvos, 2024-07-06:
### What happened?

Livarno (Lidl) lights color_power_on_behavior typo gives an error in the log. There seems to be an S missing in the word cutomized. Which most likely produces the error. 

Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:120
integration: MQTT (documentation, issues)
First occurred: 5 July 2024 at 22:11:53 (97 occurrences)
Last logged: 08:42:18

Invalid option for select.hal_2e_ledpaneel_color_power_on_behavior: '_cutomized_' (valid options: ['initial', 'previous', 'customized'])
Invalid option for select.berging_ledpaneel_color_power_on_behavior: 'cutomized' (valid options: ['initial', 'previous', 'customized'])
Invalid option for select.badkamer_lamp_color_power_on_behavior: 'cutomized' (valid options: ['initial', 'previous', 'customized'])
Invalid option for select.zzz_livarno_color_white_color_power_on_behavior: 'cutomized' (valid options: ['initial', 'previous', 'customized'])

### How to reproduce it (minimal and precise)

Set the default power on behaviour in z2m and check the HA logs. 

### Zigbee2MQTT version

1.39

### Adapter firmware version

7.3.1

### Adapter

Ezsp 

### Setup

Add-on on intel nuc

Koenkk, 2024-07-07:
This has been fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/7673 which landed in 1.39.0, are you sure you are running z2m 1.39.0?

yvos, 2024-07-07:
I double checked it. I am definitely running 1.39.0. 
Also the error in the log is recent. And keeps occurring today. 

```
Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:120
integration: MQTT (documentation, issues)
First occurred: 6 July 2024 at 10:09:30 (287 occurrences)
Last logged: 19:05:10

Invalid option for select.hal_2e_ledpaneel_color_power_on_behavior: 'cutomized' (valid options: ['initial', 'previous', 'customized'])
Invalid option for select.berging_ledpaneel_color_power_on_behavior: 'cutomized' (valid options: ['initial', 'previous', 'customized'])
Invalid option for select.badkamer_lamp_color_power_on_behavior: 'cutomized' (valid options: ['initial', 'previous', 'customized'])
Invalid option for select.zzz_livarno_color_white_color_power_on_behavior: 'cutomized' (valid options: ['initial', 'previous', 'customized'])

```
[image]

Koenkk, 2024-07-09:
It could be that you have to change the power on behavior once, this should make it report the correct value

yvos, 2024-07-10:
That worked thanks a lot!

Mvg,
Yuri Vos

closed 2024-07-10 by Koenkk