Koenkk/zigbee2mqtt #23271: Surfou Plug leaves network after pairing

oberschwabe, 2024-07-06:
### What happened?

trying to pair a Surfou Smart ZigBee Plug 
it gets detected and seems to be supported according logs:
Info 2024-07-06 11:14:02zh:controller: Succesfully interviewed '0x70b3d52b60056fd0'
Info 2024-07-06 11:14:02z2m: Successfully interviewed '0x70b3d52b60056fd0', device has successfully been paired
Info 2024-07-06 11:14:02z2m: Device '0x70b3d52b60056fd0' is supported, identified as: Tuya Smart plug (with power monitoring by polling) (BSD29_1)

after a few secs it just leaves the network again:
Info 2024-07-06 11:14:02z2m: Successfully configured '0x70b3d52b60056fd0'
Warning 2024-07-06 11:14:23z2m: Device '0x70b3d52b60056fd0' left the network
Info 2024-07-06 11:14:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b60056fd0","ieee_address":"0x70b3d52b60056fd0"},"type":"device_leave"}'

### What did you expect to happen?

device stays in network

### How to reproduce it (minimal and precise)

repair the device

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20230507

### Adapter

Sonoff

### Setup

Pi

### Debug log

2024-07-06 11:13:59z2m: Accepting joining not in blocklist device '0x70b3d52b60056fd0'
Info 2024-07-06 11:13:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b60056fd0', payload '{"child_lock":null,"current":null,"energy":null,"indicator_mode":null,"last_seen":"2024-07-06T11:13:58+02:00","linkquality":72,"power":null,"power_outage_memory":null,"state":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
Info 2024-07-06 11:13:59zh:controller: Interview for '0x70b3d52b60056fd0' started
Info 2024-07-06 11:13:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b60056fd0', payload '{"child_lock":null,"current":null,"energy":null,"indicator_mode":null,"last_seen":"2024-07-06T11:13:58+02:00","linkquality":72,"power":null,"power_outage_memory":null,"state":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
Info 2024-07-06 11:13:59z2m: Device '0x70b3d52b60056fd0' joined
Info 2024-07-06 11:14:00z2m: Starting interview of '0x70b3d52b60056fd0'
Info 2024-07-06 11:14:00z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b60056fd0","ieee_address":"0x70b3d52b60056fd0"},"type":"device_joined"}'
Info 2024-07-06 11:14:00z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b60056fd0","ieee_address":"0x70b3d52b60056fd0","status":"started"},"type":"device_interview"}'
Info 2024-07-06 11:14:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b60056fd0', payload '{"child_lock":null,"current":null,"energy":null,"indicator_mode":null,"last_seen":"2024-07-06T11:14:00+02:00","linkquality":72,"power":null,"power_outage_memory":null,"state":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
Info 2024-07-06 11:14:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b60056fd0","ieee_address":"0x70b3d52b60056fd0"},"type":"device_announce"}'
Info 2024-07-06 11:14:02zh:controller: Succesfully interviewed '0x70b3d52b60056fd0'
Info 2024-07-06 11:14:02z2m: Successfully interviewed '0x70b3d52b60056fd0', device has successfully been paired
Info 2024-07-06 11:14:02z2m: Device '0x70b3d52b60056fd0' is supported, identified as: Tuya Smart plug (with power monitoring by polling) (BSD29_1)
Info 2024-07-06 11:14:02z2m: Configuring '0x70b3d52b60056fd0'
Info 2024-07-06 11:14:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart plug (with power monitoring by polling)","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":7,"description":"Recover state after power outage","label":"Power outage memory","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":7,"description":"LED indicator mode","label":"Indicator mode","name":"indicator_mode","property":"indicator_mode","type":"enum","values":["off","off/on","on/off","on"]},{"access":1,"category":"diagnostic","description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"category":"diagnostic","description":"Instantaneous measured electrical current","label":"Current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"category":"diagnostic","description":"Measured electrical potential value","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","label":"Energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"features":[{"access":3,"description":"Enables/disables physical input on the device","label":"State","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"label":"Child lock","type":"lock"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"BSD29_1","options":[{"access":2,"description":"This device does not support reporting electric measurements so it is polled instead. The default poll interval is 60 seconds, set to -1 to disable.","label":"Measurement poll interval","name":"measurement_poll_interval","property":"measurement_poll_interval","type":"numeric","value_min":-1},{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Power precision","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","label":"Current calibration","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for current, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Current precision","name":"current_precision","property":"current_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","label":"Voltage calibration","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for voltage, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Voltage precision","name":"voltage_precision","property":"voltage_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the energy value (percentual offset), takes into effect on next report of device.","label":"Energy calibration","name":"energy_calibration","property":"energy_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for energy, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Energy precision","name":"energy_precision","property":"energy_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"Tuya"},"friendly_name":"0x70b3d52b60056fd0","ieee_address":"0x70b3d52b60056fd0","status":"successful","supported":true},"type":"device_interview"}'
[... cut ...] ull,"power_outage_memory":null,"state":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null}'
Info 2024-07-06 11:14:02z2m: Successfully configured '0x70b3d52b60056fd0'
Warning 2024-07-06 11:14:23z2m: Device '0x70b3d52b60056fd0' left the network
Info 2024-07-06 11:14:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b60056fd0","ieee_address":"0x70b3d52b60056fd0"},"type":"device_leave"}'

oberschwabe, 2024-09-17:
same issue now with a power plug.
it joins the network, in the short period of time i can turn it on/off and then it immediatelly leaves the network.

any help highly appreciated
[ts011f_plug.log](https://github.com/user-attachments/files/17023265/ts011f_plug.log)

quensen, 2024-09-18:
Servus @oberschwabe , what model?

I am not sure, if this helps, but my Surfou TS011F (`_TZ3000_okaz9tjs`) plugs (falsely recognized by Z2M as TuYa BSD29_1) work as intended.
Only issue is, that the TuYa BSD29_1 seem to be OTA-upgradable, while the Surfou are not, so the announced OTA-update always fails.

oberschwabe, 2024-09-18:
Hi Quensen, 
i am not really sure if it is the Surfou Plug itself which is the problem as since yesterday also a new Blitzwolf TUYA TS011F plug shows the same behaviour and i already have 3-4 of the same type in use.
I have the fealing something is wrong with zigbee2mqtt. As long as Z2M is in pairing mode the devices are discovered and do the interview and shows at the end "Successfully configured '0xa4c138c8c2d6d079'". 10 secs later the message " Device '0xa4c138c8c2d6d079' left the network" appears. It does the cycle again and again until the pairing mode of Z2M ends.
I restarted Z2M, rebooted PI, reseated the Sonoff Zigbee stick but didn´t solve the issue.

quensen, 2024-09-18:
Have you tried pairing to the coordinator directly/only instead of the pairing to all routers?
Have you tried to start with new configuration (delete all files from the data folder after a backup and use only minimal configuration)?
Have you tried to update Z2M to the latest version?
Have you tried to update your SONOFF dongle to 7.4.4?

oberschwabe, 2024-10-12:
not sure what exactly resolved the issue. After i updated to the Sonoff Dongle to the latest version of Koenkks firmware and bought him a coffee - everything works fine again. The Surfou plug joins the network without any issue.

closed 2024-10-12 by oberschwabe