Koenkk/zigbee2mqtt #23277: [TS130F] Regression since last update

DarkrysVakarian, 2024-07-07:
### What happened?

Hello,

I use Z2M on Home Assistant, and it seems that there is a regression since the last update concerning the TS130F shutter controls.

Z2M no longer indicates the correct state for the controls.
For example, today, out of my 7 shutters, 2 returned the correct state and all the others indicated the opposite state (closed when they were open).

Using the specific Invert Covert parameter also gives false results.

Thnaks for your help.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Home Assistant Yellow built-in adapter

### Setup

Add-on on Homa Assistant

adrienvsj, 2024-07-13:
@Koenkk same error here. But I notice this problem since HomeAssistant 2024.6.0.

Koenkk, 2024-07-15:
Could you provide the debug log when closing them and the incorrect state is reported?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

DarkrysVakarian, 2024-07-15:
Hello,

There is a debug file when I close my shutter maned "Volet roulant - Véranda porte".
[log.txt](https://github.com/user-attachments/files/16241376/log.txt)

But after all I don't know if it's Z2M fault or HA, because on Z2M the state seams to be right : 
[image]

But HA say it's open : 
[image]

@adrienvsj hum you maybe right, I've the problem since a HA update

Thanks :)

Koenkk, 2024-07-16:
Is this without `invert_cover` being set? When:
- cover is closed, position should be 0
- cover is open, position should be 100

HA docs; https://www.home-assistant.io/integrations/cover.mqtt/#position_closed

DarkrysVakarian, 2024-07-16:
Indeed, my screen was taken witout invert_covert set.

Koenkk, 2024-07-17:
Then it looks to be an HA issue

adrienvsj, 2024-07-17:
Hello @Koenkk @DarkrysVakarian 
I just found the commit (https://github.com/home-assistant/core/pull/118044) on homeassistant core that produces the bug. It has been present since HomeAssistant version 2024.6.0b0, as I suspected.
I'm going to open an issue about this bug and I'm currently working on a fix.
In the meantime, I advise you to go back to version 2024.5.X (I think the latest version is 2024.5.5).

DarkrysVakarian, 2024-07-17:
Hello @Koenkk @adrienvsj 
Thanks for your reply.

@adrienvsj  I just donwdrading to version 2024.5.5 thank you.

Koenkk, 2024-07-18:
Great that you've found the root cause, then I will close this as it's not something that can be fixed by z2m.

closed 2024-07-18 by Koenkk

adrienvsj, 2024-07-19:
@DarkrysVakarian for your information I have opened a PR to fix the problem: https://github.com/home-assistant/core/pull/122192

ElCaplar, 2024-08-16:
Update for this problem. Since the last version i have updated : 
Core 2024.8.1
Supervisor 2024.08.0
Operating System 13.0
Interface utilisateur 20240809.0
Version de Zigbee2MQTT [1.39.1]

Everything seems to be ok.