Koenkk/zigbee2mqtt #23279: Wrong firmware version displayed

Diaoul, 2024-07-07:
### What happened?

* I had a Legrand 067773 paired to zigbee2mqtt showing version 0x2a
* I removed it and paired it to the Legrand Gateway, it's now showing version 0x46 (70 in decimal)
* I removed it and paired it again to zigbee2mqtt, it's showing version 0x2a

Sometimes, it happened that I updated some devices outside zigbee2mqtt and when I paired them again they were still showing old firmware information.

### What did you expect to happen?

The correct firmware version is displayed

### How to reproduce it (minimal and precise)

This would be hard to reproduce... However, I'm interested in knowing when is a firmware version refreshed from the device? It smells like a cache invalidation issue.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20230507

### Adapter

tube_zb_gw_cc2652p2_poe

### Setup

Home Kubernetes

Diaoul, 2024-07-07:
@Koenkk could this be causing other issues like on checking firmware versions online? :thinking:

Loic691, 2024-10-08:
+1
Legrand battery device have very very strange issues....
It's not working at home...
Battery drain, event not catched, pairing issues,....

closed 2024-12-28 by github-actions[bot]