Koenkk/zigbee2mqtt #23282: 1.39.0 random crashes

Ricc68, 2024-07-07:
### What happened?

Everything was working fine on 1.38.0-dev but since I switched to 1.39.0, z2m started crashing randomly every couple of hours.
The log shows a random crash started with a transaction failed error followed by a crash related to herdsman followed by a tentative restart that crashed again on herdsman.

### What did you expect to happen?

No crashes.

### How to reproduce it (minimal and precise)

It started happening with ZBDongle-E, fw 7.4.3 and z2m 1.39.0 on ember.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

7.4.3

### Adapter

zbdongle-e 

### Setup

HAOS in vm

### Debug log

2024-07-06 16:58:15] info: 	zh:ember: [NCP COUNTERS] 418,478,137,24,0,0,14,14,96,6,0,0,60,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
[2024-07-06 16:58:15] info: 	zh:ember: [ASH COUNTERS] 219,153,7,146,0,0,0,0,4545,146,146,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
[2024-07-06 17:58:30] error: 	zh:ember:ezsp: ERROR Transaction failure; status=undefined. Last: [FRAME: ID=101:"READ_AND_CLEAR_COUNTERS" Seq=228 Len=5].
[2024-07-06 17:58:30] error: 	zh:ember: !!! NCP FATAL ERROR reason=undefined. ATTEMPTING RESET... !!!
[2024-07-06 17:58:30] info: 	zh:ember:queue: Request dispatching stopped; queue=1 priorityQueue=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Total frames: RX=23, TX=25
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   DATA frames : RX=23, TX=2
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   DATA bytes  : RX=734, TX=32
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=23
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-07-06 17:58:30] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
Error: 
    at Ezsp.ezspReadAndClearCounters (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:2012:19)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:844:42)
    at EmberRequestQueue.dispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/requestQueue.ts:117:46)
[21:18:30] INFO: Preparing to start...
[21:18:30] INFO: Socat not enabled
[21:18:32] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-07-07 21:18:40] info: 	z2m: Logging to console, file (filename: log.log)
[2024-07-07 21:18:40] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #unknown)
[2024-07-07 21:18:40] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-07-07 21:18:41] info: 	zh:ember: Using default stack config.
[2024-07-07 21:18:41] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-07-07 21:18:41] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-07-07 21:18:41] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-07 21:18:41] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-07-07 21:18:41] info: 	zh:ember:uart:ash: Serial port opened
[2024-07-07 21:18:41] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-07 21:18:44] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-07 21:18:44] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-07 21:18:46] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-07 21:18:46] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-07 21:18:49] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-07 21:18:49] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-07 21:18:51] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-07 21:18:51] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-07 21:18:54] error: 	z2m: Error while starting zigbee-herdsman
[2024-07-07 21:18:54] error: 	z2m: Failed to start zigbee
[2024-07-07 21:18:54] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-07-07 21:18:54] error: 	z2m: Exiting...
[2024-07-07 21:18:54] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:889:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2722:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:129:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)

Ricc68, 2024-07-07:
Update: after a restart of the VM, z2m started properly.
I have now enabled the debug log to catch more info about the issue.

maciekdnd, 2024-07-12:
I had similar issue but on SLZB-06M, latest firmware (updated from github repo to 7..4.3 instead from controller console which is probably 7.4.1, on Z2M 1.38.0). After that I got random crashes especially on LXC Z2M reboot. Moving to 20240507 FW (7.4.1) seems more stable.

Nerivec, 2024-07-21:
@Ricc68 Did the restart definitely fix this problem?
If not, can you check your last log from running `1.38.0-dev` for the commit hash?
Example: `z2m: Starting Zigbee2MQTT version 1.39.0-dev (commit #cda8927d)`
It will point us to [one of these commits](https://github.com/Koenkk/zigbee2mqtt/commits/dev) to see what's changed since.

_As you can see from June 1st to June 30th [in that list](https://github.com/Koenkk/zigbee-herdsman/pulls?q=is%3Apr+is%3Aclosed+author%3ANerivec), 1.39.0 had only minimal changes for `ember`.
[August release (current dev)](https://github.com/Koenkk/zigbee2mqtt/discussions/23224) will be a big one however, it has a lot of fixes from adding support for v8, and the implementation of concurrency (should smooth things out for users with misbehaving devices)._

Ricc68, 2024-07-22:
> @Ricc68 Did the restart definitely fix this problem? If not, can you check your last log from running `1.38.0-dev` for the commit hash? Example: `z2m: Starting Zigbee2MQTT version 1.39.0-dev (commit [...]

@Nerivec Since I have restarted and enabled the debug log, the problem disappeared. Now I have disabled the debug: let's see if it appears again.
My feeling is that it was a malfunction of the serial communication with the dongle because a simple restart without power down of HAOS was fixing it.

swissjon, 2024-07-22:
I am having a similar problem. While my devices are quickly available after a reboot by mqtt, the front end can take up to 1 minute to load. I just downgraded to 1.37 and the front end load withing 5-10 seconds of starting.

I have 2 instances in docker with POE coordinators, about 100 devices on each. Same problem with both instances.

Nerivec, 2024-07-22:
@swissjon That issue should already be fixed in dev/edge (caused by a quirk in Availability unearthed after fixing the startup flow).

Ricc68, 2024-07-23:
@Nerivec With log at info level, z2m crashed in less than 24h:

[2024-07-22 21:38:53] error: 	zh:ember:ezsp: ERROR Transaction failure; status=undefined. Last: [FRAME: ID=40:"GET_NETWORK_PARAMETERS" Seq=120 Len=5].
[2024-07-22 21:38:53] error: 	zh:ember: !!! NCP FATAL ERROR reason=undefined. ATTEMPTING RESET... !!!
[2024-07-22 21:38:53] info: 	zh:ember:queue: Request dispatching stopped; queue=1 priorityQueue=1
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Total frames: RX=204, TX=249
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   DATA frames : RX=204, TX=46
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   DATA bytes  : RX=5137, TX=1225
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=203
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-07-22 21:38:53] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
Error: 
    at Ezsp.ezspGetNetworkParameters (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:2659:19)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2791:55)
    at EmberRequestQueue.dispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/requestQueue.ts:117:46)

Attempting to restart z2m results again in a crash as if the communication with the dongle is somehow stuck:

[2024-07-23 21:49:10] error: 	z2m: Error while starting zigbee-herdsman
[2024-07-23 21:49:10] error: 	z2m: Failed to start zigbee
[2024-07-23 21:49:10] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-07-23 21:49:10] error: 	z2m: Exiting...
[2024-07-23 21:49:11] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:889:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2722:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:129:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)

A soft reboot of the HAOS VM, without removing power to the dongle, fixed the issue.

Nerivec, 2024-07-23:
Would need debug logs to confirm, but from what I'm seeing, I'd say the adapter is disconnecting abruptly (or at least that's what Z2M seems to perceive).
Can you check `dmesg` to see if anything weird is happening with the USBs?

swissjon, 2024-07-24:
> @swissjon That issue should already be fixed in dev/edge (caused by a quirk in Availability unearthed after fixing the startup flow in last release).

Thanks. Now that my setup is back to stable, I will probably just wait for the next stable release.

Ricc68, 2024-07-24:
@Nerivec I have checked the host dmesg and there is no sign of usb problems at the time of the crash. (Two days ago) I have only found two usb resets compatible with the first restart attempt and the second successful VM restart of yesterday.

> Would need debug logs to confirm, but from what I'm seeing, I'd say the adapter is disconnecting abruptly (or at least that's what Z2M seems to perceive). Can you check `dmesg` to see if anything weird is happening with the USBs?

@Nerivec I have checked the host dmesg and there is no sign of usb problems at the time of the crash. (Two days ago) I have only found two usb resets compatible with the first restart attempt and the second successful VM restart of yesterday.

Nerivec, 2024-07-24:
You may want to check other available logs too, like the HA ones. In case something interesting pops up around the time of the crash.

You switched back from edge, so, no chance the edge instance is trying to start again (watchdog) or anything along these lines, that could interfere with the regular one? _It's a bit far-fetched, and shouldn't, but just in case..._

Make sure to set the Z2M log level to debug in case it happens again, so we can get more details about the error (the stuff that happens right before what you posted above).

_You could also try the edge add-on again (to update it, just uninstall it, go back to the store, check for updates -top right corner dropdown-, then install it again). See if it happens there too._

Ricc68, 2024-07-25:
@Nerivec it happened again. I have checked host and VM dmesg and HA logs but nothing happened to the usb stick, the logs are not showing anything related.
No, the edge z2m was uninstalled before installing the release version. I don't enable the watchdog.
I will re enable the debug log of z2m and see if it happens but in my previous try with debug log enabled it did not happen for weeks and with debug log disabled it happens in less than 24 hours so my guess is that maybe is a timing issue that is mitigated by the debug log.

Ricc68, 2024-07-29:
@Nerivec I've got the crash with debug enabled finally.

[2024-07-28 23:53:00] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2024-07-28 23:53:00] debug: 	zh:ember:ezsp: ===> [FRAME: ID=101:"READ_AND_CLEAR_COUNTERS" Seq=76 Len=5]
[2024-07-28 23:53:00] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=4 frmRx=1]
[2024-07-28 23:53:14] debug: 	zh:ember:ezsp: =x=> [FRAME: ID=101:"READ_AND_CLEAR_COUNTERS" Seq=76 Len=5] Error: Error: timed out after 14400ms
[2024-07-28 23:53:14] error: 	zh:ember:ezsp: ERROR Transaction failure; status=undefined. Last: [FRAME: ID=101:"READ_AND_CLEAR_COUNTERS" Seq=76 Len=5].
[2024-07-28 23:53:14] error: 	zh:ember: !!! NCP FATAL ERROR reason=undefined. ATTEMPTING RESET... !!!
[2024-07-28 23:53:14] info: 	zh:ember:queue: Request dispatching stopped; queue=1 priorityQueue=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Total frames: RX=38, TX=46
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   DATA frames : RX=38, TX=8
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   DATA bytes  : RX=1047, TX=187
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=38
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-07-28 23:53:14] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
Error: 
    at Ezsp.ezspReadAndClearCounters (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:2012:19)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:844:42)
    at EmberRequestQueue.dispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/requestQueue.ts:117:46)

Ricc68, 2024-07-29:
@Nerivec trying to restart gives another crash:

[2024-07-29 22:22:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2024-07-29 22:22:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2024-07-29 22:22:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2024-07-29 22:22:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2024-07-29 22:22:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2024-07-29 22:22:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2400/2500
[2024-07-29 22:22:49] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-29 22:22:49] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-07-29 22:22:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-07-29 22:22:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-07-29 22:22:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-07-29 22:22:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-07-29 22:22:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-07-29 22:22:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-07-29 22:22:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-07-29 22:22:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-07-29 22:22:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2024-07-29 22:22:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2024-07-29 22:22:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2024-07-29 22:22:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2024-07-29 22:22:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2024-07-29 22:22:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2024-07-29 22:22:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2024-07-29 22:22:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2024-07-29 22:22:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2024-07-29 22:22:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2024-07-29 22:22:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2024-07-29 22:22:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[2024-07-29 22:22:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[2024-07-29 22:22:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
[2024-07-29 22:22:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2400/2500
[2024-07-29 22:22:52] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-29 22:22:52] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-29 22:22:52] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-07-29 22:22:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-07-29 22:22:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-07-29 22:22:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-07-29 22:22:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-07-29 22:22:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-07-29 22:22:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-07-29 22:22:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-07-29 22:22:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-07-29 22:22:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-07-29 22:22:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-07-29 22:22:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2024-07-29 22:22:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2024-07-29 22:22:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2024-07-29 22:22:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2024-07-29 22:22:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2024-07-29 22:22:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2024-07-29 22:22:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2024-07-29 22:22:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2024-07-29 22:22:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2024-07-29 22:22:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2024-07-29 22:22:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2024-07-29 22:22:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[2024-07-29 22:22:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[... cut ...]
[2024-07-29 22:22:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[2024-07-29 22:22:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
[2024-07-29 22:22:57] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2400/2500
[2024-07-29 22:22:57] error: 	z2m: Error while starting zigbee-herdsman
[2024-07-29 22:22:57] error: 	z2m: Failed to start zigbee
[2024-07-29 22:22:57] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-07-29 22:22:57] error: 	z2m: Exiting...

Nerivec, 2024-07-29:
Do you have the rest of the logs (from [this post](https://github.com/Koenkk/zigbee2mqtt/issues/23282#issuecomment-2256821047))?

Bolukan, 2024-07-30:
I have problems since 1.39.0: my guess: I can send commands to devices (turn light on/off) but do not receive messages back (no online/offline, no interviews, etc.). Did reboots and reinstallations but no success.
I just reverted back to 1.38.0 and see what happens. First signs are ok: communication is restored.

Ricc68, 2024-07-31:
@Nerivec I am attaching the full log.

[log.txt.gz](https://github.com/user-attachments/files/16446869/log.txt.gz)

Nerivec, 2024-07-31:
Something's not quite right here.

```
[2024-07-28 22:59:56]
[2024-07-28 23:53:00]
```

Zero logging in-between?

Ricc68, 2024-07-31:
@Nerivec that was grepped -i -v mqtt to remove useless mqtt info.
Now this new attach is complete, not grepped, and yes, the only activity in between is the backup of the state, it seems the z2m communication with the zbdongle-e is stalled.

[log.log.gz](https://github.com/user-attachments/files/16447333/log.log.gz)

Nerivec, 2024-08-01:
Rebooting Z2M reboots the dongle (they sync-up during startup).
But it seems everything just cuts off, not just the communication to the adapter, Z2M sending, and receiving alike, since there is no logging. Only a couple of time-based triggers remain.

Did you check all available logs (host/vm) around 2024-07-28 23:00:00 for some event that could possibly precipitate this behavior?

Ricc68, 2024-08-01:
> Rebooting Z2M reboots the dongle (they sync-up during startup). But it seems everything just cuts off, not just the communication to the adapter, Z2M sending, and receiving alike, since there is no [...]

Checked the logs but found nothing  that can possibly be related.
The fact that it is happening much less frequently in debug log than in info log may be a hint?

Ricc68, 2024-08-06:
@Nerivec happened also with 1.39.1.
[log.zip](https://github.com/user-attachments/files/16511378/log.zip)
Log attached.

Koenkk, 2024-08-06:
@Ricc68 when the crash happens, do you see anything happening in `sudo dmesg`?

Ricc68, 2024-08-07:
> @Ricc68
> 
>     * when the crash happens, do you see anything happening in `sudo dmesg`?
> 
>     * Could you check if the crash also happens when running on different hardware? (maybe you have some Pi lying around?)

@Koenkk I have checked the dmesg and there's nothing at the time of the crash.

Unfortunately I don't have other hardware on which run haos.

Koenkk, 2024-08-08:
@Ricc68 there is no need to run haos, just z2m on bare metal would be enough already ([docs](https://www.zigbee2mqtt.io/guide/getting-started/))

Ricc68, 2024-08-08:
> @Ricc68
> 
>     * when the crash happens, do you see anything happening in `sudo dmesg`?
> 
>     * Could you check if the crash also happens when running on different hardware? (maybe you have some Pi lying around?)

@Koenkk I have checked the dmesg and there's nothing at the time of the crash.

Unfortunately I don't have other hardware on which run haos.

> @Ricc68 there is no need to run haos, just z2m on bare metal would be enough already ([docs](https://www.zigbee2mqtt.io/guide/getting-started/))

@Koenkk the problem is that I don't have any available hardware to run it. Just some spare esp8266 which is useless for the task.

Koenkk, 2024-08-09:
You can also run it on a laptop (even on Windows)

Ricc68, 2024-09-20:
@Koenkk I have to report that I still see this issue on 1.40.1 with crashes every 24h or so.

Ricc68, 2024-09-21:
@Koenkk I will try to run z2m on wsl-linux to have more control in case we want to add debug code and see if I can reproduce it.

Ricc68, 2024-09-22:
@Nerivec I have run the test suggested by @Koenkk and after 24 hours on a debian-wsl z2m was still happily running.
On my low resources NAS it takes only a few hours to crash.
I have moved the dongle back to the NAS and in the while I have updated the dongle firmware to 7.4.4.
If it happens also with the new fw I wonder if there's some debug code that we can run on my z2m HAOS installation to try to nail down the problem.

Ricc68, 2024-10-13:
@Koenkk It is still happening with firmware 7.4.4 and z2m 1.14.2. I think it must be related to the limited resources of the NAS.
Is there anything I can do to help debug the issue?
We basically need to understand what's happening during that 2 minutes of silence in the logs between the last dongle reply and the crash.

Koenkk, 2024-10-14:
I'm not sure if it's the low resources, it seems the serial disconnects on your NAS somehow (I'm not sure how to debug that further).

Ricc68, 2024-10-19:
@Koenkk I have setup interceptty to monitor the 2 way serial communication between the dongle and z2m. Let's see what happens when the crash happens, like who sends the last message and if the message is correct or corrupt or whatever may hint of an issue on either side of the communication channel.
This might give some additional info on what's going on.

Ricc68, 2024-11-01:
@koenkk I have moved the HAOS in a proxmox server with plenty of resources and it is happening also in this setup.
I had captured the serial log on the NAS:

[06:11 19.422113]>      73(s):80(<80>):b1(±):a9(©):7d(}):31(1):2a(*):15([NAK]):b6(¶):58(X):90(<90>):41(A):24($):ab(«):15([NAK]):93(<93>):49(I):9c(<9c>):e1(á):47(G):1f([US]):6e(n):69(i):98(<98>):74(t):f5(õ):de(Þ):0b([VT]):83(<83>):f4(ô):7b({):1e([RS]):87(<87>):f0(ð):cf(Ï):e6(æ):ed(í):7e(~)
[06:11 19.428862]>      03([ETX]):80(<80>):b1(±):a9(©):7d(}):31(1):2a(*):15([NAK]):b6(¶):58(X):90(<90>):41(A):24($):ab(«):15([NAK]):93(<93>):49(I):9c(<9c>):fe(þ):7f([DEL]):19([EM]):6e(n):69(i):98(<98>):74(t):f5(õ):de(Þ):0a([LF]):83(<83>):f7(÷):7b({):16([SYN]):b7(·):ed(í):cf(Ï):bb(»):96(<96>):7e(~)
[06:11 19.429020]< 80(<80>):70(p):78(x):7e(~)
[06:11 19.474177]< 81(<81>):60(`):59(Y):7e(~)
[06:11 24.432634]>      7d(}):33(3):80(<80>):b1(±):a9(©):0d([CR]):2a(*):96(<96>):15([NAK]):3d(=):c4(Ä):da(Ú):27('):4b(K):6d(m):53(S):ed(í):f4(ô):f8(ø):27('):9b(<9b>):27('):7e(~)
                 >      23(#):80(<80>):b1(±):a9(©):7d(}):31(1):2a(*):15([NAK]):b6(¶):58(X):90(<90>):41(A):24($):ab(«):15([NAK]):93(<93>):49(I):9c(<9c>):ff(ÿ):4f(O):1d([GS]):6e(n):69(i):98(<98>):74(t):f5(õ):de(Þ):09([TAB]):83(<83>):f4(ô):7b({):1e([RS]):d5(Õ):f1(ñ):cf(Ï):c3(Ã):69(i):7e(~)
[06:11 24.434719]>      33(3):80(<80>):b1(±):a9(©):7d(}):31(1):2a(*):15([NAK]):b6(¶):58(X):90(<90>):41(A):24($):ab(«):15([NAK]):93(<93>):49(I):9c(<9c>):fc(ü):4f(O):1d([GS]):6e(n):69(i):98(<98>):74(t):f5(õ):de(Þ):08([BS]):83(<83>):f7(÷):7b({):16([SYN]):4d(M):ee(î):cf(Ï):6f(o):7d(}):33(3):7e(~)
[06:11 24.434789]< 83(<83>):40(@):1b([ESC]):7e(~)
[06:11 24.482267]< 84(<84>):30(0):fc(ü):7e(~)
[06:11 29.422832]>      43(C):80(<80>):b1(±):a9(©):7d(}):31(1):2a(*):15([NAK]):b6(¶):58(X):90(<90>):41(A):24($):ab(«):15([NAK]):93(<93>):49(I):9c(<9c>):fd(ý):7b({):7d(}):38(8):6e(n):69(i):98(<98>):74(t):f5(õ):de(Þ):0f([SI]):83(<83>):f4(ô):7b({):1e([RS]):9d(<9d>):f0(ð):cf(Ï):ce(Î):ee(î):7e(~)
[06:11 29.428786]< 85(<85>):20( ):dd(Ý):7e(~)
[06:11 29.432863]>      53(S):80(<80>):b1(±):a9(©):7d(}):31(1):2a(*):15([NAK]):b6(¶):58(X):90(<90>):41(A):24($):ab(«):15([NAK]):93(<93>):49(I):9c(<9c>):fa(ú):7b({):7d(}):38(8):6e(n):69(i):98(<98>):74(t):f5(õ):de(Þ):0e([SO]):83(<83>):f7(÷):7b({):16([SYN]):b4(´):ed(í):cf(Ï):8e(<8e>):b1(±):7e(~)
[06:11 29.461513]< 86(<86>):10([DLE]):be(¾):7e(~)
[06:11 37.789549]>      63(c):80(<80>):b1(±):a9(©):7d(}):31(1):2a(*):15([NAK]):b6(¶):58(X):96(<96>):4d(M):24($):ab(«):55(U):93(<93>):49(I):9c(<9c>):fb(û):53(S):12([DC2]):6e(n):69(i):98(<98>):74(t):f1(ñ):ce(Î):0d([CR]):83(<83>):fc(ü):7d(}):5e(^):7d(}):3a(:):4c(L):ed(í):cd(Í):de(Þ):6f(o):8f(<8f>):fd(ý):14([DC4]):c1(Á):7e(~)
[06:11 37.791696]< 87(<87>):00([NUL]):9f(<9f>):7e(~)
[06:11 37.812564]< 37(7):81(<81>):21(!):a9(©):60(`):2a(*):15([NAK]):31(1):fe(þ):90(<90>):4b(K):27('):ad(­):54(T):93(<93>):49(I):8d(<8d>):4e(N):27('):ab(«):9d(<9d>):cb(Ë):7f([DEL]):e5(å):f6(ö):cc(Ì):63(c):91(<91>):a6(¦):7e(~)
[06:11 37.822645]>      74(t):81(<81>):a1(¡):a9(©):60(`):2a(*):15([NAK]):c4(Ä):c6(Æ):cc(Ì):7e(~)
[06:11 37.826145]< 80(<80>):70(p):78(x):7e(~)
[06:11 37.829903]>      04([EOT]):81(<81>):b1(±):a9(©):6b(k):2a(*):15([NAK]):31(1):fe(þ):90(<90>):4b(K):27('):ad(­):54(T):93(<93>):49(I):8d(<8d>):4e(N):27('):dd(Ý):9d(<9d>):ce(Î):67(g):a8(¨):76(v):7e(~)
[06:11 37.831486]< 81(<81>):60(`):59(Y):7e(~)
[06:49 13.147603]< 41(A):86(<86>):21(!):a9(©):31(1):2a(*):1f([US]):8e(<8e>):7e(~)

The serial log shows that seems the dongle after [06:11 37.831486] stops responding to z2m messages.
Now the issue is happening also on the proxmox, I wonder if there is some firmware issue on the dongle firmware (I have 7.4.4) that for some reason is halting the serial communication on the dongle side (or a dongle hardware issue? but that's strange because a reset of the dongle restarts the communication).
The hardware connection (USB device) is not disconnecting when the issue happens, it's just that the dongle stops responding over the serial.

Koenkk, 2024-11-02:
The Ember ack frames start with `0x80` so it seems `<` is Z2M and `>` is the adapter, looks like the adapter stops responding.

Ricc68, 2024-11-02:
Ok this explains when the first crash happens the reset attempt that obviously fails:

[2024-07-28 23:53:14] error: zh:ember: !!! NCP FATAL ERROR reason=undefined. ATTEMPTING RESET... !!!

Now, if the reset was successful I guess that the situation could be resolved. Obviously the z2m reset failed because subsequent attempts to restart z2m fail.
But why the z2m reset fails and when I soft restart the VM, which you taught me it resets the dongle, afterwards the communication is restored?
What's the difference between the z2m reset and the OS reboot reset?
If the serial communication is not working, the reset must be on the hardware lines, maybe driving the lines low or high so to instruct the dongle firmware to reset.
Do we know what is the reset procedure?

Nerivec, 2024-11-03:
How about `autosuspend` as the cause of the issue (default is different from setup to setup...)?
`cat /sys/module/usbcore/parameters/autosuspend` (should return `-1` if it's disabled, else not)

Ricc68, 2024-11-04:
I have checked the auto suspend and actually it is set to 2. Should I change it to -1 to disable it?

Nerivec, 2024-11-04:
You should try. Just have to find what way is the right way for your setup (differs from OS to OS).

Ricc68, 2024-11-04:
Ok, I'm running Home Assistant OS so there's not much to configure as every time the OS is updated, everything except HA configuration is overwritten.
I'll research if it possible to set kernel parameters in a retentive way in haos.

Ricc68, 2024-11-04:
Ok I have found that in Home Assistant OS which is based on Alpine Linux, the usbcore module is compiled into the kernel so to persistently set usbcore.autosuspend=-1 it needs to be set at the boot time kernel command line. We need to see if this survives an Home Assistant OS update.

This is achieved by this procedure:

1. login into the Home Assistant OS console (not the HA terminal which is inside the docker)
2. cd /mnt/boot
_Note that in the current Home Assistant OS 13.2 this directory is read-write. If it will ever become read-only in future, a mount -o remount,rw /mnt/boot becomes necessary before proceeding_
3. edit cmdline.txt adding the kernel parameter usbcore.autosuspend=-1
    **a big WARNING here: an incorrect edit of the cmdline.txt file can render the system unbootable, or worse**

[image]

4. save and reboot

This way the parameter is set boot time as an argument passed to the Linux kernel:

[image]

where device 2-1 is my ITHEAD USB ZBDongle-E and the parameter setting is surviving reboots.

I have re-enabled the direct serial communication (disabling the interceptty logging): now let's see if it happens again or this fixes it.

Note that the boot partition is accessible only from the Home Assistant OS console because the user within the HA docker do not have enough permissions:

[image]

the only allowed thing is to check the parameter value:

[image]

Ricc68, 2024-11-09:
@Nerivec also this tentative did not work.
Reasoning about this usb power saving, the default is 2, which means 2000 ms. It is enough to chat with the dongle within 2 seconds that the power save feature do not kick in.
I am more afraid of something that happens with the dongle firmware because I have re-checked the proxmox syslog and the vm syslog and there is absolutely nothing happening to the usb when the dongle stops responding.
I will try to update the dongle to 8.0.2 and see if it fixes the issue as I saw from the changelog that it is quite an overhaul of the firmware architecture.

Nerivec, 2024-11-09:
Since you did not have the problem with the other setup, if it is something with the firmware, it would have to be something that only happens in combination with this setup. While possible, I'd say unlikely.

It's far more likely that the setup itself is doing something that does not agree with the adapter or Z2M or nodejs...
Can you see anything about node in system logs around the time of crash? Or anything at all pointing to a state change, a crash of some kind?

Since it does look like something is "going to sleep", if it's not the USB, maybe it's some kind of power saving feature on the NAS that effectively stops a critical process for Z2M? Or something is silently crashing...

About the reset, the OS would restart the whole machine, including all hardware init. A Z2M restart only restarts Z2M's logic, and syncs up the adapter's state by asking it to reboot (basically). If serial communication is prevented in any way, only the OS reset would do anything.

Ricc68, 2024-11-09:
> Since it does look like something is "going to sleep", if it's not the USB, maybe it's some kind of power saving feature on the NAS that effectively stops a critical process for Z2M? Or something is silently crashing...

And that's the strange thing: I have migrated one week ago the HA OS VM from the NAS (highly custom QNAP linux OS, low resources) to a Ryzen 5600U mini PC running Proxmox (Debian bookworm based, plenty of resources) and the issue is still happening.

So I would exclude a problem with the host OS or hardware (different OS and hardware).
What remains is the HA OS, z2m and the dongle itself.

I have checked the logs. The crash happened yesterday at 22:33:38 CET.
In the z2m addon log I have found:

[image]

while in the HA OS journalctl I have (timezone is behind by 1 hour in HA OS):

[image]

**Note that one minute before the issue, there was some housekeeping of Home Assistant: this is the only thing that seems a coincidence, HA OS is running these checks every hour so it's hard to think that they are related.**
I have also checked previous lines of this log, up to 1 hour before, and found nothing special.

Finally, in the Proxmox host journalctl there's nothing:

[image]

Ricc68, 2024-11-16:
@Nerivec @Koenkk, an update on this.

With dongle firmware 8.0.2 build from Nerivec, z2m is running for a week without issues on my proxmox, which is a good news.
We'll see if it can stay solid for more.

Ricc68, 2024-11-19:
@Koenkk It failed after 11 days, incredible, I was thinking that finally it was stable!
Why z2m took more than 30 minutes of silence before realising the timeout?

[2024-11-17 21:58:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lavatrice', payload '{"child_lock":"UNLOCK","countdown":0,"current":0.86,"energy":26.59,"indicator_mode":"off/on","linkquality":144,"power":130,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":225}'
[2024-11-17 22:30:24] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=222 Len=10].
[2024-11-17 22:30:24] error: 	zh:ember: Adapter fatal error: ASH_ERROR_TIMEOUTS

It doesn't seem quite right. Z2m should recognise the timeout much earlier, no?
I'm out of options. Any suggestion? Maybe I can reboot the proxmox HAOS VM with crontab once a day, but that's a last resort workaround, not something I'd like to do.

Koenkk, 2024-11-20:
@Ricc68 do you have any other addons installed in HA? If yes, what are those?

Ricc68, 2024-11-20:
@Koenkk yes I have these add-ons:

- file editor 
- matter server
- mosquito broker
- terminal && ssh
- z2m

Then I have HACS with:

- local tuya
- Ariston net

Nerivec, 2024-11-20:
`ASH_ERROR_TIMEOUTS` could be just what it is, a timeout because a packet was dropped or something like that. Since it lasted 11 days, we can't assume that 1 timeout is a repeat of the old pattern.
However, it may also indicate that something was missing in those 11 days, and was reintroduced recently. Anything comes to mind?

Ricc68, 2024-11-20:
@Nerivec yesterday I tried to restart z2m but had the same host error:

[2024-11-19 22:37:23] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.

Means that the dongle was not responding.
Had to reboot the VM to restart z2m so it looks like it was not just a randomly dropped packet.
I'm intentionally not changing anything during these tests so nothing happened to the setup in these 11 days.

Nerivec, 2024-11-20:
And it is looping the ASH reset like before?
That other thread you posted in, it turned out to be another HA integration... No chance you could have a remnant of an integration (that would mess with USB) somewhere that's still running/trying to run? The behavior is a bit too similar to be a coincidence... although those 11 days without issue are the "anomaly" now...

Ricc68, 2024-11-21:
Yes, it was not able to restart giving host fatal error.
I'm waiting for it to happen again and I will try to see if something else is using the serial with lsof.
I can also try something with strace to see low level what happens when I try to restart and fails.
I saw there's also a cli tool to talk with the dongle so that may be another option to see if the dongle is really unresponsive or what.
I'm certainly not happy to let it be and live with it so as long as there's a chance to catch the problem I'm not desisting.

Ricc68, 2024-11-30:
@Nerivec @Koenkk, this time it happened after 10 days.
Trying to restart z2m (and fails), I see from the host VM that the node opens the ttyACM0 serial (correct):

[image]

but then exits saying:

[2024-11-30 16:54:00] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-11-30 16:54:00] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-11-30 16:54:00] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-30 16:54:00] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-11-30 16:54:00] info: 	zh:ember:uart:ash: Serial port opened
[2024-11-30 16:54:00] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-30 16:54:02] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-30 16:54:02] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-30 16:54:05] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-30 16:54:05] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-30 16:54:07] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-30 16:54:07] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-30 16:54:10] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-30 16:54:10] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-30 16:54:12] error: 	z2m: Error while starting zigbee-herdsman
[2024-11-30 16:54:12] error: 	z2m: Failed to start zigbee
[2024-11-30 16:54:12] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-11-30 16:54:12] error: 	z2m: Exiting...
[2024-11-30 16:54:12] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:684:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)

So nothing is interfering with the serial but still z2m bails out with error.
I then tried with zigpy radio ezsp /dev/ttyACM0 info which returned error RESET_TIMEOUT.
I then tried with zigpy radio ezsp /dev/ttyACM0 reset which again returned error RESET_TIMEOUT.
I tried to unbind/bind the usb device but that did not reset it, same error.
I tried to reset the usb device by echo 0/1 to the /sys/bus/path-to-device/authorized but that unplug/replug from dmesg:

[image]

but the dongle has the same RESET_TIMEOUT error.
Tried also with the bConfigurationValue object of the device in the sys/bus: the kernel shows the reload of the device but again I get RESET_TIMEOUT.
Tried to check if it was autosuspended writing "-1" to device/power/autosuspend interrupts autosuspend but the device is still giving error so it was not in autosuspend.

Finally, out of options I have rebooted the VM and I could access the dongle again.
I have also double checked that zigpy-cli works:

[image]

Looks like the dongle is stuck in a way that only a VM restart can heal.
I tried many ways to reset the usb device without luck.
The only thing that I have not tried is the usbreset which calls an ioctl to reset the usb device.

I did all the tests from inside the HAOS VM.
Next time I will try the same tests on the host Proxmox, but I'm not expecting any different result as the cure was the reboot of the VM, not the Proxmox host.

Nerivec, 2024-11-30:
Did you check everything related to USB, serial, power on the Proxmox host just in case?
You wouldn't happen to have another EmberZNet dongle that you could backup/restore into and try with instead?

Ricc68, 2024-11-30:
I checked everything on the VM, next time I will repeat the tests on the Proxmox host.
No, I don't have a second dongle. I could buy one and use it as a router.

Ricc68, 2024-12-01:
@Nerivec @Koenkk , I've got another idea.
Let's suppose, as I have the doubt now, that the USB passtrough of Proxmox is not as stable as we like it to be.
The configuration of my HAOS VM was in the tests so far: Proxmox -(USB passtrough)->VM->USB-serial->serial for z2m.

Now I have changed the configuration to hold the USB on the Proxmox host and pass only the serial to the VM.
In practice the way to do this is very simple:

1. add a serial0 (or serial1 or 2 or 3) to the proxmox VM hardware.
2. by default the serial is a socker in proxmox hardware so we have to change this parameter in the VM config file /etc/pve/quemu-server/<VMID>.conf to point to the serial device, in my case: "serial0: /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240122184528-if00"
3. in the VM this serial is recognized as ttyS0, but the VM systemd is spawning an agetty on /dev/ttyS0 because in the VM /mnt/boot/cmdline.txt kernel commandline parameters the ttyS0 is set as console (content is "console=ttyS0 console=tty0"). We have to change this cmdline.txt file in the VM so that systemd leaves ttyS0 alone without spawning agetty, therefore we remove the ttyS0 part and the cmdline.txt becomes "console=tty0".

From now on, z2m is able to reach the dongle through /dev/ttyS0 which is a "real" serial port (COM1) inside the VM.

With this configuration I expect no issue from the USB side of this problem as the proxmox host has never had issues with it. Only the HAOS VM had issues with the USB.

Now let's see if HAOS VM has issues with a "real" serial port!!!!!

Makiese, 2024-12-04:
Greetings to everyone. I am by no means well-versed in this topic but am currently struggling with the same issue. I am using a: **Synology DS718+** with **DSM** 7.2.2-72806 Update 2 as the base. 

On this, I am running a VM with **Home Assistant:**

- Core: 2024.11.3
- Supervisor: 2024.11.4
- Operating System: 14.0
- Frontend: 20241106.2

**Dongle (connected via USB 2.0 adapter):**
- Sonoff Zigbee 3.0 USB-Dongle Plus-V 2

with **Firmware:** 
- ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl

[image]

[image]

**Zigbee2MQTT - Protocol**

```
[22:44:27] INFO: Preparing to start...
[22:44:27] INFO: Socat not enabled
[22:44:28] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-12-04 22:44:37] info: 	z2m: Logging to console, file (filename: log.log)
[2024-12-04 22:44:37] info: 	z2m: Starting Zigbee2MQTT version 1.42.0 (commit #unknown)
[2024-12-04 22:44:37] info: 	z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-04 22:44:38] info: 	zh:ember: Using default stack config.
[2024-12-04 22:44:38] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-12-04 22:44:38] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-12-04 22:44:38] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-04 22:44:38] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-12-04 22:44:38] info: 	zh:ember:uart:ash: Serial port opened
[2024-12-04 22:44:38] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-04 22:44:40] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-04 22:44:40] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-04 22:44:43] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-04 22:44:43] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-04 22:44:45] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-04 22:44:45] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-04 22:44:48] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-04 22:44:48] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-04 22:44:51] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-04 22:44:51] error: 	z2m: Failed to start zigbee
[2024-12-04 22:44:51] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-04 22:44:51] error: 	z2m: Exiting...
[2024-12-04 22:44:51] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:684:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)
```

I have already read so many posts, but I can't find a solution. 
Rebooting etc. did not solve the problem either.

Can anyone here help me?

Thnaks al lot in advance.

Nerivec, 2024-12-04:
@Makiese If you have indeed flashed `ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl`, you cannot have `baudrate: 230400` in Z2M configuration. Either set it to 115200, or remove the line (it defaults to 115200 automatically if not present). _Baudrate from firmware and baudrate in Z2M configuration must always match._

Makiese, 2024-12-06:
> @Makiese If you have indeed flashed `ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl`, you cannot have `baudrate: 230400` in Z2M configuration. Either set it to 115200, or remove the line (it defaults to [...]

Thanks for the Important Hint. But nothing changed.

Nerivec, 2024-12-06:
@Makiese before we dig further, can you ensure you have a proper firmware by re-flashing one (to remove the possibility of misconfiguration)?
https://github.com/darkxst/silabs-firmware-builder/releases/download/20241124/zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl is the latest for Dongle-E

[26 more replies]

closed 2025-01-11 by Ricc68